lovyan03/LovyanGFX #191: Cannot compile with platformio at 0.4.12

ToraTokage, 2022-01-20:
Hello,

When I updated the library of platformio's esp-idf project from 0.4.10 to 0.4.12, I got an error with #include "LoyvanGFX.hpp" and could not compile.
The line that sets "Loyvan / GFX" disappears from "c_cpp_properties.json".
When I checked, the "library.json" file did not exist in the directory where this library was installed.
Is this the cause?

lovyan03, 2022-01-20:
@ToraTokage 
Thanks for letting me know.
Hmmm, I'm stumped.
I have never registered LovyanGFX on platformio.
In other words, somehow it is registered without my permission.
So I do not know how to fix it.

Also, I've never deleted library.json, so I don't know why only that file disappeared.
(If you look at the history of the master branch, you'll see that the file has always been there.)

I'm thinking that perhaps I can fix it by raising the version number and putting it on github again, but... I don't know for sure.

ToraTokage, 2022-01-21:
@lovyan03 

Thank you for your reply.

Oops….
What is registered in Platformio is "unofficial version", isn't it?
As you said, It was registered as a master on github, so I thought it was strange ....
I will get the "official version" from github and use LovyanGFX library.

I will close this issue with your answer.

closed 2022-01-21 by ToraTokage

lovyan03, 2022-01-21:
@ToraTokage 
I just tried to use the pio command.
I was able to unpublish 0.4.12 with my operation.
I also tried to publish 0.4.13 with a higher version number.

I tried here and found that 0.4.13 includes library.json.
I'm not sure why library.json disappeared, but I think the problem has been solved.

I would like to thank you again for bringing this issue to my attention.

ToraTokage, 2022-01-21:
@lovyan03 
Thank you for your support! !!
I downloaded version 0.4.13 and confirmed that this issue has been resolved.