lovyan03/LovyanGFX #192: VLWfont::drawChar() requires more height

dotnfc, 2022-01-24:
LovyanGFX\src\lgfx\v1\lgfx_fonts.cpp | VLWfont::drawChar()
LN940          if (y0 > clip_bottom) break;
LN988          if (y0 > clip_bottom) break;

these two lines shoud be changed to: 
LN940          if ((y + y0) > clip_bottom) break;
LN988          if ((y + y0)> clip_bottom) break;
--------------------------------------------------------------------
i used a sprite (e.g.  w = 100, h = 32), with a font ( h = 20 ). if i draw text string on it, i have to change sprite(100, 42) to meet the 
required space for text drawing, otherwise, the text will show a half.

when changed the drawChar() code, no more space required.

Any suggestion?

BTW> any plan to have SDL2 panel supported for Windows/Linux? or any existing code/guide to do so?

lovyan03, 2022-01-25:
@dotnfc 
Thanks for reporting this issue !
I was able to reproduce the bug and have made the fix and updated the develop branch.

Thanks for letting me know about SDL2.
There is no guide or documentation for porting LovyanGFX, but I think it will be similar to porting to OpenCV.

https://github.com/lovyan03/LovyanGFX/tree/develop/src/lgfx/v1/platforms/opencv
I think that copying this folder and modifying it for SDL2 is the easiest way to implement.

dotnfc, 2022-01-26:
ok, i have applied the modification to my project.

and about the SDL2, i have mofied the opencv panel to adapt SDL2 in vs2019, it works.
but, just waiting for your official impl.

closed 2022-01-26 by dotnfc

tobozo, 2022-01-26:
@dotnfc I've been playing with PECL php-sdl2 recently and this makes me very curious :-) 
Can you share your SDL2 implementation?

dotnfc, 2022-01-28:
@tobozo  you can get the experimental sdl2 panel for win32 here:

https://github.com/dotnfc/LovyanGFX

i've testd on vscode, cmake, vs2019 env.