lovyan03/LovyanGFX #202: Bus_Parallel8 not work on ESP32 S2

uldara1, 2022-02-20:
Hi i try to use 8bit parallel on this library but when i upload code ESP32 S2 always restart

when i change to 16bit mode is ok 

`//FPS test
#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include <vector>

#define LCD_CS -1
#define LCD_BLK -1

class LGFX : public lgfx::LGFX_Device
{
  //lgfx::Panel_ILI9341 _panel_instance;
   lgfx::Panel_ST7796      _panel_instance;
  //lgfx::Panel_ILI9488 _panel_instance;
  lgfx::Bus_Parallel16 _bus_instance; // 8ビットパラレルバスのインスタンス (ESP32のみ)

public:
  // コンストラクタを作成し、ここで各種設定を行います。
  // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
  LGFX(void)
  {
    {                                    // バス制御の設定を行います。
      auto cfg = _bus_instance.config(); // バス設定用の構造体を取得します。

      // 16位设置
      cfg.i2s_port = I2S_NUM_0;  // 使用するI2Sポートを選択 (0 or 1) (ESP32のI2S LCDモードを使用します)
      cfg.freq_write = 16000000; // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
      cfg.pin_wr = 39;           // WR を接続しているピン番号
      cfg.pin_rd = -1;           // RD を接続しているピン番号
      cfg.pin_rs = 38;           // RS(D/C)を接続しているピン番号

      cfg.pin_d0 = 36;
      cfg.pin_d1 = 35;
      cfg.pin_d2 = 34;
      cfg.pin_d3 = 33;
      cfg.pin_d4 = 21;
      cfg.pin_d5 = 18;
      cfg.pin_d6 = 17;
      cfg.pin_d7 = 16;

      _bus_instance.config(cfg);              // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance); // バスをパネルにセットします。
    }

    {                                      // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config(); // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs = -1;   // CS要拉低
      cfg.pin_rst = 37;  // RST和开发板RST相连
      cfg.pin_busy = -1; // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      cfg.memory_width = 320;   // ドライバICがサポートしている最大の幅
      cfg.memory_height = 320;  // ドライバICがサポートしている最大の高さ
      cfg.panel_width = 320;    // 実際に表示可能な幅
      cfg.panel_height = 320;   // 実際に表示可能な高さ
      cfg.offset_x = 0;         // パネルのX方向オフセット量
      cfg.offset_y = 0;         // パネルのY方向オフセット量
      cfg.offset_rotation = 0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel = 8; // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits = 1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable = true;      // データ読出しが可能な場合 trueに設定
      cfg.invert = false;       // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order = false;    // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit = true;    // データ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared = true;    // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

LGFX lcd;

void setup(void)
{
  pinMode(LCD_CS, OUTPUT);
  pinMode(LCD_BLK, OUTPUT);

  digitalWrite(LCD_CS, LOW);
  digitalWrite(LCD_BLK, HIGH);

  lcd.init();
}

static int colors[] = {TFT_RED, TFT_GREEN, TFT_BLUE, TFT_CYAN, TFT_MAGENTA, TFT_YELLOW};

int i = 0;

void loop(void)
{
  //fps
  static int prev_sec;
  static int fps;
  ++fps;
  int sec = millis() / 1000;
  if (prev_sec != sec)
  {
    prev_sec = sec;
    lcd.setCursor(0, 320);
    lcd.setTextSize(4);
    lcd.printf("fps:%03d", fps);
    fps = 0;
  }

  lcd.fillRect(0, 0, 320, 320, colors[i++]);
  //lcd.fillScreen(colors[i++]);
  if (i > 5)
    i = 0;
}`

lovyan03, 2022-02-20:
@uldara1 
Thanks for reporting this issue !

I was not aware of this problem because I had never run it with the pin_rd setting set to -1.
I have made a fix in the develop branch, so please try it out.

uldara1, 2022-02-20:
@lovyan03 thank for your reply 
i try your update chip not restart anymore but display not display correct
as example code 
pls check 

[image]

lovyan03, 2022-02-20:
If your sample is accurate, then there is a mistake here.
```
  lgfx::Bus_Parallel16 _bus_instance;
```
```
  cfg.dlen_16bit = true;
```

Use Bus_Parallel8 and set false to dlen_16bit.

uldara1, 2022-02-21:
[image]

It work now thank you 🙏

lovyan03, 2022-02-21:
There was one thing that I forgot to tell you.

```
 cfg.memory_height = 320;
```

 This should be changed to ` 480 ` . Otherwise, the coordinates will be broken when setRotation is used.

uldara1, 2022-02-21:
https://user-images.githubusercontent.com/22394956/154885563-e48e77f2-449e-4d09-8033-857862777d71.MOV

uldara1, 2022-02-21:
@lovyan03 my LCD has touch capacitor but i don't know chip number it support i2C interface 
how can i know address of touch ic ?
 `   cfg.i2c_port = 0; 
      cfg.i2c_addr = 0x38;   
      cfg.pin_sda  = 15;  
      cfg.pin_scl  = 14;  
      cfg.freq = 400000; `

i use this configure but touch not working

lovyan03, 2022-02-21:
oh... I am not a psychic.
How am I supposed to know what you bought?

uldara1, 2022-02-21:
i just check chip number is GSL2038
your library support this chip or not 
thank for your reply

uldara1, 2022-02-21:
i found touch ic is GSLX680

lovyan03, 2022-03-25:
@uldara1 
I have updated the develop branch and added support for GSLx680.
Probably the I2C address is 0x40.
Please give it a try.

uldara1, 2022-03-31:
Hi @lovyan03 just updated 

Now your library is work 😘🥰

https://user-images.githubusercontent.com/22394956/161089284-9f632778-c220-421b-8211-980fa15548f7.MOV

lovyan03, 2022-03-31:
@uldara1 
I am glad to hear that it worked well.
BTW, where was the firmware you uploaded?

uldara1, 2022-04-01:
@lovyan03 yeah i need to update firmware to make its work , for working firmware I attached upper 
Thank

lovyan03, 2022-04-01:
@uldara1 
Where did you find that firmware?
I know almost nothing about GSLx680.
If more than one type of firmware is needed, I would like to know why.

Where did you find that firmware?

lovyan03, 2022-04-01:
@uldara1 
and, I would also like to know what product you purchased.
Do you have a URL where I can purchase the same product?

uldara1, 2022-04-01:
@lovyan03 i use "WT-86-32-3ZW1" from wireless tag 

+ you can find from link 
[https://www.alibaba.com/product-detail/3-92-Inch-320-320-86_1700007792543.html?spm=a2747.manage.0.0.2e6271d2yYJnlo](url)
[https://blog.lvgl.io/2021-09-09/WT32-86-review](url)

+ for firmware 
 i found from this example 
[https://github.com/wireless-tag-com/8ms-esp32/](url)
[https://github.com/wireless-tag-com/8ms-esp32/blob/main/components/touch_panel/gslx680/touch_data.h](url)

lovyan03, 2022-04-01:
@uldara1 
Thanks for the information !
The develop branch has been updated.

Four types with different firmware are available.

```
lgfx::Touch_GSL1680E_800x480  // GSL_1680E, 1688E, 2681B, 2682B
lgfx::Touch_GSL1680F_800x480 
lgfx::Touch_GSL1680F_480x272 
lgfx::Touch_GSLx680_320x320  
```

I think your panel will work by using `lgfx::Touch_GSLx680_320x320` .

Also, `lgfx::Touch_GSLx680` has been changed to have no firmware and a `setFirmWare` function has been added.
Use this function if you want to use your own firmware.

uldara1, 2022-04-05:
@lovyan03 I have tested with Touch_GSLx680_320x320 and it worked well.
and i have one more question related with touch .
how to make touch work correctly without calibration 
how to save calibration data from touch

thank

lovyan03, 2022-04-05:
@uldara1 

Perhaps GSLx680 will match the coordinates without calibration.
If necessary, do the following

```
  /// get calibrate data:
  uint16_t data[8];
  gfx.calibrateTouch(data, TFT_WHITE, TFT_BLACK);

  /// check calibrate data:
  Serial.print("calibrate data:\n");
  for (int i = 0 ; i < 8; ++i)
  {
    Serial.printf("[%d] = %d\n", i, data[i]);
  }

  /// set calibrate data:
  gfx.setTouchCalibrate(data);
```

uldara1, 2022-04-05:
@lovyan03 
just update about touch when i use Touch_GSLx680_320x320
touch respond seem a bit slow than before 
sometime i need swipe or touch slowly to make it work 
thank

uldara1, 2022-04-05:
> @uldara1
>
> Perhaps GSLx680 will match the coordinates without calibration. If necessary, do the following
>
> ```
>   /// get calibrate data:
>   uint16_t data[8];
>   gfx.calibrateTouch(data, TFT_WHITE, [...]

I need to calibration touch to make it work 
if not calibation touch not work at correct point 
here is calibration data that i get from touch 

     cfg.x_min      = 2;    // タッチスクリーンから得られる最小のX値(生の値)
     cfg.x_max      = 1013;  // タッチスクリーンから得られる最大のX値(生の値)
     cfg.y_min      = 50;    // タッチスクリーンから得られる最小のY値(生の値)
     cfg.y_max      = 652;  // タッチスクリーンから得られる最大のY値(生の値)

thank

uldara1, 2022-04-05:
@lovyan03 for touch seem to be error when don't touch anything around 10s , touch not work correctly 
i don't know problem with touch or firmware. but when we reset chip it's work again

lovyan03, 2022-04-06:
@uldara1 
Hmmm. I am also in the state of not knowing anything about the details of GSLx680, so I can't give you proper advice.
My guess is that its firmware was supposed to return values up to 320.
However, your calibration data shows that it actually returns values much higher than 320.
Perhaps unless I get the same hardware and try it I will not be able to confirm the problem.

closed 2022-12-25 by github-actions[bot]