lovyan03/LovyanGFX #206: ESP32-C3 with different panel and divers

eagle-lk, 2022-03-07:
Thanks for great jobs of LovyanGFX！

I have some ESP32-C3 boards，and several panels such as: 1.8' 128x160 st7735s tft, 2.0' 176x220 ili9225 tft, 1.5' 128x128 ssd1327 oled. Fortunately，all these panels are in the supported list of LovyanGFX! So I spend some time to test the panels with LovyanGFX one by one.

As mentioned in #134 and #179，ESP32-C3 was supported but need some special config。I use the source code in 2_user_setting.ino as template，modified and saved it as **src/lgfx_user/LGFX_ESP32C3-st7735-1.8-128x160-notouch.hpp**.

==========esp32-c3+st7735=========================
#pragma once

#define LGFX_USE_V1

#include <LovyanGFX.hpp>

// ESP32C3でLovyanGFXを独自設定で利用する場合の設定例

/*
このファイルを複製し、新しい名前を付けて、環境に合わせて設定内容を変更してください。
作成したファイルをユーザープログラムからincludeすることで利用可能になります。

複製したファイルはライブラリのlgfx_userフォルダに置いて利用しても構いませんが、
その場合はライブラリのアップデート時に削除される可能性があるのでご注意ください。

安全に運用したい場合はバックアップを作成しておくか、ユーザープロジェクトのフォルダに置いてください。
//*/

/// 独自の設定を行うクラスを、LGFX_Deviceから派生して作成します。
class LGFX : public lgfx::LGFX_Device
{
/*
 クラス名は"LGFX"から別の名前に変更しても構いません。
 AUTODETECTと併用する場合は"LGFX"は使用されているため、LGFX以外の名前に変更してください。
 また、複数枚のパネルを同時使用する場合もそれぞれに異なる名前を付けてください。
 ※ クラス名を変更する場合はコンストラクタの名前も併せて同じ名前に変更が必要です。

 名前の付け方は自由に決めて構いませんが、設定が増えた場合を想定し、
 例えばESP32 DevKit-CでSPI接続のILI9341の設定を行った場合、
  LGFX_DevKitC_SPI_ILI9341
 のような名前にし、ファイル名とクラス名を一致させておくことで、利用時に迷いにくくなります。
//*/

// 接続するパネルの型にあったインスタンスを用意します。
//lgfx::Panel_GC9A01      _panel_instance;
//lgfx::Panel_GDEW0154M09 _panel_instance;
//lgfx::Panel_HX8357B     _panel_instance;
//lgfx::Panel_HX8357D     _panel_instance;
//lgfx::Panel_ILI9163     _panel_instance;
//lgfx::Panel_ILI9341     _panel_instance;
//lgfx::Panel_ILI9342     _panel_instance;
//lgfx::Panel_ILI9481     _panel_instance;
//lgfx::Panel_ILI9486     _panel_instance;
//lgfx::Panel_ILI9488     _panel_instance;
//lgfx::Panel_IT8951      _panel_instance;
//lgfx::Panel_SH110x      _panel_instance; // SH1106, SH1107
//lgfx::Panel_SSD1306     _panel_instance;
//lgfx::Panel_SSD1327     _panel_instance;
//lgfx::Panel_SSD1331     _panel_instance;
//lgfx::Panel_SSD1351     _panel_instance; // SSD1351, SSD1357
//lgfx::Panel_SSD1963     _panel_instance;
//lgfx::Panel_ST7735      _panel_instance;
lgfx::Panel_ST7735S     _panel_instance;
//lgfx::Panel_ST7789      _panel_instance;
//lgfx::Panel_ST7796      _panel_instance;

// パネルを接続するバスの種類にあったインスタンスを用意します。
  lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス
//lgfx::Bus_I2C       _bus_instance;   // I2Cバスのインスタンス (ESP32のみ)
//lgfx::Bus_Parallel8 _bus_instance;   // 8ビットパラレルバスのインスタンス (ESP32のみ)

// バックライト制御が可能な場合はインスタンスを用意します。(必要なければ削除)
  lgfx::Light_PWM     _light_instance;
[... 330 lines cut ...]
      cfg.memory_width     =   128;  // ドライバICがサポートしている最大の幅
      cfg.memory_height    =   128;  // ドライバICがサポートしている最大の高さ
      cfg.panel_width      =   128;  // 実際に表示可能な幅
      cfg.panel_height     =   128;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  true;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // データ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  true;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

//*/

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};
===================================
But unfortunately，esp32-c3+ili9225 and esp32-c3+ssd1327 all failed. I know there should be many different between the panels and drivers. But what is the key issues of my config? thanks for help.

lovyan03, 2022-03-07:
@eagle-lk 
I don't know how you failed, but perhaps the SPI clock is too fast?
I recommend that you try lowering the frequency to 5 to 10 MHz when checking the operation.

If you have further questions, please tell us the circumstances of how it failed.
Did you get a compile error?
The screen doesn't show up at all?
Did the coordinates appear out of alignment?

eagle-lk, 2022-03-07:
Thanks for your reply and sorry for information losing.

My esp32c3 and ili9225 panel are connected by Dupont Line。pins map are：
gnd - gnd
3.3 - 3.3
clk - io2
sda - io3
rst - io10
dc - io6
cs - io7
bl - io5
 
As you pointed out SPI  maybe too fast for ili9225, I reduced freq to 20M/10M, but also failed (panel is blank and backlight is on). 

**Arduino_GFX** can drive the same hareware successfully with following code in Arduino:
=========
#include <Arduino_GFX_Library.h>

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */

Arduino_DataBus *bus = new Arduino_SWSPI(6 /* DC */, 7 /* CS */, 2 /* SCK */, 3 /* MOSI */, -1 /* MISO */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
//Arduino_GFX *gfx = new Arduino_ST7735(bus, 10 /* RST */, 1 /* rotation */);
Arduino_GFX *gfx = new Arduino_ILI9225(bus, 10, 1);
==========
BTW, esp32-c3+st7735 can also be driven by call Arduino_ST7735.

So I think maybe there are some apparent mistakes in my config and ask for help.

moononournation, 2022-03-07:
FYI, Arduion_GFX limited ST7735 speed to 27 MHz.

eagle-lk, 2022-03-07:
> FYI, Arduion_GFX limited ST7735 speed to 27 MHz.

Hi，Thanks for reply!  And for your great jobs of Arduion_GFX!

eagle-lk, 2022-03-07:
@lovyan03 
Thanks!
I reduced write freq to 25M，and esp32-c3+SSD1327 run ok.
The grayscale show correctly and no need to modify demo code.
=============================================
`  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

// SPIバスの設定
      cfg.spi_host = SPI2_HOST;     // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
      cfg.spi_mode = 0;             // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write = **25000000**;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_read  = 16000000;    // 受信時のSPIクロック
      cfg.spi_3wire  = true;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock   = true;        // トランザクションロックを使用する場合はtrueを設定
      //cfg.dma_channel = SPI_DMA_CH_AUTO; // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)
      cfg.dma_channel = 0;
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 2;            // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 3;            // SPIのMOSIピン番号を設定
      cfg.pin_miso = -1;            // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc   = 6;            // SPIのD/Cピン番号を設定  (-1 = disable)
     // SDカードと共通のSPIバスを使う場合、MISOは省略せず必ず設定してください。
//*/

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs           =    7;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    10;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      cfg.memory_width     =   128;  // ドライバICがサポートしている最大の幅
      cfg.memory_height    =   128;  // ドライバICがサポートしている最大の高さ
      cfg.panel_width      =   128;  // 実際に表示可能な幅
      cfg.panel_height     =   128;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  true;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // データ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  true;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

`