lovyan03/LovyanGFX #207: Request of supporting NS2009 touch screen controller

phil6868, 2022-03-12:
Can you please add the driver for NS2009 touch screen controller, it is based on I2C bus

https://datasheet.lcsc.com/lcsc/1810181630_Shenzhen-Nsiway-Tech-NS2009_C219024.pdf

Thank you very much

lovyan03, 2022-03-14:
@phil6868 
I have arranged for a product with NS2009.
I will implement it when I have time to work on it.

phil6868, 2022-03-15:
Really appreciated. Looking forward to your driver

lovyan03, 2022-07-05:
@phil6868 
Sorry for the delay in informing you.
I have completed support for NS2009, so please give it a try.

phil6868, 2022-08-05:
Apologize for taking long time to try the code.

I was using the examples (2_user_setting) from Library to test the code.  I am able to calibrate touch screen and draw lines on the screen. However, the screen seems to be divided into two vertical planes, whenever I draw something on the left plane , it draw the same on the right plane and vice versa.

I am using ESP-32-S2, 16 bit parallel bus. It would be appreciated if you could advice how should I try the code further. Thanks in advance.

[image]
[image]
`// v1.0.0 を有効にします(v0からの移行期間の特別措置です。これを書かない場合は旧v0系で動作します。)
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

// ESP32でLovyanGFXを独自設定で利用する場合の設定例

/// 独自の設定を行うクラスを、LGFX_Deviceから派生して作成します。
class LGFX : public lgfx::LGFX_Device
{
/*
 クラス名は"LGFX"から別の名前に変更しても構いません。
 AUTODETECTと併用する場合は"LGFX"は使用されているため、LGFX以外の名前に変更してください。
 また、複数枚のパネルを同時使用する場合もそれぞれに異なる名前を付けてください。
 ※ クラス名を変更する場合はコンストラクタの名前も併せて同じ名前に変更が必要です。

 名前の付け方は自由に決めて構いませんが、設定が増えた場合を想定し、
 例えばESP32 DevKit-CでSPI接続のILI9341の設定を行った場合、
  LGFX_DevKitC_SPI_ILI9341
 のような名前にし、ファイル名とクラス名を一致させておくことで、利用時に迷いにくくなります。
//*/

// 接続するパネルの型にあったインスタンスを用意します。
lgfx::Panel_ILI9481     _panel_instance;

// パネルを接続するバスの種類にあったインスタンスを用意します。
lgfx::Bus_Parallel16 _bus_instance; // 8ビットパラレルバスのインスタンス (ESP32のみ)

// バックライト制御が可能な場合はインスタンスを用意します。(必要なければ削除)
lgfx::Light_PWM     _light_instance;

// タッチスクリーンの型にあったインスタンスを用意します。(必要なければ削除)
lgfx::Touch_NS2009          _touch_instance;

public:

  // コンストラクタを作成し、ここで各種設定を行います。
  // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

 // 16位设置
      cfg.i2s_port = I2S_NUM_0;  // 使用するI2Sポートを選択 (0 or 1) (ESP32のI2S LCDモードを使用します)
      cfg.freq_write = 12000000; // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
      cfg.pin_wr = 4;           // WR を接続しているピン番号
//      cfg.pin_rd = 34;           // RD を接続しているピン番号
      cfg.pin_rs = 3;           // RS(D/C)を接続しているピン番号

      cfg.pin_d0 = 20;
      cfg.pin_d1 = 19;
      cfg.pin_d2 = 18;
      cfg.pin_d3 = 17;
      cfg.pin_d4 = 16;
      cfg.pin_d5 = 15;
      cfg.pin_d6 = 14;
      cfg.pin_d7 = 13;
      cfg.pin_d8 = 12;
      cfg.pin_d9 = 11;
      cfg.pin_d10 = 10;
      cfg.pin_d11 = 9;
[... 109 lines cut ...]
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30,30,display.width()-60,display.height()-60,count*7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  int32_t x, y;
  if (display.getTouch(&x, &y)) {
    display.fillRect(x-2, y-2, 5, 5, count*7);
  }
}`

lovyan03, 2022-08-05:
https://github.com/lovyan03/LovyanGFX/issues/231#issuecomment-1116850484

This is a specification issue with ILI9481, try it without setRotation in loop.

phil6868, 2022-08-05:
It works. I will test the code further and provide feedback if there is any issue. Many thanks again.

closed 2022-12-25 by github-actions[bot]