lovyan03/LovyanGFX #208: st7789 16bit parallel

colbytarr, 2022-03-16:
Hi,

I am using an esp32-s2 and 8bit parallel to communicate with a st7789. 

8bit works but I would like to use 16bit parallel. Is this supported yet?

Also, when using 8bit all colors are a shade a blue unless I set ```lcd.setColorDepth(24);```

Is there anything I am missing?

```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ST7789 _panel_instance;
    lgfx::Bus_Parallel16 _bus_instance; // 8ビットパラレルバスのインスタンス (ESP32のみ)
    lgfx::Light_PWM _light_instance;

public:
    LGFX(void)
    {
        {                                      // バス制御の設定を行います。
            auto cfg = _bus_instance.config(); // バス設定用の構造体を取得します。

            // 16位设置
            cfg.i2s_port = I2S_NUM_0;  // 使用するI2Sポートを選択 (0 or 1) (ESP32のI2S LCDモードを使用します)
            cfg.freq_write = 16000000; // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
            cfg.pin_wr = 45;           // WR を接続しているピン番号
            cfg.pin_rd = 17;           // RD を接続しているピン番号
            cfg.pin_rs = 18;           // RS(D/C)を接続しているピン番号

            cfg.pin_d0 = 16;
            cfg.pin_d1 = 15;
            cfg.pin_d2 = 14;
            cfg.pin_d3 = 13;
            cfg.pin_d4 = 12;
            cfg.pin_d5 = 11;
            cfg.pin_d6 = 10;
            cfg.pin_d7 = 9;
             cfg.pin_d8 = 38;
             cfg.pin_d9 = 37;
             cfg.pin_d10 = 36;
             cfg.pin_d11 = 35;
             cfg.pin_d12 = 34;
             cfg.pin_d13 = 33;
             cfg.pin_d14 = 26;
             cfg.pin_d15 = 21;

            _bus_instance.config(cfg);              // 設定値をバスに反映します。
            _panel_instance.setBus(&_bus_instance); // バスをパネルにセットします。
        }

        {                                        // 表示パネル制御の設定を行います。
            auto cfg = _panel_instance.config(); // 表示パネル設定用の構造体を取得します。

            cfg.pin_cs = -1;   // CS要拉低
            cfg.pin_rst = 8;  // RST和开发板RST相连
            cfg.pin_busy = -1; // BUSYが接続されているピン番号 (-1 = disable)

            // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

            cfg.memory_width = 240;   // ドライバICがサポートしている最大の幅
            cfg.memory_height = 320;  // ドライバICがサポートしている最大の高さ
            cfg.panel_width = 240;    // 実際に表示可能な幅
            cfg.panel_height = 320;   // 実際に表示可能な高さ
            cfg.offset_x = 0;         // パネルのX方向オフセット量
            cfg.offset_y = 0;         // パネルのY方向オフセット量
            //cfg.offset_rotation = 0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
            //cfg.dummy_read_pixel = 0; // ピクセル読出し前のダミーリードのビット数
            //cfg.dummy_read_bits = 1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
            // cfg.readable = true;      // データ読出しが可能な場合 trueに設定
            // cfg.invert = false;       // パネルの明暗が反転してしまう場合 trueに設定
            // cfg.rgb_order = false;    // パネルの赤と青が入れ替わってしまう場合 trueに設定
            // cfg.dlen_16bit = false;    // データ長を16bit単位で送信するパネルの場合 trueに設定
            // cfg.bus_shared = true;    // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

            _panel_instance.config(cfg);
        }
        { // バックライト制御の設定を行います。（必要なければ削除）
      auto cfg = _light_instance.config();    // バックライト設定用の構造体を取得します。

      cfg.pin_bl = 3;              // バックライトが接続されているピン番号
      cfg.invert = false;           // バックライトの輝度を反転させる場合 true
      cfg.freq   = 44100;           // バックライトのPWM周波数
      cfg.pwm_channel = 7;          // 使用するPWMのチャンネル番号

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);  // バックライトをパネルにセットします。
    }
        setPanel(&_panel_instance); // 使用するパネルをセットします。
    }
};
static LGFX lcd;

void setup(void)
{

    Serial.begin(115200);

    lcd.init();
    lcd.setRotation(3);
    lcd.setBrightness(255);
    lcd.setColorDepth(24);

   
}

static int colors[] = {TFT_RED, TFT_GREEN, TFT_BLUE, TFT_CYAN, TFT_MAGENTA, TFT_YELLOW};

int i = 0;

void loop(void)
{
  //fps
  static int prev_sec;
  static int fps;
  ++fps;
  int sec = millis() / 1000;
  if (prev_sec != sec)
  {
    prev_sec = sec;
    lcd.setCursor(0, 0);
    lcd.setTextSize(2);
    lcd.printf("fps:%03d", fps);
    fps = 0;
  }

  lcd.fillRect(0, 30, 320, 240, colors[i++]);
  //lcd.fillScreen(colors[i++]);
  if (i > 5)
    i = 0;
}
```

lovyan03, 2022-03-16:
@colbytarr 

```
 cfg.dlen_16bit = false;    // データ長を16bit単位で送信するパネルの場合 trueに設定
```
please set this to true .

colbytarr, 2022-03-17:
@lovyan03 

Still no luck.

From the serial monitor, it appears that the ESP is constantly resetting

lovyan03, 2022-03-18:
@colbytarr 
I tried your code without the LCD connected as I can't spare the time to do the wiring work right now.
I used Serial.printf instead of lcd.printf, with arduinoESP32 v2.0.2.
It worked without rebooting and output logs successfully.

Just to confirm, does it work with Bus_Serial8 and reboot with Bus_Serial16?

Please set the log output level to verbose, and try again with the library from the develop branch.
If the ESP32S2 reboots again, please post the log of one reboot here.

lovyan03, 2022-03-18:
Incidentally, there was a report the other day that 8-bit parallel was not working properly on the ESP32S2, which has been corrected.

#202

He wrote that 16-bit parallel was working fine.

colbytarr, 2022-03-18:
@lovyan03

Turns out my issue was ```cfg.pin_d14 = 26;```

From https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/gpio.html

`SPI0/1: GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses`

Odd since I have had this same board/pinout working with an esp-idf example.

Both 8bit and 16bit are working properly.

Your library is wonderful!

Thank you for your time!

summerfind, 2023-09-14:
Hi all.
I am still getting compiling error with above code: 
'Bus_Parallel16' in namespace 'lgfx' does not name a type

I used ESP32+ILI9341. 
Thanks
Adam

summerfind, 2023-09-14:
Hi all.
seems the code above used to ESP32 S2, can it be used to ESP32 Dev Module? or are there example for ESP32?
Thanks
Adam