lovyan03/LovyanGFX #210: touch panel :FT6236 doesn't work

JohnTaipei, 2022-03-18:
Hello Sir:
I use FT6236 + ILI9488 8-bit
Why FT6236 doesn't work??

 { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 319;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 479;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = 35;   // INTが接続されているピン番号
      cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// I2C接続の場合
      cfg.i2c_port = 1;      // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
      cfg.pin_sda  = 33;     // SDAが接続されているピン番号
      cfg.pin_scl  = 32;     // SCLが接続されているピン番号
      cfg.freq = 400000;     // I2Cクロックを設定

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }

lovyan03, 2022-03-19:
@JohnTaipei 
There is too little information.
What microcontroller are you using?
What is the firmware version?
Please provide the smallest buildable code that can reproduce the Issue.

JohnTaipei, 2022-03-19:
Hello Sir:
I am using ESP32 + ILI9488 + FT6236
version is V0.4.3

I am trying latest version V0.4.14

JohnTaipei, 2022-03-19:
Thank you Sir :
I update V0.4.14
FT6236 works
https://www.facebook.com/100044063329321/videos/1349824018857611/

closed 2022-03-20 by lovyan03