lovyan03/LovyanGFX #212: Could you support 16-bit PARALLEL for ESP32??

JohnTaipei, 2022-03-23:
Hello Sir:
Could you support 16-bit PARALLEL for ESP32??

Use 18 GPIO pins
wr,rs,d0~d15
cs=-1 (connect to GND)
rd=-1 (connect to 3V3)
rst=-1 (connect to EN pin)

Thank you

lovyan03, 2022-03-23:
@JohnTaipei 
The behavior of the ESP32-S2's I2S peripheral is straightforward, so it was fast enough even in 16-bit parallel.
On the other hand, the ESP32's I2S peripheral behaves strangely, so we had to implement it in a rather anomalous way.
Even if we could implement 16-bit parallelism in ESP32, I expect it would not be faster than 8-bit parallelism.

JohnTaipei, 2022-03-24:
OK,I see
I will desgin a ESP32-S2 + 16-bit parallel LCD
Thank for your answer

closed 2022-05-14 by lovyan03