lovyan03/LovyanGFX #214: esp32-s3

gyjmy111, 2022-03-26:
Hi could you please add the support for ESP32-s3?

lovyan03, 2022-03-26:
@gyjmy111 
I have already started to support ESP32-S3 and SPI and I2C are currently available in the develop branch.
Please use the develop branch if you want to try it immediately.

tobozo, 2022-03-29:
New release of [arduino-esp32](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) has ESP32-S3 support with a board profile for ESP32-S3-Box :partying_face: .

I've started testing the `1_simple_use.ino` example sketch with the most recent *develop* branch.

The compilation worked, but the detection failed:

```
Hello LGFX example
[ 10119][W][LGFX_AutoDetect_ESP32S3.hpp:115] init_impl(): [Autodetect] load from NVS : board:0
[ 10132][W][LGFX_AutoDetect_ESP32S3.hpp:83] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
[ 10145][W][LGFX_AutoDetect_ESP32S3.hpp:83] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
[ 10158][W][LGFX_AutoDetect_ESP32S3.hpp:83] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
[ 10171][W][LGFX_AutoDetect_ESP32S3.hpp:83] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
[ 10184][W][LGFX_AutoDetect_ESP32S3.hpp:83] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
[ 10185][W][LGFX_AutoDetect_ESP32S3.hpp:140] init_impl(): [Autodetect] save to NVS : board:0
```

lovyan03, 2022-03-30:
@tobozo 
Thanks for the report !
I too can confirm that the auto-detection is not working properly.
I have fixed the issue and updated the develop branch, please give it a try.

tobozo, 2022-03-30:
@lovyan03 thanks for this patch, the panel is now correctly detected and the `1_simple_use.ino` is running smoothly.

Some observations I've made when testing random sketches:

- `lcd.drawPng( data, data_len, x, y) `  triggers a reset:  `rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)`
- `WiFi.scanNetworks (true, true, true, 150 );` also triggers a reset when `lcd.init()` is used.

Here's the png data I'm using when the reset occurs :

[image]

```C
// 50x50 1bit palette PNG
const unsigned int data_len = 261;
const unsigned char data[data_len] =
{
  0x89, 0x50, 0x4e, 0x47, 0x0d, 0x0a, 0x1a, 0x0a, 0x00, 0x00, 0x00, 0x0d,
  0x49, 0x48, 0x44, 0x52, 0x00, 0x00, 0x00, 0x32, 0x00, 0x00, 0x00, 0x32,
  0x01, 0x03, 0x00, 0x00, 0x00, 0x24, 0xf1, 0x1a, 0xf2, 0x00, 0x00, 0x00,
  0x06, 0x50, 0x4c, 0x54, 0x45, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xa5,
  0x67, 0xb9, 0xcf, 0x00, 0x00, 0x00, 0x02, 0x74, 0x52, 0x4e, 0x53, 0xff,
  0x00, 0xe5, 0xb7, 0x30, 0x4a, 0x00, 0x00, 0x00, 0x09, 0x70, 0x48, 0x59,
  0x73, 0x00, 0x00, 0x0e, 0xc4, 0x00, 0x00, 0x0e, 0xc4, 0x01, 0x95, 0x2b,
  0x0e, 0x1b, 0x00, 0x00, 0x00, 0x97, 0x49, 0x44, 0x41, 0x54, 0x18, 0x95,
  0x55, 0xd0, 0x41, 0x0a, 0x05, 0x21, 0x08, 0x06, 0x60, 0xc1, 0xed, 0x80,
  0x57, 0x09, 0xdc, 0x06, 0x5e, 0x5d, 0x68, 0x1b, 0x74, 0x95, 0xc0, 0x6d,
  0xe0, 0xd4, 0xab, 0xa9, 0x9e, 0x9b, 0x0f, 0x04, 0xf5, 0x47, 0x00, 0x94,
  0x14, 0xa0, 0xd7, 0x52, 0x0c, 0x98, 0xfd, 0x18, 0xb8, 0x91, 0xe1, 0x2d,
  0xe3, 0xbf, 0x66, 0x7c, 0x29, 0x66, 0x34, 0xe7, 0xa7, 0x80, 0xcc, 0x73,
  0xff, 0xd4, 0x9d, 0x9a, 0xbb, 0x6e, 0x83, 0xf6, 0x43, 0x19, 0xb6, 0x20,
  0xf2, 0xc4, 0xa6, 0x5b, 0xab, 0xb9, 0x80, 0x1f, 0xe5, 0xd1, 0x5c, 0xd2,
  0x31, 0xa2, 0xfa, 0x08, 0xfe, 0x59, 0x4b, 0xd3, 0x2a, 0xc7, 0x82, 0xc4,
  0x25, 0xe8, 0x36, 0xb8, 0x51, 0xe8, 0xfd, 0x4f, 0xc3, 0xda, 0x46, 0xbc,
  0x4f, 0x77, 0x21, 0xe1, 0x23, 0x20, 0xc8, 0x2f, 0xf8, 0x52, 0x8c, 0xa3,
  0x44, 0xdd, 0x06, 0xce, 0x04, 0xcf, 0x2d, 0xa1, 0x01, 0x5c, 0x46, 0xc2,
  0x71, 0x77, 0x29, 0x96, 0x3c, 0xf1, 0x71, 0xfc, 0xa5, 0xad, 0xff, 0x74,
  0x5f, 0xd6, 0x73, 0x88, 0x3f, 0x85, 0x22, 0xee, 0x13, 0x00, 0x00, 0x00,
  0x00, 0x49, 0x45, 0x4e, 0x44, 0xae, 0x42, 0x60, 0x82
};

```

Here's a minimal snippet to reproduce the WiFi problem, just comment/uncomment `lcd.init()` in the `setup()` to see the differences.

```C++
#include <WiFi.h>

#define LGFX_AUTODETECT
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

static LGFX lcd;
uint32_t scanChanDelay = 150; // default
const uint32_t channels_count = 13;

void setup()
{
  Serial.begin(115200);
  // lcd.init();
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  Serial.println("Setup done");
}

void loop()
{
  int n = WiFi.scanComplete();
  if (n == -2) {
    WiFi.scanNetworks (true, true, true, scanChanDelay );  // async, show_hidden
  } else if (n == -1) {
    // still scanning
  } else if (n > 0) {
    Serial.printf("WiFi scan found %d devices\n", n);
    WiFi.scanDelete();
  }
}
```

lovyan03, 2022-03-30:
@tobozo 
Thanks for reporting it.
Very headache, the problem with the PNGs stopping drawing does not occur on the ESP-IDF, only on the ArduinoESP32.
I would like to investigate the cause now, but it is very painful because I have to debug with ArduinoIDE.

tobozo, 2022-03-30:
The debugging seems disabled in Arduino IDE anyway, so it may be better to wait for RC2.

lovyan03, 2022-03-31:
@tobozo 
I have now updated the develop branch.
Perhaps this fix will make it work correctly.

tobozo, 2022-03-31:
@lovyan03 thanks I can confirm the last update fixed drawPng along with the WiFi bootloop.
I can now run the WiFiChanViz demo without a glitch :partying_face: and will proceed with testing other sketches.

lovyan03, 2022-03-31:
@tobozo 
Thanks for reporting this bug. I have now fixed the develop branch.
It seems that ESP32-S3 and C3 can send up to 32768 bytes of data at a time.
Data longer than this would need to be sent to the SPI peripheral multiple times.

This fix should work correctly, but it requires some ingenuity to achieve the highest frame rate.
It is important to adjust the height of the sprite so that a single transfer fits into 32768 bytes.
In other words, for a 16-bit color with a width of 320 pixels, the height should be no more than 51 pixels.
Since the ESP32-S3-BOX is 240 pixels high, it should be 48 pixels divided by 5.

tobozo, 2022-04-01:
thanks for the fix and detailed explanation :+1:  

I will now proceed with testing the touch support.

gyjmy111, 2022-04-07:
Thanks for your help.Hah.

closed 2022-04-27 by lovyan03