lovyan03/LovyanGFX #216: rp2040 and Touch XPT2046

rcla, 2022-03-29:
I already have my 3.5" screen (SPI) working with the ESP32 chip.
Now I want to test it with rp2040 chip(Raspberry Pi pico).

I am using the example:
`2_user_setting.ino`

And I modify the parameters of my screen.

I set up:
`lgfx::Touch_XPT2046     _touch_instance;`

When compiling it shows the following error:
```
libraries\LovyanGFX\lgfx\v1\touch\Touch_XPT2046.cpp.o: In function `lgfx::v1::Touch_XPT2046::getTouchRaw(lgfx::v1::touch_point_t*, unsigned int)':
C:\Users\rcla\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\touch/Touch_XPT2046.cpp:75: undefined reference to `lgfx::v1::spi::readBytes(int, unsigned char*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
```

lovyan03, 2022-03-29:
@rcla 
Sorry...
Currently RP2040 support is for screen display only,
 touch panel support is not yet implemented.

rcla, 2022-03-29:
Oh ok.
I'll be watching when you can add touch support.

Thanks for your quick response.

twshield, 2023-01-21:
It would have been nicer and saved me some time if you had added:

```
#if defined (ARDUINO_ARCH_MBED_RP2040) || defined(ARDUINO_ARCH_RP2040)
#error rp2040 touch support is not yet completed
#endif
```

To Touch.hpp

tobozo, 2023-01-21:
> It would have been nicer and saved me some time 

@twshield please be respectful, losing patience while coding happens to all of us, but the only person responsible for the time you lose is you.

If you believe the suggested code change will help other rp2040 users, feel free to send a PR against the 'develop' branch.

iSP-Iwasaki, 2025-07-31:
im use custmaiz configretion 

#include "LGFX_RP2040_ILI9341_touch.hpp"

#if(USE_XPT_LGFX == 1) 
    { // Configure touchscreen control. タッチスクリーン制御の設定を行います。（必要なければ削除）

      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = (1023 * 3);  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = (1023 * 3);  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = TOUCH_INT;   // INTが接続されているピン番号
      cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// SPI接続の場合
      cfg.spi_host = 1;		// 使用するSPIを選択 (SPI0 or IT)
      cfg.freq = 10000000;	// SPIクロックを設定 10MHz
      cfg.pin_sclk = TFT_SCLK;  // SCLKが接続されているピン番号
      cfg.pin_mosi = TFT_MOSI;  // MOSIが接続されているピン番号
      cfg.pin_miso = TFT_MISO;	// MISOが接続されているピン番号
      cfg.pin_cs   = TOUCH_CS ;	//   CSが接続されているピン番号

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }
#endif    

I hope this transcription helps
この転記でお役にたてれば幸いです

closed 2025-09-06 by github-actions[bot]

GKich, 2025-11-21:
> I already have my 3.5" screen (SPI) working with the ESP32 chip. Now I want to test it with rp2040 chip(Raspberry Pi pico).
>
> I am using the example: `2_user_setting.ino`
>
> And I modify the parameters [...]

Could you please share the configuration for the TouchXPT2046 ? I need it. Thank you!