lovyan03/LovyanGFX #218: RP2040 and LVGL

rcla, 2022-04-07:
I have the LVGL v8 example "lv_demo_widgets()" running on an ESP32.

Now I want to test it on a RP2040.

First, I adapted the parameters of the library, to make it compatible with the RP2040.
I made a simple program, where I write some text, change the colors and check that it works.

In the code I am not using the touch part of the screen.

In ESP32 it is displayed correctly:

But with the RP2040, it gets to this part and freezes, not responding:

yasuhirok-git, 2022-04-09:
I am trying to solve this issue. I need more information.
My questions are:
- Are you using Arduino IDE or Platform IO?
- If you are using Arduino IDE, which board was selected?
 (Example: "Tool" menu->"Board"->"Raspberry Pi RP2040 Boards(1.13.1)->"Generic RP2040")
I'm waiting for your reply.

rcla, 2022-04-09:
@yasuhirok-git
Thank you for your interest and work.

I use Arduino IDE.

I have tried with two options:

* "Board"->"Arduino Mbed OS RP2040 Boards (version 3.01)->"Raspberry Pi Pico" 

* "Board"->"Raspberry Pi RP2040 Boards(1.13.1)->"Raspberry Pi Pico"
   With these parameters:
[image]

With either of these two options, you see the same behavior.

yasuhirok-git, 2022-04-09:
@rcla san, thanks for your information.
I can not reproduce the problem you found.
In my experience, programs on the rp2040 often freeze up because of infinite loops in the exception handler due to alignment violations. After freezing, can you re-write without going into bootloader mode?

rcla, 2022-04-09:
After freezing, I can't re-write.

In addition, I have noticed that when freezing, the green led (BUILTIN) stays in an infinite loop blinking:
* Three or four times every second and then
* Three or four times faster.

yasuhirok-git, 2022-04-10:
@rcla 

> After freezing, I can't re-write.

You provided me with good information.
This indicates that your program has encountered a hard fault.
I have created a function that displays information when a hard fault occurs.
Add this file to your project, build and run it.
This function displays the error message, error code, PC and LR contents on serial monitor.
Please provide me with its output and your program(ELF format).
I will be analyzing what is happening from those information.
[mbed_error.zip](https://github.com/lovyan03/LovyanGFX/files/8458004/mbed_error.zip)

Note that this function works on Mbed environment. Please select "Arduino 
[mbed_error.zip](https://github.com/lovyan03/LovyanGFX/files/8457999/mbed_error.zip)
Mbed OS RP2040 Boards".
[mbed_error.zip](https://github.com/lovyan03/LovyanGFX/files/8458020/mbed_error.zip)
[mbed_error.zip](https://github.com/lovyan03/LovyanGFX/files/8458021/mbed_error.zip)

### About LED blinking:
Mbed-os has the ability to notify using LEDs when an error occurs, Please refer the folloeing URL.
https://os.mbed.com/handbook/Debugging#runtime-bugs

rcla, 2022-04-10:
I added your mbed_error file. And Mbed environment.

In Serial Monitor it shows:
`Fault exception 80ff013d 1003628c 100304bf`

Attached Program in ELF Format:
[Program_ELF.zip](https://github.com/lovyan03/LovyanGFX/files/8458087/Program_ELF.zip)

yasuhirok-git, 2022-04-10:
@rcla san,
From the information you provided and the ELF file, I was able to identify where the fault is occurring.
```
10036214 <_free_r>:
10036214:	b5f8      	push	{r3, r4, r5, r6, r7, lr}
10036216:	46ce      	mov	lr, r9
10036218:	4647      	mov	r7, r8
1003621a:	0005      	movs	r5, r0
1003621c:	b580      	push	{r7, lr}
........
10036284:	42a3      	cmp	r3, r4
10036286:	d100      	bne.n	1003628a <_free_r+0x76>
10036288:	e07f      	b.n	1003638a <_free_r+0x176>
1003628a:	68d2      	ldr	r2, [r2, #12]
1003628c:	60da      	str	r2, [r3, #12]	<- fault occurred here
1003628e:	6093      	str	r3, [r2, #8]
10036290:	4643      	mov	r3, r8
```
However, the error is occurring in the function _free_r, not in the function within LovyaGFX.
Please let me think for a few days about how to resolve this issue.

rcla, 2022-04-10:
> However, the error is occurring in the function _free_r, not in the function within LovyaGFX. Please let me think for a few days about how to resolve this issue.

@[yasuhirok-git](https://github.com/yasuhirok-git)

Of course.

If you need more information, let me know.

Thanks for your help.

rcla, 2022-04-11:
@yasuhirok-git 

After several tests, I managed to determine that "Fault exception" occurs when calling:
`lcd.writePixels((lgfx::rgb565_t *)&color_p->full, w * h);`

And reviewing the source code Bus_SPI.cpp I find that the "writePixels" procedure also calls "writeBytes" and in both
depend on the DMA channel number.
`if (_dma_ch >= 0) ...`

I found that it is defined in the "init":
`int dma_ch = dma_claim_unused_channel(true);`
`_dma_ch = dma_ch;`

So I tried changing it to:
`int dma_ch = -1; //dma_claim_unused_channel(true);`

And now it works! It shows no errors and everything is displayed correctly.

Obviously this is not the solution, but it can help you find the root of the problem.

yasuhirok-git, 2022-04-16:
@rcla,
I confirmed that the issue reproduced itself in the newly created environment.
And I found the problem and fixed it.
Could you try the following fixes to see if they solve your problem?

file: LovyanGFX/src/lgfx/v1/platforms/rp2040/Bus_SPI.cpp
original
```
230       do
231       {
232         len = (limit <= length) ? limit : length;
233         if (limit <= 256) limit <<= 1;
234
235         auto dmabuf = _flip_buffer.getBuffer(limit * dst_bytes);
```
fixed code
```
230       do
231       {
232         if (limit <= 256) limit <<= 1;   // changed
233         len = (limit <= length) ? limit : length;  // changed
234
235         auto dmabuf = _flip_buffer.getBuffer(limit * dst_bytes);
```

rcla, 2022-04-16:
> @rcla, I confirmed that the issue reproduced itself in the newly created environment. And I found the problem and fixed it. Could you try the following fixes to see if they solve your problem?
> 
> file: LovyanGFX/src/lgfx/v1/platforms/rp2040/Bus_SPI.cpp original
> 
> ```

@yasuhirok-git 
I confirm that with this change, it is the solution!

Thank you very much for your help.

Before closing this issue, do you plan to add touch support?
My screen has the XPT2046 controller, if you need tests, I can gladly do them.

closed 2022-04-26 by rcla