lovyan03/LovyanGFX #219: Touch not working, where am I going wrong?

srduino, 2022-04-07:
[image]

[env:featheresp32]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
lib_deps = 
	plerup/EspSoftwareSerial@^6.15.2
	kisvegabor/lvgl@^8.2.0
	sstaub/Ticker@^4.4.0
	256dpi/MQTT@^2.5.0
	bblanchon/ArduinoJson@^6.19.3
	lovyan03/LovyanGFX@^0.4.15

```

#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

  lgfx::Panel_ST7789 _panel_instance;
  lgfx::Bus_SPI _bus_instance;
  lgfx::Light_PWM _light_instance;
  lgfx::Touch_FT5x06 _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = VSPI_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 40000000;
      cfg.freq_read = 16000000;
      cfg.spi_3wire = false;
      cfg.use_lock = false;
      cfg.dma_channel = 1;
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 19;
      cfg.pin_miso = -1;
      cfg.pin_dc = 16;
      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();
      cfg.pin_cs = 5;
      cfg.pin_rst = 23;
      cfg.pin_busy = -1;
      cfg.panel_width = 240;
      cfg.panel_height = 320;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      cfg.offset_rotation = 3;
      cfg.dummy_read_pixel = 8;
      cfg.dummy_read_bits = 1;
      cfg.readable = false;
      cfg.invert = false;
      cfg.rgb_order = false;
      cfg.dlen_16bit = false;
      cfg.bus_shared = false;
      _panel_instance.config(cfg);
    }

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = 4;
      cfg.invert = false;
      cfg.freq = 1023;
      cfg.pwm_channel = 8;
      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);
    }

    {
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = 239;
      cfg.y_min = 0;
      cfg.y_max = 319;
      cfg.pin_int = 2;
      cfg.bus_shared = false;
      cfg.offset_rotation = 0;

      // cfg.spi_host = VSPI_HOST;
      // cfg.freq = 1000000;
      // cfg.pin_sclk = 22;
      // cfg.pin_mosi = 21;
      // cfg.pin_miso = -1;
      // cfg.pin_cs = 5;

      cfg.i2c_port = 1;
      cfg.freq = 400000;
      cfg.i2c_addr = 0x38;
      cfg.pin_sda = 21;
      cfg.pin_scl = 22;

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }
    setPanel(&_panel_instance);
  }
};

LGFX display;

void setup(void)
{
  display.init();
  display.setBrightness(255);
  display.setColorDepth(16);

  if (display.width() < display.height())
  {
    display.setRotation(display.getRotation() ^ 1);
    display.setCursor(0, 0);
  }
}

uint32_t count = ~0;
void loop(void)
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30, 30, display.width() - 60, display.height() - 60, count * 7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  int32_t x, y;
  if (display.getTouch(&x, &y))
  {
    display.fillRect(x - 2, y - 2, 5, 5, count * 7);
  }
}
```

lovyan03, 2022-04-07:
@srduino 
I don't see any particular problem, but I don't know if the hardware information and physical connections are correct.
Why don't you implement an I2C scan and see if you can communicate?

srduino, 2022-04-08:
@lovyan03 
thank you very much for your reply!

at present, this driver can work normally：
https://github.com/aselectroworks/Arduino-FT6336U/

combined use:
[image]

looking forward to integration and revision！

lovyan03, 2022-04-08:
@srduino 
I see, I think the RST pin was not controlled.
Currently, LovyanGFX does not implement control of the RST pin of the touch panel, but I think you can handle this for now by controlling the RST yourself.

Specifically, you should be able to set GPIO17 LOW for 1msec in your program before LGFX.init(), and then set GPIO17 HIGH.

I will eventually add RST pin settings to the touch control. thank you !

srduino, 2022-04-08:
@lovyan03

changes I tried:

```

#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

  lgfx::Panel_ST7789 _panel_instance;
  lgfx::Bus_SPI _bus_instance;
  lgfx::Light_PWM _light_instance;
  lgfx::Touch_FT5x06 _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = VSPI_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 40000000;
      cfg.freq_read = 16000000;
      cfg.spi_3wire = false;
      cfg.use_lock = false;
      cfg.dma_channel = 1;
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 19;
      cfg.pin_miso = -1;
      cfg.pin_dc = 16;
      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();
      cfg.pin_cs = 5;
      cfg.pin_rst = 23;
      cfg.pin_busy = -1;
      cfg.panel_width = 240;
      cfg.panel_height = 320;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      cfg.offset_rotation = 3;
      cfg.dummy_read_pixel = 8;
      cfg.dummy_read_bits = 1;
      cfg.readable = false;
      cfg.invert = false;
      cfg.rgb_order = false;
      cfg.dlen_16bit = false;
      cfg.bus_shared = false;
      _panel_instance.config(cfg);
    }

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = 4;
      cfg.invert = false;
      cfg.freq = 1023;
      cfg.pwm_channel = 8;
      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);
    }

    {
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = 239;
      cfg.y_min = 0;
      cfg.y_max = 319;
      cfg.pin_int = 2;
      cfg.bus_shared = false;
      cfg.offset_rotation = 0;

      // cfg.spi_host = VSPI_HOST;
      // cfg.freq = 1000000;
      // cfg.pin_sclk = 22;
      // cfg.pin_mosi = 21;
      // cfg.pin_miso = -1;
      // cfg.pin_cs = 5;

      cfg.i2c_port = 1;
      cfg.freq = 400000;
      cfg.i2c_addr = 0x38;
      cfg.pin_sda = 21;
      cfg.pin_scl = 22;
      // 2022-04-08
      cfg.pin_rst = 17;

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }
    setPanel(&_panel_instance);
  }
};

LGFX display;

#define RST_N_PIN 17

void setup(void)
{

  // // 2022-04-08 Lovyan03:
  // pinMode(RST_N_PIN, OUTPUT);
  // digitalWrite(RST_N_PIN, LOW);
  // delay(10);
  // digitalWrite(RST_N_PIN, HIGH);
  // delay(500);

  display.init();
  display.setBrightness(255);
  display.setColorDepth(16);

  if (display.width() < display.height())
  {
    display.setRotation(display.getRotation() ^ 1);
    display.setCursor(0, 0);
  }
}

uint32_t count = ~0;
void loop(void)
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30, 30, display.width() - 60, display.height() - 60, count * 7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  int32_t x, y;
  if (display.getTouch(&x, &y))
  {
    display.fillRect(x - 2, y - 2, 5, 5, count * 7);
    Serial.println("???????????????????");
  }
}
```

[image]

[image]

the problem is still there!

lovyan03, 2022-04-08:
@srduino 
I'm sorry if I'm wrong.

Are you sure you are not confusing SDA with SCL?
In the source code you have presented so far, the SDA and SCL numbers do not appear to be consistent.

[image]
[image]

closed 2022-12-25 by github-actions[bot]