lovyan03/LovyanGFX #223: error: 'class TwoWire' has no member named 'end';

juanitomaille, 2022-04-08:
hello, 

see my compilation error below :

```
.pio/libdeps/esp32Makerfabs/LovyanGFX/src/lgfx/v1/platforms/esp32/common.cpp:741:16: error: 'class TwoWire' has no member named 'end'; did you mean 'read'?
       twowire->end();

```
solved that by changing by that :

`twowire->endTransmission(true);`

my platformio.ini : 

```
[env:esp32Makerfabs]
platform = espressif32
board = esp32dev
framework = arduino
board_build.mcu = esp32s2
upload_protocol = esptool
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/juanitomaille/arduino-esp32#2.0.0-alpha1
	platformio/tool-esptoolpy @ ~1.30100
src_filter = +<*> -<.git/> -<.svn/> -<example/> -<examples/> -<test/> -<tests/> -<main_sensor.cpp*> -<main_ihm.cpp*> -<main_ihm_ili9486.cpp*>
build_flags = -DLV_CONF_INCLUDE_SIMPLE -DLV_DEMO_CONF_INCLUDE_SIMPLE -Iinclude -Iinclude/logos
lib_deps = 
	lovyan03/LovyanGFX@^0.4.15
	lvgl/lvgl@^8.1.0
	https://github.com/juanitomaille/InfluxDB-Client-for-Arduino.git
```
hope this helps.

tobozo, 2022-04-08:
Hello, thanks for reporting this.

This is a known bug with platformio and `arduino-2.0.0`, looks like it also happen with the alpha version.

If you feel like editing common.cpp file, ou can eventually apply this patch manually, or just comment out the `twowire->end();`:

https://github.com/lovyan03/LovyanGFX/pull/222/commits/55ebdd989971f4798426dd72a4ff8d0b8d36010a

This fix will be included in a later release.

You can also bypass the bug by using a more recent version or arduino-esp32.

If you want to experiment with other arduino-esp32 versions, this platform package from tasmota works with 2.0.1, 2.0.2 and 2.0.3-RC1, : 

```ini
platform = https://github.com/tasmota/platform-espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3-RC1
```

juanitomaille, 2022-04-16:
Thanks for your reply.

Now it works, I will look after the future version.

closed 2022-04-27 by lovyan03