lovyan03/LovyanGFX #228: Custom Fonts

teeshq, 2022-04-26:
Hi how to use custom fonts where i can paste them ect

teeshq, 2022-04-26:
nvm sorted

closed 2022-04-26 by teeshq

martinberlin, 2022-12-01:
I have the feeling that this part is missing some documentation @lovyan03 

I got stuck for example in the latest days trying to render text in Polish, which seems that Adafruit GFX **fontconvert** tool does not support. As an example:

`` GADZIE GODY   czyli Sagi o czaroksiążcy prapoczątek, opowieść o tym jak młody Debren na praktykę magiczną do wrogiej krainy posłan oko w oko ze smokiem staje i utraconą cześć dziewiczą przywrócić musi     - Zarooobki dla dziewic! - przebił się przez uliczny gwar basowy okrzyk. - Gooodziwe zarooobki! Debren potknął się z wrażenia, a potem przyspieszył kroku.

tobozo, 2022-12-01:
@martinberlin I know how you feel

[image]

The [font example sketch](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/3_fonts/3_fonts.ino) should help you with the declarative part.

Here's another [example](https://github.com/riraosan/ESP32_8BIT_CVBS/blob/master/sample/04_unicode_fonts/u8g2/u8g2.ino) on how to import u8g2 unicode fonts into lgfx.

Contributions to the wiki of this repository are welcome btw.

martinberlin, 2022-12-01:
Thanks!
Lovyan also told me is possible to generate Fonts using https://github.com/m5stack/FontCreator so I will try with that as well just to compare the results.
[Video of my 9.7" epub reader DIY build using LovyanGFX](https://twitter.com/martinfasani/status/1597256098301374465)

martinberlin, 2022-12-01:
https://github.com/m5stack/FontCreator -> Discarded for me. Bad documented. Generating all Ubuntu fonts also does not render polish. I don't really see much of a difference with Adafruit GFX font converter.
So far lass test will be with **u8g2** and then if nothing works will do it with Open Font Render, that it worked already for me when doing a Chinese weather panel.

tobozo, 2022-12-01:
there's still a way to load font files from the filesystem with `tft.loadFont( "/path/to/ttf/file", SPIFFS )`

found [another documentation](https://github.com/lovyan03/LovyanGFX/tree/master/src/lgfx/Fonts/IPA) based on u8g2

- [ttf to bdf](http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/)
- [bdf to c](https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdfconv.exe)

bdfconv settings for polish (found in [u8g2 MakeFile](https://github.com/olikraus/u8g2/blob/7e58b87b3cd63ed4016624cd00226ffe4f0bfe1d/tools/font/bdfconv/Makefile#L66)): 

	./bdfconv -f 1 -v -m '32-127,$$104-$$107,$$118-$$119,$$141-$$144,$$15a-$$15b,$$179-$$17c,$$d3,$$f3' ../bdf/unifont.bdf -n unifont_polish -o unifont_polish.c -d ../bdf/7x13.bdf

martinberlin, 2022-12-02:
Nice @tobozo thanks a lot. That of storing the ttf in SPIFFS sounds quite useful must say.
Would like to know, the way to use it will be to do something like this?

```C
GFXfont *m_regular_font = tft.loadFont( "/path/to/ttf/file", SPIFFS);
```

Does it return a GFXfont type or it's something different. 
If I could get a complete working example I will try it today. Thanks a lot for the hints!

tobozo, 2022-12-02:
sorry it's not TTF but VLW fonts, you can generate those using the `Processing` software.

There's a [tutorial on seeed wiki](https://wiki.seeedstudio.com/Wio-Terminal-LCD-Anti-aliased-Fonts/).

Processing will only see the system fonts though (no file picker), so make sure to install them before launching the interface.

And it's just this, tft.loadFont() has no return value:

`tft.loadFont( "/my_VLW_font.vlw", SPIFFS);`

I just tried it with `AbhayaLibre-Medium-48` on LittleFS and got this result, the font is downscaled at 50% so don't mind the quality loss:

[image]

martinberlin, 2022-12-02:
Great to see that it works. I will try that then!
Can you please copy the C part where you load the font and then assign it to the display?

I still don't see how to do a:   display.setFont(&MyFont);