lovyan03/LovyanGFX #230: MovingIcons.ino 放置していると画面が真っ暗になってしまう

GOB52, 2022-05-03:
M5Stack Gray
LovyanGFX 0.4.17
macOS Mojave 10.14.6  ArduinoIDE 1.8.19 スケッチ例からビルド

12分程放置していると画面が真っ暗になってしまいました。
シリアルには特に表示なし。
printf 仕込んでみた所ハングアップしているわけではないようです。

GOB52, 2022-05-03:
描画異常でなく、objects の位置がおかしくなっているのかもしれませんね。

lovyan03, 2022-05-03:
@GOB52 
ご報告ありがとうございます～
時折こちらでも、(MovingIcons以外でも)何か動作させたまま放置していて気付いたら画面が真っ暗ということがあり、
なにが原因なのか、どういう状態になっているのか、まだ良く分かっていません…。

lovyan03, 2022-05-03:
@GOB52 
そういえば一つ思い当たることが…SDカードを挿した状態だとLCDとの通信が化ける事があります。
可能でしたらSDカードを取り外した状態での比較検証をお願いしたいです。

M5StackはLCDとSDが同じバスを共有していますが、SDカードは起動直後はSPIモードではなくMMCモード動作なので、SPIモードへの移行処理をはじめに行っておかないと、LCDとの通信によってSDが誤動作して何らかのアクションを起こそうとし、SPI通信中に影響を与える信号を発して通信を阻害する場合があります。
その結果たとえばLCDがスリープコマンド等を誤認識して画面が消えるという可能性があるかもしれません。

M5GFXの方には先日のアップデートで、起動時にSDをSPIモードにする処理を追加しましたが、LovyanGFXのAutoDetectには未実装です。

GOB52, 2022-05-03:
@lovyan03 

SDカード抜いた状態での動作ですが、30分以上経っても表示は維持されていますね。

lovyan03, 2022-05-03:
@GOB52 
ありがとうございます～。
一方こちらではSDカードを挿してMovingIconsを動作させ放置していた３台のうちひとつが何らかのモード変更コマンドを誤認識したらしく、色が青くなりました。

[image]

以前こちらで起きた症状ではSDカードを挿したまま操作していると、輝度反転したり、16bitモードと24bitモードが変更されたり、様々な表示不具合が出ましたので、今回の事例もSDカードが原因かもですね。

GOB52, 2022-05-03:
動画はこちら。
変な体勢で撮ったので上下逆ご容赦(´･ω･`)
上記画像は00:00-00:02

https://user-images.githubusercontent.com/26270227/166471105-87374d9f-5179-4f01-8cd6-cfadc7a79bf5.mp4

lovyan03, 2022-05-03:
@GOB52 
なるほど、表示位置が縦にずれた場所に描画された、ということですかね？
であれば、縦座標を指定するコマンドがSDカードによって潰された可能性が考えられますね。
MovingIconsは１フレーム描画するために何回かに分けてpushSpriteしているため、
座標指定コマンドがSDカードに潰されたとすれば、前回の描画位置にpushSpriteされる可能性があります。

アイコンの移動内容は毎回ランダムではなく決まったパターンで動きますから、描画内容は毎回同じパターンになり、SPI通信内容も毎回同じになるはずです。
ゆえに、SDカード側も同じ通信内容を受け取って同じように誤認識して通信妨害をしてくるはずです。
ということで、別のSDカードを使うと、また違った不具合のパターンが出ると思います。

GOB52, 2022-05-03:
@lovyan03 
>なるほど、表示位置が縦にずれた場所に描画された、ということですかね？
ですね。

そしてSDカード自体に依存して現象や発生時間が変化する、と。

lovyan03, 2022-05-03:
@GOB52 
そうですね、同じSDカードを使う限りは同じ描画内容で同じように再現できるはずです。
こっちのカードはここでバグる、そっちのカードはそこでバグる、みたいな感じで。

GOB52, 2022-05-03:
@lovyan03 
おおお、確かにありえる挙動ですね>SDカード破損

まぁ現象の再現性を確保できているという事で、そこは所謂謎バグと違ってマシであると前向きに考えましょう。

lovyan03, 2022-05-03:
@GOB52 
はい、それで今回のIssueについては解決ということで良いでしょうか？
気になる挙動がありましたらば引き続きお知らせください。

GOB52, 2022-05-03:
@lovyan03 
現状で現象回避策は何かありますかね?

GOB52, 2022-05-03:
あ、 M5GFX と同様な処理を入れてあげればいいのか。そちらみてみます。
毎度対応ありがとうございました。

closed 2022-05-03 by GOB52

lovyan03, 2022-05-03:
@GOB52 
SD.beginを入れておくこと、動作中にSDカードを抜いて挿し直さないこと、ですかね。

SD.beginを使わないで独自にやる場合は、M5GFXでのこの部分を参考にしてみてください。
https://github.com/m5stack/M5GFX/blob/0.0.20/src/M5GFX.cpp#L290-L316

具体的なシーケンスは以下のとおりです
 - 通信クロックは400kHzまで
 - SDのCSをHIGHにしたまま0xFFを20Byte送る(ダミークロック)
 - SDのCSをLOWにし、0x7A 0x00 0x00 0x00 0x00 0xFD 0xFF 0xFF送ると共に、受信内容を確認
 - 最後の2バイトの内容に相違があれば既にSPIモードと判断し、SDのCSをHIGHにして終了
 - 最後の2バイトの内容が同じであれば現在MMCモードと判断し、続きの処理を実行
 - SDのCSをHIGHにしたまま0xFFを20Byte送る(ダミークロック)
 - SDのCSをLOWにし、0x40 0x00 0x00 0x00 0x00 0x95 0xFF 0xFF送る
 - SDのCSをHIGHにして終了

技術解説はこちらのサイトが参考になると思います
http://elm-chan.org/docs/mmc/mmc.html#spimode

lovyan03, 2022-05-04:
@GOB52 
developブランチにて修正を反映しました。
ボード自動認識処理を行う際にSDをSPIモードに移行する処理を追加しました。
またお時間あるときにお試し頂けましたら幸いです～

GOB52, 2022-05-04:
@lovyan03 

試してみました。
develop e5f4dd8

飯前に起動して50分経過。描画維持されてます。

GOB52, 2022-05-04:
@lovyan03 
起動から約3時間半経過、描画維持されてます。

lovyan03, 2022-05-04:
@GOB52 
検証ありがとうございます！うまく機能したようですね。
なお動作中にSD抜いて挿すとMMCモードに戻っちゃうので、これに関しては運用で気を付ける、という方向で…。

commit f1f9edce (2022-05-04): tweak AUTODETECT for ESP32. ( fix #230 )