lovyan03/LovyanGFX #231: ILI9481 + FT6236 issue

JohnTaipei, 2022-05-04:
Good morning Sir:
MCU: ESP32
Arduino IDE 1.8.13
Lovyan library : 0.4.17
LCD : 8-Bit parallel mode
ILI9481 + FT6236 has a issue
video:
https://www.facebook.com/groups/797613180807626/permalink/1110317682870506/

ILI9488 + FT6236 is OK

Could you help me this issue??

Thank you

lovyan03, 2022-05-04:
@JohnTaipei 
I am very sorry, but this is due to ILI9481 specifications.  
In the sample sketch, the rotation is frequently changed during the loop to test the setting, but we do not think you will use it that way in actual applications, so please comment out `setRotation` in the loop function and try it.  

ILI9481 does not have as much freedom in terms of rotation as other panels, and changing rotation during operation is not recommended.

JohnTaipei, 2022-05-04:
@lovyan03 
I got it
ILI9488,ST7796 work well
Thank for your answer

closed 2022-06-24 by lovyan03