lovyan03/LovyanGFX #233: -Wshadow warnings everywhere

wrgallo, 2022-05-06:
Compiling with the flag `-Wshadow` will trigger **lots** of warnings/errors in the library. 
Many companies have 'coding standards' that specify that every parameter in a function must start with _param_ or the letter _p_, **only to avoid shadowing**.

Example: 
```Cpp
class MyClass{
   protected:
   int a;
   
   public:
   void foo(int a){
      a = a; // This must trigger a warning or error [-Wshadow]
   }
   
   void bar(int paramA){
      a = paramA; // This is fine
   }
   
   void foo2(int a){
      this.a = a; // This is fine too
      a = 2; // But this is not  [-Wshadow]
   }
```

[Shadowing](https://en.wikipedia.org/wiki/Variable_shadowing) occurs _at least_ 7 times in [lgfx_fonts.hpp](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/lgfx_fonts.hpp) :
1. [declaration of 'baseline' shadows a member of 'lgfx::v1::BaseFont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L54)
1. [declaration of 'baseline' shadows a member of 'lgfx::v1::GLCDfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L65)
1. [declaration of 'baseline' shadows a member of 'lgfx::v1::FixedBMPfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L73)
1. [declaration of 'baseline' shadows a member of 'lgfx::v1::BMPfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L81)
1. [declaration of 'baseline' shadows a member of 'lgfx::v1::RLEfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L89)
1. [declaration of 'y_advance' shadows a member of 'lgfx::v1::BDFfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L101)
1. [declaration of 'range_num' shadows a member of 'lgfx::v1::GFXfont' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/lgfx_fonts.hpp#L151)

[Shadowing](https://en.wikipedia.org/wiki/Variable_shadowing) occurs _at least_ 2 times in [colortype.hpp](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/misc/colortype.hpp) :
1. [declaration of 'depth' shadows a member of 'lgfx::v1::color_conv_t' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/misc/colortype.hpp#L788)
1. [declaration of 'depth' shadows a member of 'lgfx::v1::color_conv_t' [-Wshadow]](https://github.com/lovyan03/LovyanGFX/blob/5cfb85d2843ad13cea571af86154d26551991ca8/src/lgfx/v1/misc/colortype.hpp#L793)

PlatformIO compiling results with library version 0.4.15:
[image]

tobozo, 2022-05-07:
@wrgallo thanks for your feedback :+1: 

Is there an exhaustive `build_flags` value for platformio.ini you can suggest to meet said standards?

wrgallo, 2022-05-09:
> @wrgallo thanks for your feedback +1
> 
> Is there an exhaustive `build_flags` value for platformio.ini you can suggest to meet said standards?

```ini
-Wall -Wextra -Wshadow -Wdouble-promotion -Wformat -Wnonnull
```

tobozo, 2022-05-09:
thanks :+1:  will try to use it where possible

tobozo, 2022-10-23:
closing this issue as solved, feel free to reopen if needed

wrgallo, 2022-10-24:
@tobozo 
> closing this issue as solved, feel free to reopen if needed

There is still one last pending `-WShadow` warning at [`lgfx_fonts.hpp`](https://github.com/lovyan03/LovyanGFX/blob/3af472568a319bb4c5543f00edc0c30f02770c1a/src/lgfx/v1/lgfx_fonts.hpp#L162)

```C++
declaration of 'range_num' shadows a member of 'lgfx::v1::GFXfont' [-Wshadow]
```

tobozo, 2022-10-24:
oh right, looks like there's some warnings in `lgfx/utility/lgfx_qrcode.c` too, reopening

tobozo, 2022-10-24:
I managed to reduce the warnings to this.

Note: all `-Wshadow` warnings are now deliberately silent with all `v0` codebase.
As this part of the library won't be maintained any more I didn't bother fixing the warnings there and only added some pragmas to get them silent.

I've also left the `[-Wdouble-promotion]` warnings in place as I don't know what could be the effects, and it's not my decision to change that.

However, a few `-Wshadow` warnings are still emitted by the arduino packages, I'm not sure how to cope with that.

I've created a [special branch](https://github.com/lovyan03/LovyanGFX/tree/wshadow) with the `-Wshadow` changes,.

this will need @lovyan03's approval though

```log
Processing esp32@latest (board: esp32dev; platform: https://github.com/tasmota/platform-espressif32; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.5+sha.f53ca2d) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.5 
 - tool-esptoolpy @ 1.40300.220914 (4.3.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LovyanGFX @ 0.4.18
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/Fonts/IPA/lgfx_font_japan.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_cn.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_ja.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_kr.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_tw.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/utility/lgfx_pngle.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/utility/lgfx_qoi.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/utility/lgfx_qrcode.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/utility/lgfx_tjpgd.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/utility/miniz.c.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/LGFXBase.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/lgfx_fonts.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/panel/Panel_GDEW0154M09.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/panel/Panel_IT8951.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/platforms/esp32_common.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/platforms/samd51_common.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/touch/Touch_FT5x06.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/touch/Touch_GT911.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/touch/Touch_STMPE610.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v0/touch/Touch_XPT2046.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v1/LGFXBase.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v1/LGFX_Button.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v1/LGFX_Sprite.cpp.o
Compiling .pio/build/esp32@latest/lib82e/LovyanGFX/lgfx/v1/lgfx_fonts.cpp.o
.pio/libdeps/esp32@latest/LovyanGFX/src/lgfx/v1/LGFXBase.cpp: In member function 'void lgfx::v1::LGFXBase::draw_bezier_helper(int32_t, int32_t, int32_t, int32_t, int32_t, int32_t)':
.pio/libdeps/esp32@latest/LovyanGFX/src/lgfx/v1/LGFXBase.cpp:737:23: warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]
         dx = 4.0 * sy * cur * (x1 - x0) + xx - xy;
              ~~~~~~~~~^~~~~
.pio/libdeps/esp32@latest/LovyanGFX/src/lgfx/v1/LGFXBase.cpp:738:23: warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]
         dy = 4.0 * sx * cur * (y0 - y1) + yy - xy;
              ~~~~~~~~~^~~~~
.pio/libdeps/esp32@latest/LovyanGFX/src/lgfx/v1/LGFXBase.cpp: At global scope:
[... 257 lines cut ...]
Compiling .pio/build/esp32@latest/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32@latest/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32@latest/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32@latest/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32@latest/libFrameworkArduino.a
Indexing .pio/build/esp32@latest/libFrameworkArduino.a
Linking .pio/build/esp32@latest/firmware.elf
Retrieving maximum program size .pio/build/esp32@latest/firmware.elf
Checking size .pio/build/esp32@latest/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.9% (used 15904 bytes from 327680 bytes)
Flash: [==        ]  19.6% (used 257481 bytes from 1310720 bytes)
Building .pio/build/esp32@latest/firmware.bin
esptool.py v4.3
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
========================= [SUCCESS] Took 39.00 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
esp32@latest   SUCCESS   00:00:38.996
========================= 1 succeeded in 00:00:38.996 =========================

```

wrgallo, 2022-10-24:
About the `-Wdouble-promotion` warnings, just placing `f` after the float constants will silent the warnings, and **can even make the math operations faster**, as described by [GCC Warning Options](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html).

A `float` constant without the [floating-point literal](https://en.cppreference.com/w/cpp/language/floating_literal) `f` explicitly added to the number is actually a `double`, and if a double and a float actually differ (usually it does), all math operations between a float and a double will require the float number to be promoted to double.

```C
#define MY_FLOAT_CONSTANT 3.14f // Usually take 32bits (4 bytes)
#define MY_DOUBLE_CONSTANT 3.14 // Usually take 64bits (8 bytes)
```

Example:
```C
float area_wrong(float radius)
{
  // radius would be promoted from float to double, to match the constant that should be float too
   return 3.14159 * radius * radius;
}

float area_right(float radius)
{
  // radius and the PI constant are both float
   return 3.14159f * radius * radius;
}
```

tobozo, 2022-10-24:
that's also happening when multiplicating in32_t and floats, i.e. in the beziers functions, could be one of those situations where ignoring the warning is more beneficial for performances than fixing it?

that won't fix the issues in the espressif core anyway

tobozo, 2022-10-24:
based on your suggestion I've pushed some changes on the 'wshadow' branch and updated the log output in my previous comment, much smaller now :-)

lovyan03, 2022-10-25:
There are so many changes that it's very hard to check.

closed 2022-12-25 by github-actions[bot]