lovyan03/LovyanGFX #234: Strange effect which depend on color

RA-NGR, 2022-05-07:
Display: 3.2inch 16BIT Module ILI9341 SKU:MRB3205 (see at ww.lcdwiki.com)
MCU: MH ET LIVE ESP32MiniKit
Test sketch is:
`#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ILI9341 _panel_instance;
    lgfx::Bus_Parallel8 _bus_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.i2s_port = I2S_NUM_0; 
      cfg.freq_write = 20000000;
      cfg.pin_wr =  4;          
      cfg.pin_rd =  2;          
      cfg.pin_rs = 15;          
      cfg.pin_d0 = 12;          
      cfg.pin_d1 = 13;          
      cfg.pin_d2 = 5;           
      cfg.pin_d3 = 26;          
      cfg.pin_d4 = 17;          
      cfg.pin_d5 = 16;          
      cfg.pin_d6 = 27;          
      cfg.pin_d7 = 14;          
      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }
    { 
      auto cfg = _panel_instance.config();
      cfg.pin_cs           =    13;
      cfg.pin_rst          =    32;
      cfg.pin_busy         =    -1;
      cfg.panel_width      =   240;
      cfg.panel_height     =   320;
      cfg.offset_x         =     0;
      cfg.offset_y         =     0;
      cfg.offset_rotation  =     0;
      cfg.dummy_read_pixel =     8;
      cfg.dummy_read_bits  =     1;
      cfg.readable         = false;
      cfg.invert           = false;
      cfg.rgb_order        = false;
      cfg.dlen_16bit       = false;
      cfg.bus_shared       = false;
      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance);
  }
};

LGFX display;

void setup(void)
{
  Serial.begin(115200);
  display.init();

  display.setRotation(3);
  display.setColorDepth(16);
}

void loop(void)
{
    display.startWrite();
    display.fillScreen(0xFFE0);
    delay(500);
    display.fillScreen(0x001F);
    delay(500);
    display.endWrite();
}`
Result:
[image]
[image]
 If I change colors to 0xFEE0 and 0x001E respectively - fill is ok

lovyan03, 2022-05-07:
```
 cfg.dlen_16bit       = false;
```
please set `cfg.dlen_16bit` to `true`

lovyan03, 2022-05-07:
@RA-NGR 
If the state of the data bits affects the behavior, it may be possible that the wiring is shorted ?

RA-NGR, 2022-05-08:
Thanks, the problem was a wrong pin number ))
cfg.dlen_16bit = true makes blank white screen for me, so I returned 'false' value

closed 2022-05-08 by RA-NGR