lovyan03/LovyanGFX #235: Issues with SSD1306 128x32. Content are squeezed vertically

zenaro147, 2022-05-07:
Hello, I'm having some problems while setting up my project.

I'm using an ESP32 WROOM and this [I2C OLED 128x32](https://www.aliexpress.com/item/32983556053.html) from AliExpress with the LongTextScroll example.

I can display the text but not in full screen mode. It's a little tight.

I've tested it with my main project examples as well, but I have the same problem.

A friend of mine tested it with the LilyGO TTGO T2 with my main project and it works fine.

Here's the code I'm usint to test now.
-------

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64

// ESP32でLovyanGFXを独自設定で利用する場合の設定例

/// 独自の設定を行うクラスを、LGFX_Deviceから派生して作成します。
class LGFX_SSD1306 : public lgfx::LGFX_Device
{
  lgfx::Panel_SSD1306     _panel_instance;
  lgfx::Bus_I2C       _bus_instance;   // I2Cバスのインスタンス (ESP32のみ)

public:
  LGFX_SSD1306(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。
// I2Cバスの設定
      cfg.i2c_port    = 0;          // 使用するI2Cポートを選択 (0 or 1)
      cfg.freq_write  = 400000;     // 送信時のクロック
      cfg.freq_read   = 400000;     // 受信時のクロック
      cfg.pin_sda     = 21;         // SDAを接続しているピン番号
      cfg.pin_scl     = 22;         // SCLを接続しているピン番号
      cfg.i2c_addr    = 0x3C;       // I2Cデバイスのアドレス

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

//      cfg.pin_cs           =    -1;  // CSが接続されているピン番号   (-1 = disable)
//      cfg.pin_rst          =    -1;  // RSTが接続されているピン番号  (-1 = disable)
//      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

//      cfg.panel_width      =   SCREEN_WIDTH;  // 実際に表示可能な幅
//      cfg.panel_height     =   SCREEN_HEIGHT;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
//      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
//      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
//      cfg.readable         =  false;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
//      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
//      cfg.dlen_16bit       = false;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
//      cfg.bus_shared       =  false;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

// 以下はST7735やILI9163のようにピクセル数が可変のドライバで表示がずれる場合にのみ設定してください。
    cfg.memory_width     =   SCREEN_WIDTH;  // ドライバICがサポートしている最大の幅
    cfg.memory_height    =   SCREEN_HEIGHT;  // ドライバICがサポートしている最大の高さ

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

// 準備したクラスのインスタンスを作成します。
//LGFX display;
static LGFX_SSD1306 lcd; // LGFX_SSD1306のインスタンス（クラスLGFX_SSD1306を使ってlcdでいろいろできるようにする）を作成
static LGFX_Sprite canvas(&lcd);  // スプライトを使うためのLGFX_Spriteのインスタンスを作成

static constexpr char text[] = "Hello world ! こんにちは世界！ this is long long string sample. 寿限無、寿限無、五劫の擦り切れ、海砂利水魚の、水行末・雲来末・風来末、喰う寝る処に住む処、藪ら柑子の藪柑子、パイポ・パイポ・パイポのシューリンガン、シューリンガンのグーリンダイ、グーリンダイのポンポコピーのポンポコナの、長久命の長助";
static constexpr size_t textlen = sizeof(text) / sizeof(text[0]);
size_t textpos = 0;

void setup(void) 
{
  lcd.init();

  // 画面が横長になるように回転
  if (lcd.width() < lcd.height()) lcd.setRotation(lcd.getRotation() ^ 1);

  canvas.setColorDepth(8);
  canvas.setFont(&fonts::lgfxJapanMinchoP_32);
  canvas.setTextWrap(false);        // 右端到達時のカーソル折り返しを禁止
  canvas.createSprite(lcd.width() + 36, 36); // 画面幅+１文字分の横幅を用意
}

void loop(void)
{
  int32_t cursor_x = canvas.getCursorX() - 1;   // 現在のカーソル位置を取得し、1ドット左に移動
  if (cursor_x <= 0) // カーソル位置が左端に到達している場合は一周したと判断
  {
    textpos = 0;            // 文字列の読取り位置をリセット
    cursor_x = lcd.width(); // 新たな文字が画面右端に描画されるようにカーソル位置を変更
  }

  canvas.setCursor(cursor_x, 0); // カーソル位置を更新
  canvas.scroll(-1, 0);          // キャンバスの内容を1ドット左にスクロール
  while (textpos < textlen && cursor_x <= lcd.width()) // 画面右端に文字が書けるか判定
  {
    canvas.print(text[textpos++]);   // 1バイトずつ出力 (マルチバイト文字でもこの処理で動作します)
    cursor_x = canvas.getCursorX();  // 出力後のカーソル位置を取得
  }

  canvas.pushSprite(&lcd, 0, 0);
}

lovyan03, 2022-05-08:
Hello @zenaro147 
LovyanGFX has recently added a new function called `setComPins` to `Panel_SSD1306`.
If your screen display size is halved, try using this `setComPins` function.
If the function is not found, update the library.

Please add the following line to your source code
```
    _panel_instance.config(cfg);

    _panel_instance.setComPins(0x02); // please add this line;
  }
  setPanel(&_panel_instance); // 使用するパネルをセットします。
 }
};
```

The specific behavior of this function specifies the parameters for sending a 0xDA command to `SSD1306`.
The `SSD1306` may have different connection patterns between the panel and controller depending on the product, and the 0xDA command adjusts for this.
If it does not display correctly, try the following values of the `setComPins` function one at a time.
```
    _panel_instance.setComPins(0x02);
//  _panel_instance.setComPins(0x12);
//  _panel_instance.setComPins(0x22);
//  _panel_instance.setComPins(0x32);
```

If you want to see the details of the 0xDA command, please search the SSD1306 specifications.
The following images are excerpts from the specifications

[image]

zenaro147, 2022-05-08:
Hi @lovyan03 !

Thanks for the quick awnser!

I added the setComPins line and works fine now!
Huge thanks!

I'll add this config and a comment about that in my main project too!

closed 2022-05-08 by zenaro147