lovyan03/LovyanGFX #237: How to use 16bit parallel with ESP32 S3

uldara1, 2022-05-12:
Hi @lovyan03 i just start with 16bit parallel for ESP S3

but when i add pin d0-d15 it show error 
pls check attached file 
[image]

lovyan03, 2022-05-12:
@uldara1 
The ESP32-S3 has a new LCD_CAM peripheral in place of I2S.
Therefore, I2S is no longer used and the `i2s_port` setting has been removed, so comment out the line with the error.
Also, if you want to use 16-bit parallel, use `Bus_Parallel16` instead of `Bus_Parallel8`.

uldara1, 2022-05-12:
@lovyan03 
thank for your support now no error 
unfortunately library not support my display yet  
my display is RM68120

closed 2022-06-06 by lovyan03