lovyan03/LovyanGFX #238: Set coordinate to TFT write *.BMP from SD

sgamoct, 2022-05-19:
Hi, I am using a makerkfab esp32 card with ili 320x480 and capacitive touch.
I would like to view some images, uploaded from sd, in a specific position.
how can I set the coordinates of a non-full-screen image loaded from SD?
the example code is this:
`int print_img(fs::FS &fs, String filename, int x, int y)
{
  SPI_ON_SD;
  File f = fs.open(filename, "r");
  if (!f)
  {
    Serial.println("Failed to open file for reading");
    f.close();
    return 0;
  }

  f.seek(54);
  int X = x;
  int Y = y;
  uint8_t RGB[3 * X];
  for (int row = 0; row < Y; row++)
  {
    f.seek(54 + 3 * X * row);
    f.read(RGB, 3 * X);
    SPI_OFF_SD;
//     tft.pushImage(0, row, X, 1, (lgfx::rgb888_t *)RGB);
    tft.pushImage(0, row, X, 1, (lgfx::rgb888_t *)RGB);
    SPI_ON_SD;
  }

  f.close();
  SPI_OFF_SD;
  return 0;
}`

lovyan03, 2022-05-19:
@sgamoct 
LovyanGFX provides functions for drawing BMP, JPG, and PNG.
Perhaps you can solve this by simply using drawBmp or drawBmpFile.

 - drawBmpFile  // use filename access
 - drawBmp        // use data pointer access

```
#include <SD.h>

#define LGFX_AUTODETECT
#include <LGFX_AUTODETECT.h>

static LGFX display;

void setup(void)
{
  display.init();

  display.drawBmpFile(SD, "filename.bmp", 10, 20);
}
```

sgamoct, 2022-05-23:
Hi, thanks for the reply. I tried your suggestion but to no avail.
I changed
#include <LGFX_AUTODETECT.h>
to
#include <LGFX_AUTODETECT.hpp>
because he did not compile.
this is the code i used.
the image is on the root of the SD.
the screen remains black.
in your example:
display.drawBmpFile (SD, "/dry.bmp", 10, 20)
10 and 20 are x, y coordinates?
Hi, thanks for the reply. I tried your suggestion but to no avail.
I changed
#include <LGFX_AUTODETECT.h>
to
#include <LGFX_AUTODETECT.hpp>
because he did not compile.
this is the code i used.
the image is on the root of the SD.
the screen remains black.
in your example:
display.drawBmpFile (SD, "/dry.bmp", 10, 20)
10 and 20 are x, y coordinates?:
`

#include "SPI.h"
#include "ILI9488.h"
#include "SD.h"
#include "FS.h"
#define LGFX_AUTODETECT
#include <LGFX_AUTODETECT.hpp>

static LGFX display;

//SPI
#define SPI_MOSI 13
#define SPI_MISO 12
#define SPI_SCK 14

//SD Card
#define SD_CS 4

//TFT
#define TFT_CS 15
#define TFT_DC 33
#define TFT_LED -1 //1//-1
#define TFT_RST -1 //3//-1

//SPI control
#define SPI_ON_TFT digitalWrite(TFT_CS, LOW)
#define SPI_OFF_TFT digitalWrite(TFT_CS, HIGH)
#define SPI_ON_SD digitalWrite(SD_CS, LOW)
#define SPI_OFF_SD digitalWrite(SD_CS, HIGH)

// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
ILI9488 tft = ILI9488(TFT_CS, TFT_DC, TFT_RST);

void setup()
{
  Serial.begin(115200);

  //SPI init
  pinMode(SD_CS, OUTPUT);
  pinMode(TFT_CS, OUTPUT);
  SPI_OFF_SD;
  SPI_OFF_TFT;
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPISettings(60000000, MSBFIRST, SPI_MODE0);
  SPI.setFrequency(60000000);

  SPI_ON_SD;
  //SD(SPI) init
  if (!SD.begin(SD_CS, SPI, 80000000))
  {
    Serial.println("Card Mount Failed");
    while (1)
      ;
  }
  else
  {
    Serial.println("Card Mount Successed");
  }
  //sd_test();
  SPI_OFF_SD;

  Serial.println("SD init over.");

  //TFT(SPI) init
  SPI_ON_TFT;
  tft.begin();
  tft.setRotation(1);
  SPI_OFF_TFT;
  Serial.println("TFT init over.");
//  print_img(SD, "/dry.bmp");

  display.init();

  display.drawBmpFile(SD, "/1.bmp", 10, 20); // test for backslah
  display.drawBmpFile(SD, "1.bmp", 10, 20);
}

void loop()
{
}`

sgamoct, 2022-05-23:
> Hi, thanks for the reply, I tried your suggestion but to no avail. I changed `#include <LGFX_AUTODETECT.h>` to `#include <LGFX_AUTODETECT.hpp>` because he did not compile. The image is on the root of the SD, but the screen remains black. in your example: `display.drawBmpFile (SD, "/dry.bmp", 10, 20) ` 10 and 20 are x, y coordinates? this is the code i used:
> 
> ```
> #include "SPI.h"
> #include "ILI9488.h"
> #include "SD.h"
> #include "FS.h"
> #define LGFX_AUTODETECT
> #include <LGFX_AUTODETECT.hpp>
> 
> static LGFX display;
> 
> //SPI
> #define SPI_MOSI 13
> #define SPI_MISO 12
> #define SPI_SCK 14
> 
> //SD Card
> #define SD_CS 4
> 
> //TFT
> #define TFT_CS 15
> #define TFT_DC 33
> #define TFT_LED -1 //1//-1
> #define TFT_RST -1 //3//-1
> 
> //SPI control
> #define SPI_ON_TFT digitalWrite(TFT_CS, LOW)
> #define SPI_OFF_TFT digitalWrite(TFT_CS, HIGH)
> #define SPI_ON_SD digitalWrite(SD_CS, LOW)
> #define SPI_OFF_SD digitalWrite(SD_CS, HIGH)
> 
> // Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
> ILI9488 tft = ILI9488(TFT_CS, TFT_DC, TFT_RST);
> 
> 
> void setup()
> {
>   Serial.begin(115200);
> 
>   //SPI init
>   pinMode(SD_CS, OUTPUT);
>   pinMode(TFT_CS, OUTPUT);
>   SPI_OFF_SD;
>   SPI_OFF_TFT;
>   SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
>   SPISettings(60000000, MSBFIRST, SPI_MODE0);
>   SPI.setFrequency(60000000);
> 
>   SPI_ON_SD;
>   //SD(SPI) init
>   if (!SD.begin(SD_CS, SPI, 80000000))
>   {
>     Serial.println("Card Mount Failed");
>     while (1)
>       ;
>   }
>   else
>   {
>     Serial.println("Card Mount Successed");
>   }
>   //sd_test();
>   SPI_OFF_SD;
> 
>   Serial.println("SD init over.");
> 
>   //TFT(SPI) init
>   SPI_ON_TFT;
>   tft.begin();
>   tft.setRotation(1);
>   SPI_OFF_TFT;
>   Serial.println("TFT init over.");
> 
>   display.init();
> 
>   display.drawBmpFile(SD, "/1.bmp", 10, 20); // test for backslah
>   display.drawBmpFile(SD, "1.bmp", 10, 20);
> }
> 
> void loop()
> {
> }
> ```

Hi, the error was keep
`tft.begin ();`
without this the image is loaded  :D
:S but it is not possible to set the rotation of the display with the command 
 `tft.setRotation();`
why?

lovyan03, 2022-05-23:
@sgamoct 
Sorry, the example code I gave was insufficient.
Try the following code.

```
#include <SD.h>
#define SDCARD_SPI SPI
#ifndef SDCARD_SS_PIN
#define SDCARD_SS_PIN 4
#endif

#define LGFX_USE_V1
#define LGFX_AUTODETECT
#include <LGFX_AUTODETECT.hpp>

static LGFX display;

void setup(void)
{
// Initialize display before SD.
  display.init();

// The maximum communication speed of the SD card is 25 MHz in SPI mode.
// even if 80 MHz is set, it is capped internally.
  while (!SD.begin(SDCARD_SS_PIN, SDCARD_SPI, 25000000))
  {
    delay(100);
  }
}

void loop(void)
{
  display.setRotation(display.getRotation() + 1);

  for (int y = 0; y <= 256; y += 64)
  {
    for (int x = 0; x <= 256; x += 64)
    {
      display.drawBmpFile(SD, "/1.bmp", x, y);
    }
  }

  auto file = SD.open("/1.bmp");
  for (float scale = 0.1; scale < 3; scale += 0.1)
  {
    file.seek(0);
    display.drawBmp(&file, 0, 0, 0, 0, 0, 0, scale);
  }
  file.close();
}
```

sgamoct, 2022-05-31:
> @sgamoct Sorry, the example code I gave was insufficient. Try the following code.
>
> ```
> #include <SD.h>
> #define SDCARD_SPI SPI
> #ifndef SDCARD_SS_PIN
> #define SDCARD_SS_PIN 4
> #endif
>
> #define [...]

Thanks, I tried it and it works fine!

closed 2022-05-31 by sgamoct