lovyan03/LovyanGFX #239: Can't find #include <driver/i2c.h> and  #include <driver/gpio.h> when trying to build for esp32

Techseeker-404, 2022-05-21:
I was getting a really horrible error when trying to build with esp32 tool chain 
```
idf.py build
....
....
/home/WORK/ESP_PROJECT/esp32-resnova/components/LovyanGFX/src/lgfx/v1/platforms/esp32/common.hpp:28:10: fatal error: driver/gpio.h: No such file or directory
 #include <driver/gpio.h>
          ^~~~~~~~~~~~~~~
...
...
/home/WORK/ESP_PROJECT/esp32-resnova/components/LovyanGFX/src/lgfx/v0/panel/../platforms/esp32_common.hpp:7:10: fatal error: driver/i2c.h: No such file or directory
 #include <driver/i2c.h>
          ^~~~~~~~~~~~~~
...
...
compilation terminated.
make[2]: *** [esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/build.make:398: esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/lgfx_fonts.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4437: esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2
```
The build is failing once after including the LovyanGFX library as a component , with all other components the build is working fine.

tobozo, 2022-05-21:
hey @Techseeker-404 thanks for your feedback :+1: 

can you please provide more info on your esp-idf version, system/architecture and development environment so we can reproduce this error?

if you're using esp-idf 4.4.1 then you probably found a bug in LovyanGFX, and I need to update the CI script which only handles 4.4-RC1 as the highest version number.

Techseeker-404, 2022-05-21:
@tobozo 
Hey Thanks for the reply .
The version of esp-idf is 4.4.1
Other details:
Os: PopOs  - Linux X86_64
Development environment: "Just bare bones; with a text editor and terminal as  I generated  C project template with idf.py"
Build manager : idf.py  (From esp-idf toolchain)

tobozo, 2022-05-21:
[image] and nothing to report with [4.4.1 build](https://github.com/lovyan03/LovyanGFX/runs/6539014941?check_suite_focus=true)

this opts out esp-idf version as the source of the problem

is your project source tree sharable?

Techseeker-404, 2022-05-22:
```
$tree . -d  (truncated to directory level)
.
├── build
│   ├── bootloader
│   │   ├── CMakeFiles
│   │   │   ├── 3.22.1
│   │   │   │   ├── CompilerIdASM
│   │   │   │   ├── CompilerIdC
│   │   │   │   │   └── tmp
│   │   │   │   └── CompilerIdCXX
│   │   │   │       └── tmp
│   │   │   ├── app.dir
│   │   │   ├── bootloader.elf.dir
│   │   │   ├── CheckTypeSize
│   │   │   ├── CMakeTmp
│   │   │   ├── confserver.dir
│   │   │   ├── erase_flash.dir
│   │   │   ├── gen_project_binary.dir
│   │   │   ├── git-data
│   │   │   ├── menuconfig.dir
│   │   │   ├── monitor.dir
│   │   │   ├── _project_elf_src.dir
│   │   │   ├── save-defconfig.dir
│   │   │   ├── size-components.dir
│   │   │   ├── size.dir
│   │   │   ├── size-files.dir
│   │   │   ├── uf2-app.dir
│   │   │   └── uf2.dir
│   │   ├── config
│   │   └── esp-idf
│   │       ├── bootloader
│   │       │   └── CMakeFiles
│   │       ├── bootloader_support
│   │       │   └── CMakeFiles
│   │       │       └── __idf_bootloader_support.dir
│   │       │           ├── bootloader_flash
│   │       │           │   └── src
│   │       │           └── src
│   │       │               └── esp32
│   │       ├── CMakeFiles
│   │       ├── efuse
│   │       │   └── CMakeFiles
│   │       │       ├── efuse-common-table.dir
│   │       │       ├── efuse_common_table.dir
│   │       │       ├── efuse-custom-table.dir
│   │       │       ├── efuse_custom_table.dir
│   │       │       ├── efuse_test_table.dir
│   │       │       ├── __idf_efuse.dir
│   │       │       │   ├── esp32
│   │       │       │   └── src
│   │       │       │       └── efuse_controller
│   │       │       │           └── keys
│   │       │       │               └── without_key_purposes
│   │       │       │                   └── three_key_blocks
│   │       │       ├── show-efuse-table.dir
│   │       │       └── show_efuse_table.dir
│   │       ├── esp32
│   │       │   └── CMakeFiles
│   │       ├── esp_common
│   │       │   └── CMakeFiles
│   │       │       └── __idf_esp_common.dir
│   │       │           └── src
│   │       ├── esp_hw_support
│   │       │   ├── CMakeFiles
│   │       │   │   └── __idf_esp_hw_support.dir
│   │       │   │       └── port
│   │       │   │           └── esp32
│   │       │   └── port
│   │       │       └── esp32
[... 808 lines cut ...]
│           │   └── test_fonts
│           └── unity
└── main
```
I only included the directory level tree structure

$ tree main
```
main
├── CMakeLists.txt
└── esp32-resnova.c

0 directories, 2 files
```
$ cat main/CMakeLists.txt
```
idf_component_register(
    SRCS "esp32-resnova.c"
    INCLUDE_DIRS "."
    REQUIRES lvgl LovyanGFX driver
)
```

As mentioned on the previous comment in the thread "idf.py build" works fine after adding lvgl as a component . But the error started to show once LovyanGFX was added as a git submodule in component.
so its not likely a source tree problem i guess .

tobozo, 2022-05-22:
Sorry for my bad choice of words, I meant sharing the source, so it's possible to checkout and reproduce the error,

Most LGFX examples I've seen working with esp-idf were using the `.cpp` extension in the main folder though, can you afford to rename your main file ?

```
main
├── CMakeLists.txt
└── esp32-resnova.c
```

also found this recent example, maybe it can be of some help ? https://github.com/sukesh-ak/LVGL8-WT32-SC01-IDF

Techseeker-404, 2022-05-22:
SORRY! It was a version issue. Eventhough i changed the commit to version of esp-idf to 4.4.1 . I forget to install the the required dependency of the paerticular version once i roll back everything form version 5 to 4.4.1 everything works fine. As the project involves some Rust code I decided to go with the V 5.0 initially, that made entire thing break.

Once again sorry for troubling your time.

closed 2022-05-22 by Techseeker-404