lovyan03/LovyanGFX #240: Could you support ESP32-S3 for RGB panle??

JohnTaipei, 2022-05-22:
Hello Sir:
Could you support ESP32-S3 for RGB panle(4~7" LCD)??

for example:
1.esp-idf
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel

2. Line43~52
https://github.com/moononournation/Arduino_GFX/blob/master/examples/PDQgraphicstest/PDQgraphicstest.ino

Thank you

lovyan03, 2022-05-23:
@JohnTaipei 
I don't know or own an RGB panel yet so I can't make it.
Please let me know if you have any recommendations or online stores where I can easily purchase the product.

JohnTaipei, 2022-05-24:
@lovyan03 
Ok, I see
But it is too expensive $80
or waiting for my PCB design
Thank you
https://www.aliexpress.com/item/1005004013252670.html?spm=a2g0o.detail.1000014.11.25d0714eDignr0&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.14452.226710.0&scm_id=1007.14452.226710.0&scm-url=1007.14452.226710.0&pvid=7a55b207-e154-40a9-b459-e1dd64923ac0&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.14452.226710.0,pvid:7a55b207-e154-40a9-b459-e1dd64923ac0,tpp_buckets:668%232846%238115%232000&pdp_ext_f=%7B%22sku_id%22%3A%2212000027738395551%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21USD%21%2180.0%21%21%21%21%21%402103239f16533562382435274e24e6%2112000027738395551%21rec

tobozo, 2022-05-26:
@JohnTaipei found this https://github.com/W00ng/ESP32-S3-RGB-Panel

JohnTaipei, 2022-05-27:
@tobozo Thank for your infomation
But I can not use esp-idf
I think LovyanGFX library is better

htvekov, 2022-09-28:
Hi'

I'm adding a request to this thread, as there's now several, additional devices 'out there' using same 480x480 res. 4" panel with identical display driver and touch controller.

Eg. the GS-T3E 4" wall display device with 3 gang relay switch.
This device has an ESP32-S3R8 onboard, 8MB PSRAM, 16MB external flash, 4" Capacitive 480x480 res. HD IPS LCD, ST7701S display driver and Goodix GT911 touch controller.

@fvanroie, the founder of openHASP, is currently implementing ESP32-S3 support in openHASP. LovyanGFX is already implemented. So I would really appreciate if this display could be supported in LovyanGFX as well 😍🤗

Currently the display is ,as you mention, only implemented in the esp-idf library and actually also in ArduinoGFX.

Here's a [link](https://github.com/HASwitchPlate/openHASP/discussions/383) to all our gathered info on the GS-T3E device 

Regards
Henning

KamilLenica, 2023-03-18:
Any chance to get support for this type of panel?

lovyan03, 2023-03-21:
@JohnTaipei 
Please set freq_write low. It acts as a pixel clock.
Because PSRAM is used as a buffer, if the freq_write is too high, accesses will not be made in time and glitches will occur.

JohnTaipei, 2023-03-23:
@lovyan03 
yes, I did it
try 10,12,15MHz
but still jitter

lovyan03, 2023-03-23:
umm,, please try develop branch version.

JohnTaipei, 2023-03-24:
@lovyan03 
Hello Sir:
i tried develop version
jitter is disappear
https://www.youtube.com/watch?v=q4bKGo4KR5k
https://www.youtube.com/watch?v=3_RdF-g1FH0
Thanks a lot

but the performance is slow fps

tobozo, 2023-03-24:
@JohnTaipei ~20fps on a 800*480 at @16bpp is 15MB/s, it's possible the performances for this configuration are already nominal.

JohnTaipei, 2023-03-24:
@tobozo 
yes , about 21fps

i tried to use ArduinoGFX ,about 29fps
but still slow fps

this  video is about 48fps
https://www.bilibili.com/video/BV1de411G7QM/?spm_id_from=333.999.0.0&vd_source=1d45c77050dae7a74103ea87ff23afae

i tried NT35501 800*480 parallel 16-bits + LoyanGFX = 55pfs
https://www.youtube.com/watch?v=uYeeqBHdWEI

lovyan03, 2023-03-24:
In RGB panel operation, all drawing processing is done to PSRAM and output processing is done by DMA transfer from PSRAM.
In other words, it depends on the speed setting of PSRAM.
The PSRAM speed and QIO, DIO, etc. settings at build time must be the fastest, but I don't know what settings you have made.

If you say it is slow slow we have no way to find out.
If you want us to compare and find out, you need to present all the buildable code.

lovyan03, 2023-03-24:
By the way, since LVGL does differential rendering, I don't think it makes sense to compare FPS values with benchmarks that do full screen rendering.

JohnTaipei, 2023-03-26:
@lovyan03 
OK , i see
Thanks

closed 2023-05-02 by github-actions[bot]