lovyan03/LovyanGFX #247: Request of supporting FT5436 touch screen controller

gyzhaoyimei, 2022-06-20:
Can you please add the driver for FT5436 touch screen controller, it is based on I2C bus
https://www.good-display.cn/companyfile/543.html
Thank you very much

lovyan03, 2022-06-20:
Hello @gyzhaoyimei 
Perhaps Touch_FT5x06 would work.
I don't have a FT5436 so you could try it and let me know your results.

gyzhaoyimei, 2022-06-20:
I have tried using FT5x06. The touch does not work properly. FT6236 can work normally

lovyan03, 2022-06-20:
@gyzhaoyimei 
That is unfortunate.
I do not have it and cannot verify it.
I can't do anything at this time because I don't have enough information.

lovyan03, 2022-06-21:
By the way, what MCU are you using? Maybe there is a problem with the I2C implementation of LovyanGFX.
I am in the process of finding and correcting a problem with the touch panel not working properly on the ESP32-S3.

lovyan03, 2022-06-25:
@gyzhaoyimei 
I tried FT5436 and saw it. It is working fine.

Please make sure your settings and wiring are correct.
You could also try to see if you can communicate using the Wire class without LovyanGFX.

gyzhaoyimei, 2022-06-28:
@lovyan03 
Thanks. I found that I2C did not work, and the pin line sequence of ft6236 and ft5436 was different，
I'll try again。

closed 2022-06-30 by gyzhaoyimei