lovyan03/LovyanGFX #249: How can eliminate flickering display

MisterAG, 2022-06-20:
Hi,

I'm trying to make a gauge with a moving needle, using a gc9a01 round display (240x240 resolution).
I got my code up and running and I get good performance. However I see a terrible flickering on the display.
When I let it run full speed I get about 506 FPS (mydelay=0), when I set mydelay to 16ms I get as I expect about 63 FPS.
If you look at the two videos below you will see what I mean. With the flickering.

How my code basically works: I divided the screen in 5 parts, each part is a Sprite. The outer sprites (the white, red, blue, green part) are written once to the LCD. The inner Sprite (magenta color) is the only part that gets rewritten to the LCD.

Is there way to avoid the flickering? I admit I'm not the best developer, so I might have made some rookie mistake.

[https://www.youtube.com/watch?v=CokpqxwYtk4](url)
[https://www.youtube.com/watch?v=4kmfR0G6qcc](url)

` 
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

static LGFX_Sprite icons;

extern const unsigned short needle[];
extern const unsigned short vumeter1a[];
extern const unsigned short vumeter1b[];
extern const unsigned short vumeter1c[];
extern const unsigned short vumeter1d[];
extern const unsigned short vumeter1e[];

 

class LGFX : public lgfx::LGFX_Device
{
lgfx::Panel_GC9A01     _panel_instance;
lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス
//lgfx::Light_PWM     _light_instance;
//lgfx::Touch_XPT2046     _touch_instance;

public:
  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。
      cfg.spi_host = VSPI_HOST;     // 使用するSPIを選択  (VSPI_HOST or HSPI_HOST)
      cfg.spi_mode = 0;             // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write = 80000000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_read  = 16000000;    // 受信時のSPIクロック
      cfg.spi_3wire  = false;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock   = true;        // トランザクションロックを使用する場合はtrueを設定
      cfg.dma_channel = 1;          // Set the DMA channel (1 or 2. 0=disable)   使用するDMAチャンネルを設定 (0=DMA不使用)
      cfg.pin_sclk = 18;            // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 23;            // SPIのMOSIピン番号を設定
      cfg.pin_miso = -1;            // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc   = 33;            // SPIのD/Cピン番号を設定  (-1 = disable)
      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。
      cfg.pin_cs           =    5;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    4;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)
      cfg.memory_width     =   240;  // ドライバICがサポートしている最大の幅
      cfg.memory_height    =   240;  // ドライバICがサポートしている最大の高さ
      cfg.panel_width      =   240;  // 実際に表示可能な幅
      cfg.panel_height     =   240;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  false;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = true;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // データ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  true;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

static LGFX lcd;

static uint32_t lcd_width ;
static uint32_t lcd_height;
static LGFX_Sprite img[6];
int coordinates[6][4]={{60,240,0,0},{180,80,60,0},{50,160,190,80},{130,105,60,135},{130,55,60,80},{130,55,60,80}};
static LGFX_Sprite needleSprite;

static uint32_t sec, psec;
static size_t fps = 0, frame_count = 0;

void setup(void)
{

  lcd.init();
  lcd.setRotation(0);
  lcd_width = lcd.width();
  lcd_height = lcd.height();
  lcd.setSwapBytes(true); 
  for(int t=0;t<6;t++){
   img[t].createSprite(coordinates[t][0],coordinates[t][1]);
   img[t].setSwapBytes(true);
//   img[x].fillSprite(TFT_BLACK);  
 }

img[0].fillSprite(TFT_WHITE);
img[1].fillSprite(TFT_RED);
img[2].fillSprite(TFT_BLUE);
img[3].fillSprite(TFT_GREEN);
img[4].fillSprite(TFT_MAGENTA);

//Create basic vumeter needle
needleSprite.createSprite(5,50);
needleSprite.setSwapBytes(true);
needleSprite.fillSprite(TFT_RED);
needleSprite.setPivot(1,100);
 

img[0].pushSprite(&lcd,coordinates[0][2],coordinates[0][3]);
img[1].pushSprite(&lcd,coordinates[1][2],coordinates[1][3]);
img[2].pushSprite(&lcd,coordinates[2][2],coordinates[2][3]);
img[3].pushSprite(&lcd,coordinates[3][2],coordinates[3][3]);
img[4].pushSprite(&lcd,coordinates[4][2],coordinates[4][3]);

}
 
int v = 100;  //pivot point y axis
int sa = 327;  //start angle
int aw = 72; //angle width
int tempa=sa; //temp angle to calculate with
int mymilli=millis();
int mydelay=0; // every mydelay (ms) a frame is drawn.
bool goingup=true;// needle going to the right = true

void loop(void)
{

if((millis() - mymilli)>= mydelay){
mymilli=millis();
//Draw the middle frame
img[5].pushSprite(&lcd,coordinates[4][2],coordinates[4][3]);
//FPS Calculation
++frame_count;
 sec = lgfx::millis() / 1000;
  if (psec != sec) {
   psec = sec;
    fps = frame_count;
    frame_count = 0;
  }

//Determine next angle of needle
if(goingup==true) tempa++; else tempa--;
if(tempa == sa) goingup = true;
if(tempa == (sa+aw)) goingup= false;
  int k = tempa % 360;

//prepare the next frame but don't draw it yet
img[4].pushSprite(&img[5],0,0);
img[5].setCursor(0,20);
img[5].setFont(&fonts::Font4);
img[5].setTextColor(0xFFFFFFU);
img[5].printf("FPS: %d", fps);
needleSprite.pushRotateZoom(&img[5],60, v, k, 1, 1);
 }
}`

MisterAG, 2022-06-21:
_Do you mean that there is a striped luminance fluctuation across the entire screen?_
=> yes, that's what I'm referring to. 

_If so, check to see if it still wavers when you stop drawing with delay(1000)._
=> I tested this and I can see a horizontal luminance across the screen every second. See also video below.
On the video you can also see a lot of smaller vertical 'lines' across the screen as well.

[https://www.youtube.com/shorts/-47bi8WXlds](url)

lovyan03, 2022-06-24:
I have checked the operation with my GC9A01 panel.
There seems to be no particular problem with the display.

https://user-images.githubusercontent.com/42724151/175452762-a3b81cae-4da4-41ba-a6d4-779c0f36bc9c.mp4

Even though the driver IC is the same GC9A01, there may be differences in the specifications of the panel beyond the driver.
In that case, I think you need to adjust the parameters in the initialization command set.
This cannot be confirmed without getting exactly the same one.

lovyan03, 2022-06-24:
@MisterAG 
Perhaps the communication clock is too fast ?  Try lowering it to 10 to 20 MHz as a trial.
Also, if the backlight momentarily dims during communication, the condition of the wiring may be affected.
Is there any interference between the backlight power line and the MOSI?

MisterAG, 2022-06-24:
Hi, 

thank you for your testing. I'm starting to think the wiring might be a problem. I soldered wires to the flex cable. I ordered a display with a connected PCB like you have to do some further testing. 

The strange thing however is that the MovingIcons demo runs fine without any noticable interference on the screen. So I suspected my code to be the culprit.

MisterAG, 2022-06-24:
I had a PCB laying around with the correct 0.7mm pitch, so I soldered the screen to it and tested again. 
It works now without any flickering. 

Apologies for the inconvenience.
This issue can be closed.

closed 2022-06-24 by MisterAG