lovyan03/LovyanGFX #250: misleading indentation error in miniz.c

ftab, 2022-06-23:
I'm in dangerous territory working with the unreleased esp-idf version 5. I had to make a few changes to LGFX to make it work - see them all [here](https://github.com/lovyan03/LovyanGFX/compare/develop...radiosound-com:ftab/idf5?expand=1)

One in particular is a misleading indentation error that has me confused about what it was supposed to do.

https://github.com/lovyan03/LovyanGFX/commit/e295643f3f4cbac1a02ac1d916f1bb92994127d6

The indentation makes it look like it wants braces. However, putting braces around it technically changes what it does. As I am unfamiliar with what this is supposed to do, I don't know if that's the intended behavior or not.

lovyan03, 2022-06-23:
@ftab 
thanks for your report.
I did not create miniz so I am not concerned about the quality of the code.
If the impact of the change cannot be properly assessed, then it should be ignored.

ftab, 2022-06-23:
Based on the latest version of miniz.c, it looks like I guessed wrong, the braces shouldn't have been around the whole thing.

The full macro from miniz 2.2.0 is

```c
#define TDEFL_PUT_BITS(b, l)                                       \
    do                                                             \
    {                                                              \
        mz_uint bits = b;                                          \
        mz_uint len = l;                                           \
        MZ_ASSERT(bits <= ((1U << len) - 1U));                     \
        d->m_bit_buffer |= (bits << d->m_bits_in);                 \
        d->m_bits_in += len;                                       \
        while (d->m_bits_in >= 8)                                  \
        {                                                          \
            if (d->m_pOutput_buf < d->m_pOutput_buf_end)           \
                *d->m_pOutput_buf++ = (mz_uint8)(d->m_bit_buffer); \
            d->m_bit_buffer >>= 8;                                 \
            d->m_bits_in -= 8;                                     \
        }                                                          \
    }                                                              \
    MZ_MACRO_END
```

Do you think it would be feasible to update? I tried to use the latest miniz.c release (2.2.0), but it's not a direct drop-in replacement (8 years newer code will do that sometimes)

Otherwise, maybe just fix the indentation

lovyan03, 2022-06-23:
This is only used for PNG rendering. This is based on the PNGLE library implementation and I was not involved in the selection process.
I tried upgrading miniz once before, but the binary size increased a bit and the execution speed slowed down a bit, so I decided to revert back to the original version.
Perhaps some configuration or adjustment might improve it, but I didn't really feel the need to spend my time there.
In other words, I'm not particularly interested in raising a version of this.

ftab, 2022-06-23:
Good call. I like it small and fast

closed 2022-06-24 by lovyan03