lovyan03/LovyanGFX #253: pushimage and uneven y resolution problem.

MisterAG, 2022-06-26:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP-WROOM-32]
- Panel Driver IC:    [gc9a01]
- Bus type:           [SPI]
- LovyanGFX version:  [v0.4.17|
- Platform version:   [Arduino IDE 1.8.14]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )
I need to display some images on the screen. This works with images that have an even resolution for x and y.
For images that have and uneven y resolution, the last few pixels are drawn in black and not in the correct color.
When I do a fillSprite() on a sprite with uneven y resolution and display it, there is no problem. Only when I push an image (with uneven y resolution) to a sprite.

[image]

### Expected Behavior ( 期待される動作 )
The correct colors should be displayed
### Actual Behavior ( 実際の動作 )
The last few pixels are drawn in black
### Steps to reproduce ( 再現のための前提条件 )
I create a jpg in MS paint with different resolutions (see picture) and fill it with the color yellow.
I then convert the image using:  http://rinkydinkelectronics.com/t_imageconverter565.php
Then copy and paste it into my code.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )
[mycode-to-reproduce.zip](https://github.com/lovyan03/LovyanGFX/files/8986316/mycode-to-reproduce.zip)

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2022-06-26:
hello,

although it's not exactly adressing the odd/even situation, maybe `setClipRect()` can be of some help for your issue?

you can apply it before drawing, and eventually get rid of sprites the  e.g. 

```C++
tft.setClipRect( 0, 141, 148, 67 );
tft.pushImage(  0,141, 148, 67, test148x67);
```

lovyan03, 2022-06-26:
@MisterAG 
The number of data in the array appears to be insufficient.
Compare the number of elements with the actual number of data.

lovyan03, 2022-06-26:
@MisterAG 
I used the conversion tool at the URL you provided.
There appears to be a bug that prevents data less than a multiple of 16 from being output.

MisterAG, 2022-06-26:
Could you recommend a decent image converter?

closed 2022-06-26 by lovyan03