lovyan03/LovyanGFX #256: Not able to work with touch part of User settings example

zuhairmst, 2022-07-04:
I am using ESP32-S3-Devkit-C1 with an LCD having panel controller IC ILI9488 and touch controller IC GT911. Using user setting example to get started. Any modification to the code make touch of the LCD not working. If I don't make any change, then it work fine. Simple change like opening serial port creates issue. Guide me regarding this issue, how to use this example properly. If you have any other example related to touch part then plz share it.

Here is the code.
#include <Arduino.h>
#include<Wire.h>
//#include<TFT_eSPI.h>
#include <SPI.h>
#include "FS.h"
#include "SPIFFS.h"
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include "M3_Logo.h"
class LGFX : public lgfx::LGFX_Device
{

 lgfx::Panel_ILI9488     _panel_instance;
 lgfx::Bus_SPI           _bus_instance;   
 lgfx::Touch_GT911       _touch_instance;

public:
  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

// SPIバスの設定
      cfg.spi_host = SPI2_HOST;     // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
      cfg.spi_mode = 0;             // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write = 27000000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_read  = 16000000;    // 受信時のSPIクロック
      cfg.spi_3wire  = true;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock   = true;        // トランザクションロックを使用する場合はtrueを設定
      cfg.dma_channel = SPI_DMA_CH_AUTO; // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 12;            // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 11;            // SPIのMOSIピン番号を設定
      cfg.pin_miso = 13;            // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc   = 10;            // SPIのD/Cピン番号を設定  (-1 = disable)

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs           =    4;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    5;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      cfg.panel_width      =   320;  // 実際に表示可能な幅
      cfg.panel_height     =   480;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     2;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  true;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = true;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  false;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)
    //以下はST7735やILI9163のようにピクセル数が可変のドライバで表示がずれる場合にのみ設定してください。
   cfg.memory_width     =   320;  // ドライバICがサポートしている最大の幅
   cfg.memory_height    =   480;  // ドライバICがサポートしている最大の高さ
      _panel_instance.config(cfg);
    }

    
    { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 2047;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 2047;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = 42;   // INTが接続されているピン番号
      cfg.bus_shared = false; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 2;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// I2C
      cfg.i2c_port = 0;      // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x5D;   // I2Cデバイスアドレス番号
      cfg.pin_sda  = 1;     // SDAが接続されているピン番号
      cfg.pin_scl  = 2;     // SCLが接続されているピン番号
      cfg.freq =  400000;     // I2Cクロックを設定

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }
//*/

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

// 準備したクラスのインスタンスを作成します。
LGFX display;

void setup(void)
{
   display.init();  
   display.setTextSize((std::max(display.width(), display.height()) + 255) >> 8);
   // タッチが使用可能な場合のキャリブレーションを行います。（省略可）
   if (display.touch())
   {
     if (display.width() < display.height())
       display.setRotation(display.getRotation() ^ 1);

     // 画面に案内文章を描画します。
     display.setTextDatum(textdatum_t::middle_center);
     display.drawString("touch the arrow marker.", display.width() >> 1, display.height() >> 1);
     display.setTextDatum(textdatum_t::top_left);

     // タッチを使用する場合、キャリブレーションを行います。画面の四隅に表示される矢印の先端を順にタッチしてください。
     std::uint16_t fg = TFT_WHITE;
     std::uint16_t bg = TFT_BLACK;
     if (display.isEPD())
       std::swap(fg, bg);
     display.calibrateTouch(NULL, fg, bg, std::max(display.width(), display.height()) >> 3);
  }

  display.fillScreen(TFT_BLACK);
}

uint32_t count = ~0;
void loop(void)
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30,30,display.width()-60,display.height()-60,count*7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  long x, y;
  if (display.getTouch(&x, &y)) {
    display.fillRect(x-2, y-2, 5, 5, count*7);
  }
}

lovyan03, 2022-07-04:
Try using the develop branch version.

zuhairmst, 2022-07-05:
Thanks. Working now.

closed 2022-07-05 by zuhairmst