lovyan03/LovyanGFX #258: Use lovyangfx library to drive 8-bit parallel port screen to appear flower screen

gyzhaoyimei, 2022-07-08:
## Environment ( 実行環境 )

- MCU or Board name:  ESP-WROOM-32
- Panel Driver IC:    ILI9488
- Bus type:           Parallel8
- LovyanGFX version:  LovyanGFX @ 0.4.17
- FrameWork version:   framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
- Build Environment :   PlatformIO
（PLATFORM: Espressif 32 (4.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash）
- Operating System:   Windows 10
--FTF LCD
[image]

## Problem Description ( 問題の内容 )
The same hardware and circuit, using this library TFT_ ESPI is displayed normally, but the library lovyangfx cannot be displayed normally

### Expected Behavior ( 期待される動作 )
I hope to use this library lovyangfx to drive my screen。

### Steps to reproduce ( 再現のための前提条件 )

1.  Using TFT_ ESPI Library 
User_Setup.h is as follows
```cpp

#define USER_SETUP_INFO "User_Setup"

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   1//33  // Chip select control pin (library pulls permanently low 可以不连接
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  3//32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
```
The main functions are as follows
```cpp
#include "SPI.h"
#include "TFT_eSPI.h"

TFT_eSPI tft = TFT_eSPI();

unsigned long testText()
{
  tft.fillScreen(TFT_BLACK);
  unsigned long start = micros();
  tft.setCursor(0, 0);
  tft.setTextColor(TFT_WHITE);
  tft.setTextSize(1);
  tft.println("Hello World!");
  tft.setTextColor(TFT_YELLOW);
  tft.setTextSize(2);
  tft.println(1234.56);
  tft.setTextColor(TFT_RED);
  tft.setTextSize(3);
  tft.println(0xDEADBEEF, HEX);
  tft.println();
  tft.setTextColor(TFT_GREEN);
  tft.setTextSize(5);
  tft.println("Groop");
  tft.setTextSize(2);
  tft.println("I implore thee,");
  // tft.setTextColor(TFT_GREEN,TFT_BLACK);
  tft.setTextSize(1);
  tft.println("my foonting turlingdromes.");
  tft.println("And hooptiously drangle me");
  tft.println("with crinkly bindlewurdles,");
  tft.println("Or I will rend thee");
  tft.println("in the gobberwarts");
  tft.println("with my blurglecruncheon,");
  tft.println("see if I don't!");
  return micros() - start;
}
void setup()
{
  Serial.begin(9600);
  Serial.println("TFT_eSPI library test!");

  tft.init();
}

void loop(void)
{
  for (uint8_t rotation = 0; rotation < 4; rotation++)
  {
    tft.setRotation(rotation);
    testText();
    delay(2000);
  }
}
``` 

Screen display：
https://user-images.githubusercontent.com/24825476/177986093-4ac00e55-5ac7-4b10-8fe1-9e4ba62a2906.mp4

2.The same hardware and circuit are displayed abnormally with this LovyanGFX library。The code is as follows。
```cpp
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ILI9488 _panel_instance; 
  lgfx::Bus_Parallel8 _bus_instance; // 8位并行总线实例（仅限ESP32）
 
public:
  LGFX(void)
  {
    {                                    //设置总线控制。
      auto cfg = _bus_instance.config(); // 获取用于总线设置的结构。
  
      // 8位并行总线设置
      cfg.i2s_port = I2S_NUM_0;  // 选择要使用的I2S端口（I2S_NUM_0 or I2S_NUM_1）（使用ESP32的I2S LCD模式）
      cfg.freq_write = 20000000; // 发送时钟（最大可将20MHz、80MHz舍入为整数）
      cfg.pin_wr = 4;            // 连接WR的针脚编号
      cfg.pin_rd = 2;            // 连接RD的针脚编号
      cfg.pin_rs = 15;           // 连接RS（D/C）的针脚编号
     
      cfg.pin_d0 = 12;           // 连接D0的管脚编号
      cfg.pin_d1 = 13;           // 连接D1的管脚编号
      cfg.pin_d2 = 26;           // 连接D2的管脚编号
      cfg.pin_d3 = 25;           // 连接D3的管脚编号
      cfg.pin_d4 = 17;           // 连接D4的管脚编号
      cfg.pin_d5 = 16;           // 连接D5的管脚编号
      cfg.pin_d6 = 27;           // 连接D6的管脚编号
      cfg.pin_d7 = 14;           // 连接D7的管脚编号
      
      _bus_instance.config(cfg);              // 在总线上反映设置值。
      _panel_instance.setBus(&_bus_instance); // 把总线设置在面板上。
    }

    {                                      //设置显示面板控制。
      auto cfg = _panel_instance.config(); // 获取用于设置显示面板的结构。
      cfg.pin_cs = 1;                      // 连接CS的管脚编号（-1=disable）
      cfg.pin_rst = 3;                    // 连接RST的管脚编号（-1=disable）
      cfg.pin_busy = -1;                   // 连接BUSY的管脚编号（-1=disable）
      cfg.memory_width = 320;              // 驱动器IC支持的最大宽度
      cfg.memory_height = 480;             // 驱动器IC支持的最大高度
      cfg.panel_width = 320;               // 实际可见宽度
      cfg.panel_height = 480;              // 实际可见高度
      cfg.offset_x = 0;                    // 面板X方向偏移量
      cfg.offset_y = 0;                    // 面板Y方向偏移量
      cfg.offset_rotation = 0;             // 旋转方向值的偏移0~7（4~7上下反转）
      cfg.dummy_read_pixel = 8;            // 像素读取前虚拟读位数
      cfg.dummy_read_bits = 1;             // 读取非像素数据之前的虚拟读位数
      cfg.readable = true;                 // 可读取数据时设置为真
      cfg.invert = false;                  // 如果面板的明暗颠倒，则设置为真
      cfg.rgb_order = false;               // 如果面板中的红色和蓝色替换为真
      cfg.dlen_16bit = true;              // 在以16 bit为单位发送数据长度的面板的情况下设定为真
      cfg.bus_shared = true;               // 与SD卡共享总线时设置为真（通过drawJpgFile等进行总线控制）
     
      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance); // 设置要使用的面板。
  }
};

static LGFX lcd;                 // LGFXのインスタンスを作成。
static LGFX_Sprite sprite(&lcd); // スプライトを使う場合はLGFX_Spriteのインスタンスを作成。

void setup(void)
{
// 最初に初期化関数を呼び出します。
  lcd.init();
// 回転方向を 0～3 の4方向から設定します。(4～7を使用すると上下反転になります。)
  lcd.setRotation(1);
 
 // バックライトの輝度を 0～255 の範囲で設定します。
  lcd.setBrightness(128);

 // int8_t型、uint8_t型を使用すると、RGB332の8ビットとして扱われます。
  uint8_t blue = 0x03;
  lcd.fillRect(20, 20, 20, 20, (uint8_t)0xE0);  // 赤で矩形の塗りを描画
  lcd.fillRect(30, 30, 20, 20, (uint8_t)0x1C);  // 緑で矩形の塗りを描画
  lcd.fillRect(40, 40, 20, 20, blue);           // 青で矩形の塗りを描画

}

void loop(void)
{
  static int count = 0;
  static int a = 0;
  static int x = 0;
  static int y = 0;
  static float zoom = 3;
  ++count;
  if ((a += 1) >= 360) a -= 360;
  if ((x += 2) >= lcd.width()) x -= lcd.width();
  if ((y += 1) >= lcd.height()) y -= lcd.height();
  sprite.setPaletteColor(1, lcd.color888( 0, 0, count & 0xFF));
  sprite.setPaletteColor(2, lcd.color888( 0,~count & 0xFF, 0));
  sprite.setPaletteColor(3, lcd.color888( count & 0xFF, 0, 0));

  sprite.pushRotateZoom(x, y, a, zoom, zoom, 0);

  if ((count % 100) == 0) lcd.display(); // 電子ペーパーの場合の表示更新を 100回に一度行う
}
 
```

Screen display：
[image]

lovyan03, 2022-07-08:
```
      cfg.dlen_16bit = true;              // 在以16 bit为单位发送数据长度的面板的情况下设定为真
```
please set to false.

gyzhaoyimei, 2022-07-08:
> ```
>       cfg.dlen_16bit = true;              // 在以16 bit为单位发送数据长度的面板的情况下设定为真
> ```
> 
> please set to false.

I think this is also the problem. True or false does not affect,

lovyan03, 2022-07-08:
The area given to createSprite is too large.
Try drawing it directly like TFT_eSPI.

lovyan03, 2022-07-08:
> I guess this library does not support 8-bit parallel port?

Don't blame me for your bad usage.

lovyan03, 2022-07-09:
Oh… you are using an uninitialized sprite. There is no way that will work.

try deleting the contents of the loop and give it a try.

```
void loop() 
{
delay(100);
}
```

gyzhaoyimei, 2022-07-09:
> GPIO 2 and 12 is Bootstrapping pin GPIO 1 and 3 is UART pin
> 
> You should avoid using this pin unless you want to suffer from unexplained problems.
> 
> and, may not work properly if the clock is too fast with jumper wire connection. try lowering it to about 10 MHz.

Thank you for your reply。It can be displayed normally after changing the pin
```
 {                                    //设置总线控制。
      auto cfg = _bus_instance.config(); // 获取用于总线设置的结构。
  
      // 8位并行总线设置
      cfg.i2s_port = I2S_NUM_0;  // 选择要使用的I2S端口（I2S_NUM_0 or I2S_NUM_1）（使用ESP32的I2S LCD模式）
      cfg.freq_write = 20000000; // 发送时钟（最大可将20MHz、80MHz舍入为整数）
      cfg.pin_wr = 4;            // 连接WR的针脚编号
      cfg.pin_rd = 5;            // 连接RD的针脚编号
      cfg.pin_rs = 15;           // 连接RS（D/C）的针脚编号  1 2 3  12
     
      cfg.pin_d0 = 18;           // 连接D0的管脚编号
      cfg.pin_d1 = 13;           // 连接D1的管脚编号
      cfg.pin_d2 = 26;           // 连接D2的管脚编号
      cfg.pin_d3 = 25;           // 连接D3的管脚编号
      cfg.pin_d4 = 17;           // 连接D4的管脚编号
      cfg.pin_d5 = 16;           // 连接D5的管脚编号
      cfg.pin_d6 = 27;           // 连接D6的管脚编号
      cfg.pin_d7 = 14;           // 连接D7的管脚编号
      
      _bus_instance.config(cfg);              // 在总线上反映设置值。
      _panel_instance.setBus(&_bus_instance); // 把总线设置在面板上。
    }

    {                                      //设置显示面板控制。
      auto cfg = _panel_instance.config(); // 获取用于设置显示面板的结构。
      cfg.pin_cs = 32;                      // 连接CS的管脚编号（-1=disable）
      cfg.pin_rst = 33;                    // 连接RST的管脚编号（-1=disable）
      cfg.pin_busy = -1;                   // 连接BUSY的管脚编号（-1=disable）
      cfg.memory_width = 320;              // 驱动器IC支持的最大宽度
      cfg.memory_height = 480;             // 驱动器IC支持的最大高度
      cfg.panel_width = 320;               // 实际可见宽度
      cfg.panel_height = 480;              // 实际可见高度
      cfg.offset_x = 0;                    // 面板X方向偏移量
      cfg.offset_y = 0;                    // 面板Y方向偏移量
      cfg.offset_rotation = 0;             // 旋转方向值的偏移0~7（4~7上下反转）
      cfg.dummy_read_pixel = 8;            // 像素读取前虚拟读位数
      cfg.dummy_read_bits = 1;             // 读取非像素数据之前的虚拟读位数
      cfg.readable = true;                 // 可读取数据时设置为真
      cfg.invert = false;                  // 如果面板的明暗颠倒，则设置为真
      cfg.rgb_order = false;               // 如果面板中的红色和蓝色替换为真
      cfg.dlen_16bit = false;              // 在以16 bit为单位发送数据长度的面板的情况下设定为真
      cfg.bus_shared = true;               // 与SD卡共享总线时设置为真（通过drawJpgFile等进行总线控制）
     
      _panel_instance.config(cfg);
    }

```
Many thanks for your help.

closed 2022-07-10 by gyzhaoyimei