lovyan03/LovyanGFX #261: crash when using psram and a sprite

paxo-rch, 2022-07-14:
## Environment

- MCU or Board name: ESP32 Wrover B
- Panel Driver IC:    ILI9488
- Bus type:           SPI
- LovyanGFX version:  [v0.4.17|
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment :   ArduinoIDE
- Operating System:   Linux

## Problem Description
Hello, I tried to use your library in my project. The problem I had is a problem between the psram and the sprites. When allocating a sprite after using setPsram(true), I received the following error: "Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled."
I specify that I used an example of your library (MovingCircles).
I can use the psram without problem at the beginning of the setup but at the end, any allocation in the psram seems to fail.

### Expected Behavior ( 期待される動作 )
"Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled."

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1. allocate a sprite or just memory in the psram after the exemple code in setup
2. that's all

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
void setup(void)
{
  Serial.begin(115200);
  
  void* ptr = ps_malloc(1000); // works
  
  lcd.init();

  if (lcd.width() < lcd.height())
  {
    lcd.setRotation(lcd.getRotation() ^ 1);
  }
  lcd_width = lcd.width();
  lcd_height = lcd.height();
  obj_info_t *a;
  for (std::uint32_t i = 0; i < obj_count; ++i) {
    a = &objects[i];
    a->color = (uint32_t)rand() | 0x808080U;
    a->x = rand() % lcd_width;
    a->y = rand() % lcd_height;
    a->dx = ((rand() & 3) + 1) * (i & 1 ? 1 : -1);
    a->dy = ((rand() & 3) + 1) * (i & 2 ? 1 : -1);
    a->r = 8 + (i & 0x07);
  }

  uint32_t div = 2;
  for (;;) {
    sprite_height = (lcd_height + div - 1) / div;
    bool fail = false;
    for (std::uint32_t i = 0; !fail && i < 2; ++i)
    {
      sprites[i].setColorDepth(lcd.getColorDepth());
      sprites[i].setFont(&fonts::Font2);
      sprites[i].setPsram(true);
      fail = !sprites[i].createSprite(lcd_width, sprite_height);
    }
    if (!fail) break;
    for (std::uint32_t i = 0; i < 2; ++i)
    {
      sprites[i].deleteSprite();
    }
    ++div;
  }
  void* ptr = ps_malloc(1000); // does not work
  lcd.startWrite();
}

...
```

thank you in advance, I would really like to use this library instead of tft_espi for a question of speed, but with tft_espi the bug does not appear

lovyan03, 2022-07-14:
Hello @paxo-rch 
I don't see any information on which GPIOs you are using.
Are you using GPIO16,GPIO17?
If so, change those pins to other pins as they are necessary for PSRAM operation.

paxo-rch, 2022-07-14:
I just use gpio 25 and 22 in addition to the example.

lovyan03, 2022-07-14:
?????
are you use autodetect ?
your setting is gone ?
I don't understand.

lovyan03, 2022-07-14:
- Your sample code will compile error because the variable name ptr is duplicated in the second ps_malloc.

 - I have enabled PSRAM on M5StackCore2 and it works fine.

I am convinced that you have the wrong settings.

paxo-rch, 2022-07-15:
ok, I do this now

paxo-rch, 2022-07-15:
code: [https://docs.google.com/document/d/1KCVpsQDI6cMY-z6a4Oxl5Ai2LSC6trpI/edit?usp=sharing&ouid=116451164311963326444&rtpof=true&sd=true](url)
(include the configuration)

I specify that I have just reinstalled the library and completely redone the configuration from this tutorial: [https://www.youtube.com/watch?v=IPCvQ4o_WP8](url)

crash report:
```[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
before create sprite
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008d66f  PS      : 0x00060330  A0      : 0x8008dc27  A1      : 0x3ffb1ec0  
A2      : 0x3f800000  A3      : 0x0004b010  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x3f800010  A11     : 0x00000000  A12     : 0x3f800014  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x400890c1  LEND    : 0x400890d1  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d66f:0x3ffb1ec0 0x4008dc24:0x3ffb1ee0 0x400817f9:0x3ffb1f00 0x400d3290:0x3ffb1f20 0x400d2db7:0x3ffb1f40 0x400d18d8:0x3ffb1f60 0x400d1d96:0x3ffb1f80 0x400d8eb6:0x3ffb1fb0 0x4008a9d5:0x3ffb1fd0

Rebooting...
```

paxo-rch, 2022-07-15:
if i use ps_malloc before the sprite allocation, it works fine, but the sprite allocation crash. I put Serial.println() to act as a debugger to show where the code crashed.

lovyan03, 2022-07-15:
[image]

hey. you using GPIO17.

closed 2022-07-15 by lovyan03