lovyan03/LovyanGFX #263: Init LovyanGFX without touch panel enabled

xlla, 2022-07-16:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

Can we init LovyanGFX without touch panel for M5paper?

**Is your feature request related to a problem? Please describe.**
for power consume reduce, I want interface just by hardware buttons.

**Describe the solution you'd like**
add an optional parameter or global macro to tell underlay does not activate/init touch panel. 

**Describe alternatives you've considered**
add an explicit method to init touch panel.

**Additional context**

lovyan03, 2022-07-17:
hello @xlla 

Conclusion first.
After completing the initialization process, please do the following

```
 /// The instance name `gfx` should be rewritten to match your environment.
  gfx.init();
  gfx.touch()->sleep();
```

Since you are using M5Paper, I assume that the touch controller GT911 is activated simply by energizing it.
In other words,  Before GFX performs the initialization process, the touch controller GT911 has already started working, Even if you omit the initialization process of the GT911, your wish will not be fulfilled.

To achieve the desired result, the sleep command should be sent to GT911 after startup.

…Note that in order to wake up the GT911, the INT pin of the GT911 must be set to HIGH, but this is not possible with the M5Paper because the INT pin is connected to a read-only GPIO.
Keep in mind that once you put it to sleep, the only way to do this is to reset the entire system.

xlla, 2022-07-17:
@lovyan03 thank you, this is what I want.

if I put the device into deep sleep mode after some tasks, then when it wake up again, the touch panel should wake up again too? since it lost power while in deep sleep mode, the INT pin goes LOW, when device wake up, the INT pin goes HIGH.

by the way, is there any example for 4 bit image render? I found all examples using RGB565 2 bytes array to store images, it is a waste and wearing flash for m5paper.

lovyan03, 2022-07-17:
@xlla 
Sorry, I have not experimented with it in detail, so I have no idea. Please try it yourself and check.

Because LovyanGFX focuses on compatibility with color LCDs, it does not have the ability to render 4-bit images directly to EPDs without conversion.
If you think it's waste and wearing flash, I suggest you use the official M5Stack M5EPD.

xlla, 2022-07-18:
all right, no problem here.

closed 2022-07-18 by xlla