lovyan03/LovyanGFX #264: Is parallel ST7789 1.9" 320x170 TFT supported

patfelst, 2022-07-17:
Hello Lovyan,

Lilygo have just release a new ESP32-S3 with a parallel interface 1.9" TFT with resolution 320x170. I have not seen this display on dev board before.
https://www.lilygo.cc/products/t-display-s3

They have an adapted version of TFT_eSPI to drive it. Do you know if your library will support this display and if yes, how to configure it? The TFT_eSPI code can be seen here
https://github.com/Xinyuan-LilyGO/T-Display-S3

Thank you!

lovyan03, 2022-07-17:
Hello @patfelst .
If it is the ST7789 with parallel connection, the operation has already been verified.
I think it will work as long as you set it up using Bus_Parallel8.

patfelst, 2022-07-18:
ok thanks. I have a few on order so I will try with Bus_Parallel8 when I receive in a week or so.

closed 2022-07-18 by patfelst