lovyan03/LovyanGFX #270: Confused about color handling

marcelstoer, 2022-07-26:
I have this simple example below. I am confused as to why the rectangle in the lower right corner is blue rather than red. It's as if RGB were interpreted as BGR.

```cpp
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

static LGFX display;

int32_t w, h;
void setup(void) {
  display.init();

  w = display.width();
  h = display.height();

  log_i("Display width: %d, height: %d", w, h);
  log_i("Rotation: %d", display.getRotation());

  int8_t l = 10;
  display.fillRect(0, 0, l, l, TFT_WHITE);
  display.fillRect(w - l, h - l, l, l, TFT_RED);

  display.setTextColor(TFT_WHITE);
  display.setFont(&fonts::FreeSans9pt7b);
  display.drawString("0", 20, 20);
  display.drawString(String(w) + "/" + String(h), w - 70, h - 40);
}
```
[image]

lovyan03, 2022-07-27:
You need to adjust the setting items in the panel. `rgb_order`

marcelstoer, 2022-07-27:
I have `cfg.rgb_order = true;` which I confirmed with `log_i("Color mode RGB: %d", display.panel()->config().rgb_order);`.

> 08:01:01.031 > [I][main.cpp:26] setup(): Color mode RGB: 1

From what I understand i.e. based on the Google translate of the Japanese comments this should give me RGB, no?

lovyan03, 2022-07-27:
Rewrite and try before asking me questions.

marcelstoer, 2022-07-27:
Of course I did. I am asking because the code is not clear to me and the comments are in Japanese. I can set `cfg.rgb_order = false;` which gives me red but this is contrary to what I understand from the comments.

It looks like `rgb_order = false` means "do NOT swap red and blue"?

lovyan03, 2022-07-27:
パネルの仕様によって違うので何とも言えませんね。

closed 2022-07-27 by lovyan03