lovyan03/LovyanGFX #279: Would you support LT7685 SPI mode LCD controller with LovyanGFX?

CBLeong1985, 2022-08-19:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

### Device Name (デバイスの名称・型番等)
WKS70WV009-WCT

### URL of Device Specifications document 
(仕様書等のURL)
[WKS70WV009-WCT_SPEC_V0.1.pdf](https://github.com/lovyan03/LovyanGFX/files/9379103/WKS70WV009-WCT_SPEC_V0.1.pdf)
[LT768_AP-Note_V11_CH.pdf](https://github.com/lovyan03/LovyanGFX/files/9379109/LT768_AP-Note_V11_CH.pdf)
[LT768x_DS_V21_CH.pdf](https://github.com/lovyan03/LovyanGFX/files/9379110/LT768x_DS_V21_CH.pdf)

### URL of the store where we can purchase (商品を購入できるURL)
https://item.taobao.com/item.htm?spm=a1z0k.7628869.0.0.150c37de1X8Jeu&id=584174745676&_u=t2dmg8j26111

CBLeong1985, 2022-11-24:
https://user-images.githubusercontent.com/62048568/203723504-8b40e0ab-2224-4478-928e-1b9cf35574e4.mp4

Done porting to 7 inch LT7686 spi mode with ESP32.

tobozo, 2022-12-26:
sorry for the misfire, are you willing to share the configuration for this device?

CBLeong1985, 2022-12-27:
[Panel_LT7686.zip](https://github.com/lovyan03/LovyanGFX/files/10305684/Panel_LT7686.zip)

Sure, i try to follow "Panel_RM68120.cpp" modify the code, so i just implement the SPI interface only. The rotation function also no implement. Hope someone can combine the function. Thanks.

class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_LT7686  _panel_instance;
    lgfx::Bus_SPI       _bus_instance;
    // lgfx::Light_PWM     _light_instance;
    lgfx::Touch_FT5x06  _touch_instance;

public:
    LGFX(void)
    {
        {
            auto cfg = _bus_instance.config();
            cfg.spi_host    = VSPI_HOST;
            cfg.spi_mode    = 3;
            cfg.freq_write  = 79000000;
            cfg.freq_read   = 39000000;
            cfg.spi_3wire   = false;
            cfg.use_lock    = true;
            cfg.dma_channel = SPI_DMA_CH_AUTO;
            // cfg.dma_channel = 1;
            cfg.pin_sclk    = GPIO_NUM_21;
            cfg.pin_mosi    = GPIO_NUM_22;
            cfg.pin_miso    = GPIO_NUM_18;
            cfg.pin_dc      = -1;
            _bus_instance.config(cfg);
            _panel_instance.setBus(&_bus_instance);
        }
        {
            auto cfg = _panel_instance.config();
            cfg.pin_cs      = GPIO_NUM_23;
            cfg.pin_rst     = GPIO_NUM_19;
            cfg.pin_busy    = -1;
            cfg.panel_width         = 800;
            cfg.panel_height        = 480;
            cfg.offset_x            = 0;  // パネルのX方向オフセット量
            cfg.offset_y            = 0;  // パネルのY方向オフセット量
            cfg.offset_rotation     = 0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
            cfg.dummy_read_pixel    = 8;  // ピクセル読出し前のダミーリードのビット数
            cfg.dummy_read_bits     = 1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
            cfg.readable            = false;  // データ読出しが可能な場合 trueに設定
            cfg.invert              = false;  // パネルの明暗が反転してしまう場合 trueに設定
            cfg.rgb_order           = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
            cfg.dlen_16bit          = false;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
            cfg.bus_shared          = false;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

    { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 799;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 479;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = GPIO_NUM_39;   // INTが接続されているピン番号
      cfg.bus_shared = false; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// I2C接続の場合
      cfg.i2c_port = 1;      // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
      cfg.pin_sda  = GPIO_NUM_4;     // SDAが接続されているピン番号
      cfg.pin_scl  = GPIO_NUM_5;     // SCLが接続されているピン番号
      cfg.freq = 400000;     // I2Cクロックを設定

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

CBLeong1985, 2022-12-27:
In "..\LovyanGFX\src\lgfx\v1_init.hpp" file also need add this

#include "v1/panel/Panel_LT7686.hpp"

tobozo, 2022-12-27:
I have pushed the changes on a [new branch](https://github.com/lovyan03/LovyanGFX/tree/LT7686) and left some annotations in the code (look for TODO).

I tried to reorganize the code to stick with the LGFX panel design pattern but I was only be able to test compilation, so I may also have created bugs, please confirm I didn't break anything :)

- removed `#define` statements, created protected class members
- added `_waitForStatus()` and `_waitForReg()` to prevent endless loops
- refactored some redundant code blocks

> The rotation function also no implement. Hope someone can combine the function.

can't do that for you as I don't have a display to test with, but I found some [documentation](https://raw.githubusercontent.com/b1b2b3b4b5b6/lt768_ad/master/LT7680_DS_V12B_CH.pdf#page=42) for a different model of the same brand, rotation registers are 02h and 12h, maybe it can be used as a start?

CBLeong1985, 2023-01-12:
Now i using LT7381 LCD controller interface is parallel 16 with ESP32-S3. I using "writeData(uint32_t data, uint_fast8_t bit_length)" in "Bus_Parallel16.cpp" everything is ok. I hope change to "writeBytes(const uint8_t* data, uint32_t length, bool dc, bool use_dma)" because this api got DMA i think is more faster than "writeData(uint32_t data, uint_fast8_t bit_length)" but i am fail. Anyone know how to do it? Thanks.

```cpp 

void Panel_LT7381::writePixels(pixelcopy_t* param, uint32_t length, bool use_dma)
{
	uint32_t xpos = _xpos;
	uint32_t ypos = _ypos;

	bool flg_setpos = false;
	if(xpos > _xe) { flg_setpos = true; xpos = _xs; ++ypos; }
	if(ypos > _ye) { flg_setpos = true; ypos = _ys;         }
	if(flg_setpos)
	{
		_set_write_pos(xpos, ypos);
		writeCommand(0x04, 1);
	}

	FlipBuffer _flip_buffer;
	auto dmabuf = _flip_buffer.getBuffer(length * (param->dst_bits >> 3));
	param->fp_copy(dmabuf, 0, length, param);
	// This is NG
	// _bus->writeBytes((uint8_t*)dmabuf, length * param->dst_bits, true, true);
	// This is OK
	for(uint32_t i = 0; i < (length * param->dst_bits >> 3) / 2; i++)
	{
		writeData((dmabuf[i * 2 + 0] << 0) + (dmabuf[i * 2 + 1] << 8), 2);
	}

	return;
}

```

tobozo, 2023-01-17:
I ordered two units but the package was lost and I got refunded.

It would probably be a bad idea to renew the order just before the Chinese New Year so I'll wait a few weeks.

closed 2023-02-23 by github-actions[bot]