lovyan03/LovyanGFX #280: Setting custom color palette

loxai, 2022-08-21:
Hi (to tobozo, lovyan or any other welcomed expert :)

I'm using the lib on an m5stack. I've been working on a large project and bumping into memory problems.
Up until now, I have been using a framebuffer at 320x240x16bits to do fast graphics. But that's a good chunk of mem.
I tried using begin/end transactions, but I always get flicker. maybe I'm not doing it right (I might open a separate issue, after closing this one).

if I do a 320x240x4bit buffer, I'm saving a lot of mem and still have enough colors, but I don't see a way to set the palette (ie. a custom palette for 2 or 4 bit buffer). is there a way to set an rgb array as palette for a sprite or the LGFX_Device (currently I have the latter at 16bit, while testing sprite/buffer at 2 or 4 (only get color at 8)?

Kind regards

m1cr0lab, 2022-08-22:
Hi, @loxai,

Take a cue from these examples:
- [1_simple_use.ino](https://github.com/lovyan03/LovyanGFX/blob/2b642b78833fba0748d8e9f44ca88f009abf5315/examples/HowToUse/1_simple_use/1_simple_use.ino#L293-L304)
- [RotatedZoomSample.ino](https://github.com/lovyan03/LovyanGFX/blob/a31438b4929f71c3eb3159398f31cd1f504b1a72/examples/Sprite/RotatedZoomSample/RotatedZoomSample.ino)

You can also define your palette from an array (16-bit colors here):

```cpp
uint16_t const constexpr PALETTE[] = {

    0x220c, // hsl(210,  50,  25) | #204060
    0x43f7, // hsl(210,  50,  50) | #407fbf
    0x8dfd, // hsl(210,  75,  75) | #8fbfef
    0xffff  // hsl(210, 100, 100) | #ffffff

};

sprite.createPalette(PALETTE, 4)
```

On the other hand, if you don't want to be limited by the number of colors, you can also consider rendering in slices (horizontal or vertical) with micro-buffers.

loxai, 2022-08-22:
thanks! I got it working using the setPaletteColor method, as mentioned in the samples

closed 2022-08-22 by loxai