lovyan03/LovyanGFX #282: The path to ESP-IDF version 5

martinberlin, 2022-08-24:
It seems the nice people from Espressif has made our lives miserable with some breaking changes:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/index.html

So far I had also trouble adapting my (mostly SPI) epaper component to build in this [new version of IDF](https://github.com/martinberlin/cale-idf/compare/develop...idf_v5?expand=1). 

This is the list that I get so far when trying to compile Lovyan GFX in **develop** branch and I can help with ideas / testing so it will build also in this version:

- [ ] /platforms/esp32/common.hpp:28:10: fatal error: driver/gpio.h: No such file or directory
      This simply because it does not find gpio.h unless  "REQUIRES driver" is also present in the CMakeLists **idf_component_register**  (Easy fix)

- [ ] /v1/panel/../Panel.hpp:160:19: error: 'buf' may be used uninitialized [-Werror=maybe-uninitialized]
 This one I'm not sure why it is but I will research a bit more.
 
 Will be updating this with new findings soon. 
 Like always very grateful to have this component. And I would like to make it build and flash also in this new version of IDF.

martinberlin, 2022-08-28:
Impressive update @lovyan03 thanks a lot!

Only remaining thing that I had to comment in order to get it built was: 

```
../components/lovyanGFX/src/lgfx/v1/lgfx_filesystem_support.hpp: At global scope:
../components/lovyanGFX/src/lgfx/v1/lgfx_filesystem_support.hpp:60:13: error: template-id not allowed for destructor
   60 |     virtual ~LGFX_FILESYSTEM_Support<Base>()
```

I don't get it, now then on GCC 11, you can't add a destructor in a Template?
How to do it then?

lovyan03, 2022-08-29:
@martinberlin 
Thanks for letting me know!

I removed the destructor from the template class.
Let me know if you still have problems.

martinberlin, 2022-08-29:
Hi @lovyan03 
Now it builds correctly without any problem in IDF v5. Good work!

dizcza, 2022-09-10:
I'm facing the same issue. LGFX is a dependency for https://github.com/sukesh-ak/LVGL8-WT32-SC01-IDF and my hands are dirty fixing the LGFX issues regarding ESP-IDF v5.x. I've already fixed the abovementioned problem with `miniz.c`, missing required components, missing `esp_timer.h` includes, but still don't see light at the end of the tunnel.

Should I switch to the develop branch? It looks broken according to CI.

lovyan03, 2022-09-13:
@dizcza 
I have just readjusted the develop branch. I think it should build now.
There seems to be a build error in IDFv4.3.1 of ESP32-S2, but this should not affect this case.

dizcza, 2022-09-13:
@lovyan03 
thanks! The developer branch works now with ESP-IDF v5.1-dev-96-gafc6fba7f5-dirty

dizcza, 2022-09-13:
It outputs a number of annoying warnings though:
```
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:4:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.hpp: In constructor 'lgfx::v0::Panel_GDEW0154M09::Panel_GDEW0154M09()':
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.hpp:52:20: warning: implicitly-declared 'constexpr lgfx::v0::range_rect_t& lgfx::v0::range_rect_t::operator=(const lgfx::v0::range_rect_t&)' is deprecated [-Wdeprecated-copy]
   52 |       _range_new = _range_old;
      |                    ^~~~~~~~~~
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:1:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/../lgfx_common.hpp:1634:5: note: because 'lgfx::v0::range_rect_t' has user-provided 'lgfx::v0::range_rect_t::range_rect_t(const lgfx::v0::range_rect_t&)'
 1634 |     range_rect_t(const range_rect_t& rhs) : horizon(rhs.horizon), vertical(rhs.vertical) {}
      |     ^~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp: In member function 'virtual void lgfx::v0::Panel_GDEW0154M09::post_init(lgfx::v0::LGFX_Device*, bool)':
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:26:18: warning: implicitly-declared 'constexpr lgfx::v0::range_rect_t& lgfx::v0::range_rect_t::operator=(const lgfx::v0::range_rect_t&)' is deprecated [-Wdeprecated-copy]
   26 |     _range_new = _range_old;
      |                  ^~~~~~~~~~
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:1:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/../lgfx_common.hpp:1634:5: note: because 'lgfx::v0::range_rect_t' has user-provided 'lgfx::v0::range_rect_t::range_rect_t(const lgfx::v0::range_rect_t&)'
 1634 |     range_rect_t(const range_rect_t& rhs) : horizon(rhs.horizon), vertical(rhs.vertical) {}
      |     ^~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp: In static member function 'static void lgfx::v0::Panel_GDEW0154M09::display(lgfx::v0::PanelCommon*, lgfx::v0::LGFX_Device*, int32_t, int32_t, int32_t, int32_t)':
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:302:26: warning: implicitly-declared 'constexpr lgfx::v0::range_rect_t& lgfx::v0::range_rect_t::operator=(const lgfx::v0::range_rect_t&)' is deprecated [-Wdeprecated-copy]
  302 |     me->_range_old = me->_range_new;
      |                          ^~~~~~~~~~
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_GDEW0154M09.cpp:1:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/../lgfx_common.hpp:1634:5: note: because 'lgfx::v0::range_rect_t' has user-provided 'lgfx::v0::range_rect_t::range_rect_t(const lgfx::v0::range_rect_t&)'
 1634 |     range_rect_t(const range_rect_t& rhs) : horizon(rhs.horizon), vertical(rhs.vertical) {}
      |     ^~~~~~~~~~~~
[850/1071] Building CXX object esp-idf/LovyanGFX/CMakeF...df_LovyanGFX.dir/src/lgfx/v0/panel/Panel_IT8951.cpp.obj/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_IT8951.cpp: In static member function 'static void lgfx::v0::Panel_IT8951::display(lgfx::v0::PanelCommon*, lgfx::v0::LGFX_Device*, int32_t, int32_t, int32_t, int32_t)':
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_IT8951.cpp:672:26: warning: implicitly-declared 'constexpr lgfx::v0::range_rect_t& lgfx::v0::range_rect_t::operator=(const lgfx::v0::range_rect_t&)' is deprecated [-Wdeprecated-copy]
  672 |     me->_range_old = me->_range_new;
      |                          ^~~~~~~~~~
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/Panel_IT8951.cpp:1:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v0/panel/../lgfx_common.hpp:1634:5: note: because 'lgfx::v0::range_rect_t' has user-provided 'lgfx::v0::range_rect_t::range_rect_t(const lgfx::v0::range_rect_t&)'
 1634 |     range_rect_t(const range_rect_t& rhs) : horizon(rhs.horizon), vertical(rhs.vertical) {}
      |     ^~~~~~~~~~~~
[1065/1071] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.objIn file included from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/lvgl/src/core/lv_obj.h:140,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/lvgl/lvgl.h:35,
[... cut ...]
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/main/main.cpp:16:
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:766:43: note: declared here
  766 |     [[deprecated("use pushPixels")]] void pushColors(const uint16_t* data, int32_t len, bool swap) { startWrite(); writePixels(data, len, swap); endWrite(); }
      |                                           ^~~~~~~~~~
```
Something to be fixed on a rainy evening.

tobozo, 2022-09-13:
`lgfx::v0` should probably be filtered out with esp-idf 5.0, maybe with a conditional `list(REMOVE_ITEM)`in the CMakeLists.txt ?

lovyan03, 2022-09-17:
I have made the adjustment again. Perhaps this will eliminate the warning.

sukesh-ak, 2022-12-02:
@lovyan03 
Since IDF 5.0 just got released, do you have any timeline on these updates in release branch?

sukesh-ak, 2022-12-07:
@tobozo why does it say beta?
IDF 5.0 was released few days ago.

tobozo, 2022-12-07:
> workflow CI has been updated v5.0-beta1 => v5.0

@sukesh-ak it says v5.0

sukesh-ak, 2022-12-07:
> > workflow CI has been updated v5.0-beta1 => v5.0
> 
> @sukesh-ak it says v5.0

Sorry for the confusion :)

closed 2022-12-25 by github-actions[bot]