lovyan03/LovyanGFX #283: Extend ram access for esp32 for larger sprite buffers

XanCraft21, 2022-09-04:
Hello, i don’t know if anyone else already posted this, but i love this library because of its enhanced performance and display support, but i have a request about the sprite engine. With esp32 i can use the sprite engine to make a full frame buffer for displays up to a resolution of 240x240 or 240x230, but if i go any higher, the sprite doesn’t load. I would really like to make a full display sprite buffer with a 240x320 resolution, with for esp32 only uses a third of the ram (not including wifi functions). If this were applied for an esp32 s3 chip the maximum size is estimated to be 320x480 using two thirds of the available ram. I know this is possible with a psram chip, but those are generally slower than the built-in on-chip ram.

If it is possible to increase the maximum sprite size limit to achieve this goal, i would gladly appreciate that. Thank you for your help.

tobozo, 2022-09-05:
Hi, thanks for your feedback :+1: 

Sprites use DRAM for faster transfer, and the amount is limited.

[image]

[source](https://blog.espressif.com/esp32-programmers-memory-model-259444d89387)

XanCraft21, 2022-09-05:
Ok. Do you know which compatible controller will allow the largest sprite buffer possible? And will it be possible to modify the ram usage to allow larger buffer sizes (this is just a rough estimate, but use SRAM2 for the buffer for the sprite, take the equation 240 x 320 x 2 for a 2.8 inch display with 16 bit color for using more than 3 quarters of the ram, leaving the rest for general use, depending on how that affects the controller’s performance.)? As i do not fully understand how ram usage works on higher end controllers with multiple or split banks, i just know total memory allocation for all available ram. Thank you.

tobozo, 2022-09-06:
If ESP32 psram is too slow for your expectations maybe try with the ESP32S3's faster psram along with a parallel display?

XanCraft21, 2022-09-17:
Hmm, i will look into that. I currently do not have an esp32 s3 but i do plan to get one in the future. Thanks for the tip, i will come back later if i need anything else.

lovyan03, 2022-09-17:
The LGFX_Sprite implementation requires allocating contiguous memory, so it cannot allocate the maximum capacity in situations where the free memory is fragmented.

I am considering introducing a new framebuffer function that works by allocating memory in fragments.
I do not know yet when this will be possible, but if successful, it may be possible to create a 320x240 x16bpp buffer with ESP32.

XanCraft21, 2022-09-17:
Ok. I will just wait and be patient about this. Thank you for the help.

XanCraft21, 2022-09-26:
Accidentally closed this for the wrong reason. Sorry about that.

closed 2022-09-27 by XanCraft21

bmedici, 2023-07-27:
Same here: trying to create a framebuffer of 240x240 fails (createSprite(à returns false/nullptr).
What's the correct way to implement such a frame-buffering to avoid flicker when updating the display ?
Thanks!

ammaree, 2023-12-20:
+99 for PSRAM support.

In some cases the speed difference SRAM vs PSRAM is a worthwhile tradeoff.