lovyan03/LovyanGFX #285: Add autodetect for Makerfabs ESP32-S3 with 480x320 parallel 3.5"

sukesh-ak, 2022-09-14:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**
Request to add autodetect for this device. This is the S3 version. S2 version is already supported by LovyanGFX I think.

### Device Name (デバイスの名称・型番等)
Makerfabs ESP32-S3

### URL of Device Specifications document (仕様書等のURL)
https://github.com/Makerfabs/Makerfabs-ESP32-S3-Parallel-TFT-with-Touch

### URL of the store where we can purchase (商品を購入できるURL)
https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html

lovyan03, 2022-09-16:
@sukesh-ak 
Device has been arranged.
Will implement upon arrival.

ammaree, 2022-11-11:
@lovyan03 
Any progress with this item. I have this specific board and [un]fortunately I am tied to ESP-IDF v5.
I would like to add GUI support either with just LGFX or LGFX + LVGL
Any assistance appreciated.

sukesh-ak, 2022-11-11:
> @lovyan03 Any progress with this item. I have this specific board and [un]fortunately I am tied to ESP-IDF v5. I would like to add GUI support either with just LGFX or LGFX + LVGL Any assistance appreciated.

You can use this config to get started with LovyanGFX
https://gist.github.com/sukesh-ak/610508bc84779a26efdcf969bf51a2d1

ammaree, 2022-11-11:
@sukesh-ak 
Thanks for the feedback, but a couple of points I need clarity.

Makerfabs-ESP32-S3-Parallel-TFT-With-Touch...
> use ILI9488 which is supported
> interconnection is parallel16 which as far as I can see is not yet supported by LGFX?

sukesh-ak, 2022-11-11:
> @sukesh-ak Thanks for the feedback, but a couple of points I need clarity.
> 
> Makerfabs-ESP32-S3-Parallel-TFT-With-Touch...
> 
> > use ILI9488 which is supported
> > interconnection is parallel16 which as far as I can see is not yet supported by LGFX?

Readme says S3 Parallel 16bit is supported.

sukesh-ak, 2022-11-11:
Btw, I am waiting for the board as well.

@ammaree 
I have the config here but not tested yet.
https://github.com/sukesh-ak/ESP32-TUX/blob/master/main/devices/conf_Makerfabs_S3_PTFT.h

lovyan03, 2022-11-11:
I apologize for the delay.
Please wait for a while for the release.

If you want to try it right away, try using the develop branch.
However, the develop branch is updated daily, so it may not work properly at times.

sukesh-ak, 2022-11-24:
@lovyan03 

I just got this device with ESP32-S3 16bit Parellel 
Display: ili9488 
Touch: FT6236

https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html

I got the display working but `touch is not working` yet with LovyanGFX.

They have an Arduino sample [here](https://github.com/Makerfabs/Makerfabs-ESP32-S3-Parallel-TFT-with-Touch/tree/main/firmware/SD16_3.5) where touch is working. The sample uses LovyanGFX for display but not for touch. Can you please take a look.

My touch configuration below which is not working
```c++
    lgfx::Touch_FT5x06      _touch_instance;

    { 
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;
      cfg.x_max      = TFT_WIDTH-1;
      cfg.y_min      = 0;  
      cfg.y_max      = TFT_HEIGHT-1;
      cfg.pin_int    = 40;  
      cfg.bus_shared = true; 
      cfg.offset_rotation = 0;

      cfg.i2c_port = 0;     
      cfg.i2c_addr = 0x38;
      cfg.pin_sda  = 38;   
      cfg.pin_scl  = 39;   
      cfg.freq = 400000;  

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  
    }

```

sukesh-ak, 2022-11-24:
I tried this from your develop branch but doesn't work either

https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx_user/LGFX_ESP32S2_MakerabsParallelTFTwithTouch.hpp

sukesh-ak, 2022-11-24:
I got this working. Thanks.

ESP32-S3 16bit Parallel
Display: ili9488
Touch: FT6236

https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html

lovyan03, 2022-11-24:
Have you used LGFX_AUTODETECT ?

sukesh-ak, 2022-11-24:
> Have you used LGFX_AUTODETECT ?

No, I don't need autodetect in this case so just created the entire definition.

lovyan03, 2022-11-24:
The title of this Issue is `Add autodetect for Makerfabs ESP32-S3 with 480x320 parallel 3.5`. 
I have reflected it in the develop branch.
The definition file you just tried is for ESP32-S2, so I don't think it will work.

sukesh-ak, 2022-11-24:
I use autodetect for other samples. That's why I opened this request.

But for this project, I need more flexibility with shared SPI etc configurable. So went with directly without autodetect.
We discussed this before [here](https://github.com/lovyan03/LovyanGFX/issues/294)

Its working fine. Here is the config file.

https://github.com/sukesh-ak/ESP32-TUX/blob/master/main/devices/conf_Makerfabs_S3_PTFT.h

lovyan03, 2022-11-24:
??
What should I do?

sukesh-ak, 2022-11-24:
> ?? What should I do?

Its working now. Nothing expected from you :)
Thank you.

closed 2022-11-24 by sukesh-ak

ammaree, 2022-11-29:
@lovyan03 

Can we please reopen this request. As far as I can remember @sukesh-ak opened this issue after we had some communication. My requirement are fairly straightforward.
https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html
ESP IDF v5 (master)
Main application written in "C" not "C++"
If possible auto detection of the hardware used else manual/fixed confirm OK

Very simple example with 2 functions, both to be called from existing "C" code
gui_init() - initialise the display
gui_update(char * pBuf) - clear screen and display contents of text buffer on screen as fast as possible.

Once we have this working we will add a basic GUI

lovyan03, 2022-11-29:
@ammaree 
I have no idea what you are talking about.

sukesh-ak, 2022-11-30:
@ammaree Sounds like you are talking about my sample. 
The display you mentioned is already supported now with LovyanGFX driver in my sample.

Feel free to open a discussion here if you need to
https://github.com/sukesh-ak/ESP32-TUX/discussions

ammaree, 2022-11-30:
@lovyan03 
Apologies, sorry I bothered you.

hayschan, 2022-12-03:
@sukesh-ak does auto-detect support this display?

sukesh-ak, 2022-12-03:
> @sukesh-ak does auto-detect support this display?

I don't think it does yet. But here is the configuration for it
https://gist.github.com/sukesh-ak/7bcafd2227a2c2fa727208f4ffa47308