lovyan03/LovyanGFX #288: Touch is not working with parallel-8 bit

bil7317, 2022-10-04:
## Environment ( 実行環境 )

- MCU or Board name:  [ESP-WROOM-Dev-kit-V1]
- Panel Driver IC:    [ILI9486]
- Bus type:           [Parallel8]
- LovyanGFX version:  [latest develop]
- FrameWork version:   [ArduinoESP32 v2.0.3]
- Build Environment :   [ArduinoIDE]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )
 The screen touch is not working.
### Expected Behavior ( 期待される動作 )
The touch should work so I can get the calibration data for touch.
### Actual Behavior ( 実際の動作 )
When I start touch calibration, it displays the "touch the arrow marker" on the screen but the touch is not working at all. 

[image]

### Steps to reproduce ( 再現のための前提条件 )

1. Use Arduino IDE to run the code.
2. Schematic is as mentioned in code.

### Code to reproduce this issue ( 再現させるためのコード )

```
https://gist.github.com/bil7317/7809f2276828a78648bedfdd69d29606
```

lovyan03, 2022-10-04:
```
    { // タッチスクリーン制御の設定を行います。（必要なければ削除）Configure settings for touch screen control. (delete if not necessary)
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 319;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 479;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = -1;   // INTが接続されているピン番号

////////////////////////////////////////////////
// ↓ bus_shared should be false.
////////////////////////////////////////////////
      cfg.bus_shared = false; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation =0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

////////////////////////////////////////////////
// ↓ You will need to add this setting.
// The pin numbers should be set according to your environment.
////////////////////////////////////////////////
      cfg.spi_host = VSPI_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST)
      cfg.freq = 1000000;     // SPIクロックを設定
      cfg.pin_sclk = 18;     // SCLKが接続されているピン番号
      cfg.pin_mosi = 23;     // MOSIが接続されているピン番号
      cfg.pin_miso = 19;     // MISOが接続されているピン番号
      cfg.pin_cs   =  5;     //   CSが接続されているピン番号
////////////////////////////////////////////////
// ↑ You will need to add this setting.
// The pin numbers should be set according to your environment.
////////////////////////////////////////////////

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }
```

bil7317, 2022-10-07:
I have made the recommended changes and implemented my schematics according to the pins defined in code, but the touch is still not working. It might be possible that my LCD is damaged.
[Code Here=](https://gist.github.com/bil7317/02ef40ab4ea75d8a15e2ace15e05e57d) 
                            https://gist.github.com/bil7317/02ef40ab4ea75d8a15e2ace15e05e57d

lovyan03, 2022-10-07:
Your code uses Touch_XPT2046, is the touch controller on the board XPT2046?
If so, please reconfigure the pin numbers as you have wired them.
I am not an esper, so I don't know how you wired it.

closed 2022-12-25 by github-actions[bot]