lovyan03/LovyanGFX #290: Esp32-cam + ili9488

Sentinel8000, 2022-10-07:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32 wrover modul
- Panel Driver IC:    ILI9488
- Bus type:           SPI
- LovyanGFX version:  [v0.4.17|latest develop|other]
- FrameWork version:   Arduino IDE 1.8.12
- Build Environment :   ArduinoIDE
- Operating System:   Windows 10

## Problem Description ( 問題の内容 )

I try use the this display driver with esp32-cam board + 3.5" TFT MSP3520 what have ili9488

### Expected Behavior ( 期待される動作 )

I have succesfully set (without using touch) the user setting example and i see the R G B text without problem. I want add my esp camera code to push the picture directly from camera to 3.5" display.

### Actual Behavior ( 実際の動作 )

After i try add the esp32 cam code to example, after compile i can see the picture looks fast but distorted, i see several weird lines. 

### Steps to reproduce ( 再現のための前提条件 )

`
#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include <esp_camera.h>

// Camera pins:

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

camera_fb_t *fb = NULL;

class LGFX : public lgfx::LGFX_Device
{

lgfx::Panel_ILI9488     _panel_instance;
lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス
lgfx::Light_PWM     _light_instance;
lgfx::Touch_XPT2046          _touch_instance;

public:

  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

[... 195 lines cut ...]
    s->set_vflip(s, 1);//flip it back
    s->set_hmirror(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    //s->set_saturation(s, -2);//lower the saturation
  //}
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);
  
//    show_log(2);
}

void readCamera() {
  //camera_fb_t *fb = NULL;
  fb = NULL;
  fb = esp_camera_fb_get();
  //if (stream_flag == 1) {
      display.pushImage(11, 50, fb->width, fb->height, (lgfx::swap565_t *)fb->buf);

  //}
}
`

My code is bad / something missing or i have wrong set or display driver is incompatible with the esp32 cam driver?

thank you for check or for any help

Sentinel8000, 2022-10-07:
[image]

lovyan03, 2022-10-07:
```
#define Y4_GPIO_NUM       19
```
```
      cfg.pin_miso = 19;            // SPIのMISOピン番号を設定 (-1 = disable)
```

conflict

```
#define HREF_GPIO_NUM     23
```
```
      cfg.pin_mosi = 23;     // MOSIが接続されているピン番号
```

conflict

```
      cfg.panel_width      =   480;  // 実際に表示可能な幅
      cfg.panel_height     =   320;  // 実際に表示可能な高さ
```
This setting cannot be displayed correctly.
ILI9488 : width320  x  height480. 

I am not a correction worker. You should review your setup more carefully.

Sentinel8000, 2022-10-07:
Sure you are not ,,correction worker" as im not pro developer like you just a hobby code ,,hacker" lbut you can prevent same cases, if make your examples more understandable and transparent:> 

Currently was for me first not realy clear what i need change see this million parameter.

Now after change the resolution back to 240x320 picture is clear. But here comes - if i have wroted for you a false bug, what is the highest reolsution what i can set in this modded example?

Sentinel8000, 2022-10-07:
[image]

Sentinel8000, 2022-10-07:
[image]

tobozo, 2022-10-07:
congrats! 

feel free to document and submit your configuration, see the `src/lgfx_user/` folder for the suggested formats

Sentinel8000, 2022-10-07:
I want still check what is the maximum in resolution, or this 320x240 is the maximum.

Picture looks good and fluent, refresh is good - just looks weird on a 3.5" inch display this small window without any strech or bigger resolution.

Sentinel8000, 2022-10-07:
I have maked myself some tests, these also works:

cfg.panel_width      =   296;  
cfg.panel_height     =   400;  

and

cfg.panel_width      =   320;
cfg.panel_height     =   480;

What i have noticed in lamp light is a small ,,ghost" effect, maybe this is a refresh limitation, im not sure.

Now if im put my hands between the coala bear toy i see few seconds the coala shape and disappear or im a ghost:>

I have tested the esp32 cam itself in this code:  If i try my camera resolution set higher from config.frame_size = FRAMESIZE_QVGA; // 340x240 -> FRAMESIZE_CIF; or FRAMESIZE_VGA set esp32 goes into reboot loop, there is a problem with the fb:

Serial monitor:

19:50:10.855 -> E (7360) cam_hal: FB-SIZE: 153600 != 236800
19:50:10.946 -> E (7440) cam_hal: FB-SIZE: 153600 != 236800
19:50:10.946 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
19:50:10.946 -> 
19:50:10.946 -> Core  1 register dump:
19:50:10.946 -> PC      : 0x400d2334  PS      : 0x00060130  A0      : 0x800d2366  A1      : 0x3ffb2780  
19:50:10.994 -> A2      : 0x00000000  A3      : 0x3ffc2418  A4      : 0x00000000  A5      : 0x00000140  
19:50:10.994 -> A6      : 0x000001e0  A7      : 0x003fffff  A8      : 0x01000000  A9      : 0x3ffb2760  
19:50:10.994 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000010  
19:50:10.994 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
19:50:10.994 -> EXCVADDR: 0x00000008  LBEG    : 0x40087f58  LEND    : 0x40087f74  LCOUNT  : 0x00000000  
19:50:10.994 -> 
19:50:10.994 -> 
19:50:10.994 -> Backtrace:0x400d2331:0x3ffb27800x400d2363:0x3ffb2800 0x400d7edd:0x3ffb2820 
19:50:10.994 -> 
19:50:10.994 -> 
19:50:10.994 -> 
19:50:11.040 -> 
19:50:11.040 -> ELF file SHA256: 0000000000000000
19:50:11.040 -> 
19:50:11.040 -> Rebooting...

Maybe this is also a limitation, i dont know.