lovyan03/LovyanGFX #292: How to config the board that use 3 difference SPI-bus ?

TridentTD, 2022-10-13:
I have a board that use 3 difference SPI-bus.

The first for TFT SPI.          (MISO 12, MOSI 13, SCLK 14, CS 15)
The second for Touch SPI. (MISO 39, MOSI 32, SCLK 25, CS 33)
And the third for SD SPI.   (MISO 19, MOSI 23, SCLK 18, CS  5)

Now I can config for TFT and touch interface SPI. It's worked finely.
However, I wish to use SD SPI that difference from TFT and Touch SPI.

I have tried by the latest "LovyanGFX-develope" version, 
but I don't know how to config for SD.

For using sd too, how to config SD SPI? 

Thank you.

lovyan03, 2022-10-13:
If the touch controller is an XPT2046, use the develop branch version and set `spi_host = -1`.
With this setting, the XPT2046 can operate on software SPI.

TridentTD, 2022-10-13:
> If the touch controller is an XPT2046, use the develop branch version and set `spi_host = -1`. With this setting, the XPT2046 can operate on software SPI.

It has already worked.
Thank you.

closed 2022-10-16 by lovyan03