lovyan03/LovyanGFX #293: 5' inch - ssd1963 - gsl1680F - 8 bit Parallel - ld.exe region `dram0_0_seg' overflowed by 39768 bytes

iw7dmh, 2022-10-16:
**## Environment:**

- MCU or Board name: **DOIT ESP32 DEVKIT V1** 
- Panel Driver IC: **SSD1963**
- Bus type: **Parallel8**
- LovyanGFX version:  **v0.4.18**
- FrameWork version:   **ArduinoESP32 v2.0.3**
- Build Environment :   **ArduinoIDE 2.0**
- Operating System:   **Windows 11**

**## Problem Description**
Hi, 
I followed your install guide on youtube [https://www.youtube.com/watch?v=IPCvQ4o_WP8&t=320s](https://www.youtube.com/watch?v=IPCvQ4o_WP8&t=320s) and everything went fine until the calibration data collect.
Then I installed and configured the lvgl library (9.0.0). Code compile is fine but I am having the following error:
c:/users/enzo/appdata/local/ ...  **../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 39768 bytes**

### Code to reproduce this issue
Code is identical to the one shown in the video tutorial - Compilation is fine, the problem is in the building/linking step.
The display is from here https://www.buydisplay.com/lcd-5-ssd1963-tft-module-touch-screen-display-800x480-mcu-arduino

I suspect I won't be able to use such a large screen - can you confirm it or can you give a workaround solution?

Thank you in advance for any help

tobozo, 2022-10-16:
hi, thanks for your feedback.

lvgl is big so you probably need to select a bigger partition scheme to avoid this linking error

closing this as not a LovyanGFX issue

closed 2022-10-16 by tobozo

iw7dmh, 2022-10-16:
Hi,

thank you very much for your prompt replay.
I did a test with LVGL 8.3 and code works as expected.

I did another test trying to compile the official LVGL 9.0 Arduino example and I got the same `dram0_0_seg' error.

I don't know if I can post this message on your github repository, maybe it could help someone else. In the mean time I'll add an issue on LVGL github.

Thank you again and best regards
Vincenzo Stefanazzi
Italy

tobozo, 2022-10-16:
it's not a LVGL issue either unless the test sketch doesn't compile

things you can do:

- try with ArduinoESP32 package v2.0.5 instead of v2.03
- check your sketch and move the heavy stuff (images, binaries) to some filesystem like spiffs/littlefs or SD
- refactor your code to minimize pre-allocations and avoid framebuffers ( 800x480 is a lot )

iw7dmh, 2022-10-17:
> it's not a LVGL issue either unless the test sketch doesn't compile

Yes, that's just the test sketch in the example directory (**LVGL_Arduino.ino**) configured with plain TFT_eSPI libs.
I did a test on a second computer and the result is the same.

tobozo, 2022-10-17:
> LVGL version: **v9.0 dev**

@iw7dmh Dev versions are notoriously unstable and unsupported.

I guess the solution is: don't use the `dev` version unless you're contributing as a developer and want to submit a fix.