lovyan03/LovyanGFX #294: Add MISO PIN for WT32-SC01 autodetect

sukesh-ak, 2022-10-17:
## Problem Description ( 問題の内容 )
Please add correct MISO pin here, so that we can use other SPI devices like SD card using LovyanGFX shared SPI option.

https://github.com/lovyan03/LovyanGFX/blob/9de99d25496413f133ce3c057fa3edb60148f523/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32.hpp#L1717

Thanks to @dizcza for pointing this out.

tobozo, 2022-10-17:
hi,

I don't know what the *correct MISO pin* is in this context, but if you do please feel free to submit a pull request :+1:

dizcza, 2022-10-17:
@tobozo add an option to menuconfig and set -1 by default.

tobozo, 2022-10-17:
@dizcza unless there's an official value for this MISO pin, it sounds like those menuconfig values should be project specific rather than driver specific

@sukesh-ak your custom config can be achieved by disabling autodetect and managing the LGFX configuration as seen in the [how to use](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino) example, just import the `LGFX_WT32_SC01` board/bus config values from autodetect and assign your esp-idf `MY_CONFIG_MISO_PIN*` macro from there.

sukesh-ak, 2022-10-17:
> @dizcza unless there's an official value for this MISO pin, it sounds like those menuconfig values should be project specific rather than driver specific
>
> @sukesh-ak your custom config can be [...]

Yes sorry guys we can close this. 
I have created custom config and disabled autodetect. 

Here is the updated sample
https://github.com/sukesh-ak/WT32-SC01-PLUS-LVGL-IDF

closed 2022-10-17 by tobozo

dizcza, 2022-10-20:
Such a pity that the users are forced to write up a script to initialize SPI bus, connect a touch I2C periphery, light screen *just because of one pin*. I'd gladly leave it to autodetect. This will also ensure code maintenance rather than one-stone code.

What if instead, you provide a way to subclass the LGFX class and reinitialize the SPI bus only and leave everything as it is? Would it work @tobozo ?

Currently, these members are private:
```c++
    lgfx::Panel_Device* _panel_last = nullptr;
    lgfx::ILight* _light_last = nullptr;
    lgfx::ITouch* _touch_last = nullptr;
    lgfx::Bus_SPI _bus_spi;
```

tobozo, 2022-10-20:
@dizcza only non standard builds (e.g. M5Stack + secondary display) should disable autodetect.

Did you add an extra device not listed in the product specifications? 

If so the build is custom and it's up to you to manage your configuration as detailed in the [how to use](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino) example.

dizcza, 2022-10-21:
I do use a device supported by your autodetect - WT32-SC01. As the header of the issue outlines.

I just need to make SPI bus full-duplex. Reinitialize it.

dizcza, 2022-10-21:
> If so the build is custom and it's up to you to manage your configuration as detailed in the [how to use](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino) example.

This is precisely what I do not want to do as WT32-SC01 is already supported. That's why I wrote

> Such a pity that the users are forced to write up a script to initialize SPI bus, connect a touch I2C periphery, light screen just because of one pin. I'd gladly leave it to autodetect. This will also ensure code maintenance rather than one-stone code.

dizcza, 2022-10-21:
What I'm asking is being able to issue
```c++
      auto cfg = _bus_spi.config(); // gets the structure for bus settings.
      cfg.pin_miso = 33;
      _bus_spi.config(cfg);              // reflects the setting value on the bus.
```
in the subclassed constructor.

dizcza, 2022-10-21:
3 lines instead of 130 hardcoded commands.

tobozo, 2022-10-21:
here's an experimental snippet for you, courtesy of @lovyan03 

```cpp
LGFX display;

void setup()
{
  display.init();
  auto bus = (lgfx::Bus_SPI*)(display.panel()->bus());
  autp cfg = bus->config();
  cfg.pin_miso = 33;
  bus->init();
}

```

dizcza, 2022-10-21:
It works with a minor and yet important change: you need to deinitialize a previously initialized SPI bus prior to initialing it for the second time. I wonder why I didn't get any `spi bus init error` at line `bus->init()` - do you hide these logs somehow?

Here is the complete code that works.
```c
        lcd.init();
        auto bus = (lgfx::Bus_SPI*)(lcd.panel()->bus());
        bus->release();
        auto cfg = bus->config();
        cfg.pin_miso = 33;
        bus->config(cfg);
        bus->init();
```

----

But guys! I see a large change on develop branch. With the latest changes, the project used to work fails to build.

```
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32_all.hpp:3114:1: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
 3114 |   ESP_LOGE("LGFX","pkg:%d", pkg_ver);
      |                             ~
      |                             |
      |                             uint32_t {aka long unsigned int}
/home/dizcza/tools/esp-idf/components/log/include/esp_log.h:282:59: note: in definition of macro 'LOG_FORMAT'
  282 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                                           ^~~~~~
/home/dizcza/tools/esp-idf/components/log/include/esp_log.h:432:41: note: in expansion of macro 'ESP_LOG_LEVEL'
  432 |         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                         ^~~~~~~~~~~~~
/home/dizcza/tools/esp-idf/components/log/include/esp_log.h:334:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  334 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format __VA_OPT__(,) __VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/components/LovyanGFX/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32_all.hpp:3114:3: note: in expansion of macro 'ESP_LOGE'
 3114 |   ESP_LOGE("LGFX","pkg:%d", pkg_ver);
      |   ^~~~~~~~
```

To resolve the issues, I added a hotfix to the project cmake file:

```cmake
idf_component_get_property(lib main COMPONENT_LIB)
target_compile_options(${lib} PRIVATE -Wno-error=format)
```

I hope it's temporary and you'll fix this soon.

BTW, why did you mark the package version log as an error? It looks scary in the logs.

tobozo, 2022-10-21:
> BTW, why did you mark the package version log as an error? It looks scary in the logs.

some leftovers from halloween maybe ? :rofl:

dizcza, 2022-10-23:
I have another issue that might be spelled on a separate page but in favor of our cozy chatting I'm posting it here.

## The problem

On `abort()` and subsequent kernel reset autodetect doesn't find the right board. First attempt: success. Second: failure. It might be due to RAM keeping old values between resets or I don't know.

## Script

Borrowed from https://github.com/sukesh-ak/LVGL8-WT32-SC01-IDF/blob/master/main/main.cpp.

Just add `abort()` at the end.

```c++
void app_main(void)
    {
        lcd.init(); // Initialize LovyanGFX

        lv_init();  // Initialize lvgl

        // Setting display to landscape
        if (lcd.width() < lcd.height())
            lcd.setRotation(lcd.getRotation() ^ 1);

        /* LVGL : Setting up buffer to use for display */
        lv_disp_draw_buf_init(&draw_buf, buf, NULL, screenWidth * 10);

        /*** LVGL : Setup & Initialize the display device driver ***/
        static lv_disp_drv_t disp_drv;
        lv_disp_drv_init(&disp_drv);
        disp_drv.hor_res = screenWidth;
        disp_drv.ver_res = screenHeight;
        disp_drv.flush_cb = display_flush;
        disp_drv.draw_buf = &draw_buf;
        lv_disp_drv_register(&disp_drv);

        /*** LVGL : Setup & Initialize the input device driver ***/
        static lv_indev_drv_t indev_drv;
        lv_indev_drv_init(&indev_drv);
        indev_drv.type = LV_INDEV_TYPE_POINTER;
        indev_drv.read_cb = touchpad_read;
        lv_indev_drv_register(&indev_drv);

        /* Create and start a periodic timer interrupt to call lv_tick_inc */
        const esp_timer_create_args_t periodic_timer_args = {
            .callback = &lv_tick_task,
            .name = "periodic_gui"};
        esp_timer_handle_t periodic_timer;
        ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
        ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, LV_TICK_PERIOD_MS * 1000));

        /*** Create simple label and show LVGL version ***/

        sprintf(txt, "WT32-SC01 with LVGL v%d.%d.%d", lv_version_major(), lv_version_minor(), lv_version_patch());

        lv_obj_t *label = lv_label_create(lv_scr_act()); // full screen as the parent
        lv_label_set_text(label, txt);                   // set label text
        lv_obj_align(label, LV_ALIGN_TOP_MID, 0, 20);    // Center but 20 from the top

        tlabel = lv_label_create(lv_scr_act());         // full screen as the parent
        lv_label_set_text(tlabel, "Touch:(000,000)");   // set label text
        lv_obj_align(tlabel, LV_ALIGN_TOP_RIGHT, 0, 0); // Center but 20 from the top

        lv_button_demo(); // lvl buttons

        while (1)
        {
            lv_timer_handler(); /* let the GUI do its work */
            vTaskDelay(1);
            abort();
        }
[... 174 lines cut ...]
I (928) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (928) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (938) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (948) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (958) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (968) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (978) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (988) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

abort() was called at PC 0x400d827b on core 0
0x400d827b: app_main at /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/main/main.cpp:114

Backtrace: 0x40081e2e:0x3ffc0430 0x400862e5:0x3ffc0450 0x4008b726:0x3ffc0470 0x400d827b:0x3ffc04e0 0x4011f557:0x3ffc0520
0x40081e2e: panic_abort at /home/dizcza/tools/esp-idf/components/esp_system/panic.c:417

0x400862e5: esp_system_abort at /home/dizcza/tools/esp-idf/components/esp_system/esp_system.c:135

0x4008b726: abort at /home/dizcza/tools/esp-idf/components/newlib/abort.c:38

0x400d827b: app_main at /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/main/main.cpp:114

0x4011f557: main_task at /home/dizcza/tools/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)
```

As you see, the first attempt succeeds - WT32-SC01 has  been correctly detected - but not the following (all others fail).

dizcza, 2022-10-23:
Keeping simple & same behavior:
```c
    void app_main(void)
    {
        lcd.init(); // Initialize LovyanGFX

        while (1)
        {
            vTaskDelay(1);
            abort();
        }
    }
```

tobozo, 2022-10-23:
unless the MISO pin of this original thread is involved, I suggest you create a new issue for that

meanwhile here some thoughts about the issue:

esp32-arduino packkage had an issue with similar symptoms involving SPI and SD Card, where SD.begin() would only work once per power cycle and fail across resets.

it turned out to fail 7 times in a row, then work once, then fail 7 times again... some bit shifting was occuring and affecting CS pin handling, causing timeouts and other side effects such as preventing the next init to respond normally.

it was fixed with the appearance of a new esp-idf config flag that affected SPI behaviour, it came with esp-idf 4.4

wild suggestion: make this a feature, detect when a reset has occured, and skip the init :-)

```cpp
if( rtc_get_reset_reason(0)!=12 /* SW_CPU_RESET */ ) 
{
  lcd.init();
}
```