lovyan03/LovyanGFX #295: CVBS output flickers/glitches when ESP32 SPIRAM is enabled

alex1115alex, 2022-10-18:
## Environment ( 実行環境 )

- MCU or Board name:  ESP-WROOM-32
- Panel Driver IC:    CVBS
- Bus type:           I2C
- LovyanGFX version:  latest develop
- FrameWork version:   ESP-IDF v4.4
- Build Environment :   PlatformIO
- Operating System:   Linux

## Problem Description ( 問題の内容 )

Output becomes glitchy when "CONFIG_ESP32_SPIRAM_SUPPORT" is enabled and set to "CONFIG_SPIRAM_USE_MALLOC" or "CONFIG_SPIRAM_USE_CAPS_ALLOC".

[image]

### Expected Behavior ( 期待される動作 )

1. Enable SPIRAM support
2. Display works as intended

### Actual Behavior ( 実際の動作 )

1. Enable SPIRAM support
2. Display becomes glitchy (see GIF above)

### Steps to reproduce ( 再現のための前提条件 )

1. Set LovyanGFX to CVBS
2. Enable ESP32 SPIRAM support in sdkconfig.[environment]

// If possible, attach a picture of your setup/wiring here.

(TinyPico ESP32 + CVBS to USB capture card)
[image]

### Code to reproduce this issue ( 再現させるためのコード )

https://github.com/alex1115alex/LovyanGFX-CVBS-Glitch

lovyan03, 2022-10-19:
@alex1115alex 
This symptom occurs when data is not read from PSRAM in time.

When I run your sample code, glitching does indeed occur.
On the other hand, if only the framework is changed to arduino, no glitching occurs.
As far as I have tested here, it is stable even at 640x480 resolution.

Perhaps there is a difference in the MenuConfig settings.
Please review your CPU speed, SPIRAM settings, etc.
I am not very familiar with MenuConfig of ESP-IDF.

alex1115alex, 2022-10-19:
Hey @lovyan03, thanks for giving me a good starting point to work from. I'll poke around and report back if I'm able to find a fix...

alex1115alex, 2022-10-27:
@lovyan03 I'm back with two solutions.

The first (quick and dirty): Set `use_psram = 0` in the display settings file. This fixes the issue of flickering while SPIRAM is enabled for the rest of the application, but obviously doesn't allow Lovyan to access SPIRAM...

The second (slightly better): Change the preferred core from 0 to 1 in Panel_CVBS.cpp:87. I've made a PR for this [here](https://github.com/lovyan03/LovyanGFX/pull/303). I'm not 100% certain why this works, but I got the idea from [this issue](https://github.com/espressif/esp-idf/issues/2892) which suggests that different cores use different sections of SPIRAM while using the EspIdf environment with older ESP32 revisions.

I'll leave this issue open in case I stumble upon any new info in the next week or so...

tobozo, 2022-10-27:
idea : add `setTaskCoreISR( uint8_t )` and `uint8_t core_num = 0` to `config_detail_t`, so it becomes an editable setting.

also add `uint8_t core_num = 0` to `scanline_cache_t` and set the value from `Panel_CVBS::init` by doing `_scanline_cache.core_num = _config_detail.core_num`, then use `core_num`  with `xTaskCreatePinnedToCore`

lovyan03, 2022-10-29:
Thanks for your report !
Panel_CVBS is scheduled for revision. The model-dependent codes will be moved to the newly established Bus_CVBS, and the dependent codes will be removed from Panel_CVBS.
I will resolve this Issue when I make that correction.

closed 2022-12-25 by github-actions[bot]

lovyan03, 2023-01-08:
I apologize for the long wait.
I had planned to establish a new Bus_CVBS, but since the work has been difficult, we have added task priority settings to Panel_CVBS for the time being.
You can currently try this in the develop branch.