lovyan03/LovyanGFX #296: ILI9486 8 bit parallel display with esp32 not working

Sentinel8000, 2022-10-22:
I try set my ILI9486 8 bit parallel display but, lookslike not working.

Not realy clear what i have forget or missing, realy hard understand the example with the japan comments:/ and there is no tutorial.

Connection what im using for the ili9488 - ESP32

TFT_DC -> 18 // Data Command control pin (Labelled as LCD_RS on my display)
TFT_CS -> 33 // Chip select control pin
TFT_RST -> 32 // Reset pin
TFT_RD -> 2
TFT_WR -> 4 // Write strobe control pin
TFT_D0 -> 12 // Must use pins in the range 0-31 for the data bus
TFT_D1 -> 13 // so a single register write sets/clears all bits
TFT_D2 -> 26
TFT_D3 -> 25
TFT_D4 - > 21
TFT_D5 -> 5
TFT_D6 -> 27
TFT_D7 -> 14

`
// v1.0.0 を有効にします(v0からの移行期間の特別措置です。これを書かない場合は旧v0系で動作します。)
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

// ESP32でLovyanGFXを独自設定で利用する場合の設定例

/// 独自の設定を行うクラスを、LGFX_Deviceから派生して作成します。
class LGFX : public lgfx::LGFX_Device
{
/*
 クラス名は"LGFX"から別の名前に変更しても構いません。
 AUTODETECTと併用する場合は"LGFX"は使用されているため、LGFX以外の名前に変更してください。
 また、複数枚のパネルを同時使用する場合もそれぞれに異なる名前を付けてください。
 ※ クラス名を変更する場合はコンストラクタの名前も併せて同じ名前に変更が必要です。

 名前の付け方は自由に決めて構いませんが、設定が増えた場合を想定し、
 例えばESP32 DevKit-CでSPI接続のILI9341の設定を行った場合、
  LGFX_DevKitC_SPI_ILI9341
 のような名前にし、ファイル名とクラス名を一致させておくことで、利用時に迷いにくくなります。
//*/

// 接続するパネルの型にあったインスタンスを用意します。
//lgfx::Panel_GC9A01      _panel_instance;
//lgfx::Panel_GDEW0154M09 _panel_instance;
//lgfx::Panel_HX8357B     _panel_instance;
//lgfx::Panel_HX8357D     _panel_instance;
//lgfx::Panel_ILI9163     _panel_instance;
//lgfx::Panel_ILI9341     _panel_instance;
//lgfx::Panel_ILI9342     _panel_instance;
//lgfx::Panel_ILI9481     _panel_instance;
lgfx::Panel_ILI9486     _panel_instance;
//lgfx::Panel_ILI9488     _panel_instance;
//lgfx::Panel_IT8951      _panel_instance;
//lgfx::Panel_RA8875      _panel_instance;
//lgfx::Panel_SH110x      _panel_instance; // SH1106, SH1107
//lgfx::Panel_SSD1306     _panel_instance;
//lgfx::Panel_SSD1327     _panel_instance;
//lgfx::Panel_SSD1331     _panel_instance;
//lgfx::Panel_SSD1351     _panel_instance; // SSD1351, SSD1357
//lgfx::Panel_SSD1963     _panel_instance;
//lgfx::Panel_ST7735      _panel_instance;
//lgfx::Panel_ST7735S     _panel_instance;
//lgfx::Panel_ST7789      _panel_instance;
//lgfx::Panel_ST7796      _panel_instance;

// パネルを接続するバスの種類にあったインスタンスを用意します。
//  lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス
//lgfx::Bus_I2C       _bus_instance;   // I2Cバスのインスタンス (ESP32のみ)
[... 183 lines cut ...]
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30,30,display.width()-60,display.height()-60,count*7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  int32_t x, y;
  if (display.getTouch(&x, &y)) {
    display.fillRect(x-2, y-2, 5, 5, count*7);
  }
}`

lovyan03, 2022-10-23:
タッチパネルとバックライトのピン設定が競合しています。
使用しない場合、タッチパネルとバックライトの項目をコメントアウトするか削除してください。

Sentinel8000, 2022-10-23:
これが私の問題です。私は開発者ではありません。この例から何を削除する必要があるのか 明確ではありません。 ありがとうございました

lovyan03, 2022-10-23:
If you want touch control, need setup `_touch_instance`.
If no need touch control, delete `_touch_instance`.

If you want backlight control, need setup `_light_instance`.
If no need backlight control, delete `_light_instance`.

Sentinel8000, 2022-10-23:
I have trouble, because lookslike i have not uncomment all required line and the compiler runned to error. After few try, now compiler ok, now i need re connect display to test the parallel display.

For me was not clear where start, where end these blocks.

lovyan03, 2022-10-23:
no idea.

closed 2022-10-23 by lovyan03

Sentinel8000, 2022-10-23:
If somebody looks around  in closed issues and want a simple, working example for ILI9486 without touch, light control, here is it:

```
/*
PINS ESp32-ILI9486

TFT_DC -> 18 // Data Command control pin (Labelled as LCD_RS on my display)
TFT_CS -> 33 // Chip select control pin
TFT_RST -> 32 // Reset pin
TFT_RD -> 2
TFT_WR -> 4 // Write strobe control pin
TFT_D0 -> 12 // Must use pins in the range 0-31 for the data bus
TFT_D1 -> 13 // so a single register write sets/clears all bits
TFT_D2 -> 26
TFT_D3 -> 25
TFT_D4 - > 21
TFT_D5 -> 5
TFT_D6 -> 27
TFT_D7 -> 14
*/

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

lgfx::Panel_ILI9486     _panel_instance;
lgfx::Bus_Parallel8 _bus_instance;   // 8ビットパラレルバスのインスタンス (ESP32のみ)
lgfx::Light_PWM     _light_instance;
lgfx::Touch_XPT2046          _touch_instance;

public:

  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

// 8ビットパラレルバスの設定
      cfg.i2s_port = I2S_NUM_0;     // 使用するI2Sポートを選択 (I2S_NUM_0 or I2S_NUM_1) (ESP32のI2S LCDモードを使用します)
      cfg.freq_write = 20000000;    // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
      cfg.pin_wr =  4;              // WR を接続しているピン番号
      cfg.pin_rd =  2;              // RD を接続しているピン番号
      cfg.pin_rs = 18;              // RS(D/C)を接続しているピン番号
      cfg.pin_d0 = 12;              // D0を接続しているピン番号
      cfg.pin_d1 = 13;              // D1を接続しているピン番号
      cfg.pin_d2 = 26;              // D2を接続しているピン番号
      cfg.pin_d3 = 25;              // D3を接続しているピン番号
      cfg.pin_d4 = 21;              // D4を接続しているピン番号
      cfg.pin_d5 = 5;              // D5を接続しているピン番号
      cfg.pin_d6 = 27;              // D6を接続しているピン番号
      cfg.pin_d7 = 14;              // D7を接続しているピン番号

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs           =    33;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    32;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      cfg.panel_width      =   320;  // 実際に表示可能な幅
      cfg.panel_height     =   480;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  true;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  true;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

LGFX display;

void setup(void)
{
  display.init();
  display.fillScreen(TFT_BLACK);
}

uint32_t count = ~0;
void loop(void)
{
  display.startWrite();
  display.setColorDepth((count & 8) ? 16 : 24);
  display.setTextColor(TFT_WHITE);
   display.setTextSize((std::max(display.width(), display.height()) + 255) >> 8);
  display.drawString("HELLO WORLD", 30, 16);
  display.endWrite();
}
```

tobozo, 2022-10-23:
@Sentinel8000 thanks for sharing you configuration, I'll edit the title of this issue to increase searchability.