lovyan03/LovyanGFX #298: Cannot detect the right board after abort and reset

dizcza, 2022-10-24:
Originally asked in https://github.com/lovyan03/LovyanGFX/issues/294.

## Environment ( 実行環境 )

- MCU or Board name: WT32-SC01
- Panel Driver IC:    ST7796S
- Bus type:           SPI
- LovyanGFX version:  develop
- FrameWork version:   ESP-IDF v5.1 master
- Build Environment :   terminal
- Operating System:   Ubuntu 22

## Code

```c
    void app_main(void)
    {
        lcd.init(); // Initialize LovyanGFX

        while (1)
        {
            vTaskDelay(1);
            abort();
        }
    }
```

causes LovyanGFX failure to detect the board after reset.

## Logs

```
I (0) cpu_start: App cpu up.
I (304) cpu_start: Pro cpu start user code
I (304) cpu_start: cpu freq: 240000000 Hz
I (304) cpu_start: Application information:
I (309) cpu_start: Project name:     lvgl-wt32-sc01
I (315) cpu_start: App version:      730fce4-dirty
I (320) cpu_start: Compile time:     Oct 23 2022 20:56:10
I (326) cpu_start: ELF file SHA256:  0a9ed23c2d3f7a5f...
I (332) cpu_start: ESP-IDF:          v5.1-dev-1146-g1308ff3d21
I (339) heap_init: Initializing. RAM available for dynamic allocation:
I (346) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (352) heap_init: At 3FFBDE10 len 000221F0 (136 KiB): DRAM
I (358) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (365) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (371) heap_init: At 4008CB6C len 00013494 (77 KiB): IRAM
I (378) spi_flash: detected chip: generic
I (382) spi_flash: flash io: dio
I (387) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (397) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (407) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (417) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (447) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (447) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (447) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (457) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (467) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (477) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (487) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (497) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (507) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (517) LovyanGFX: [Autodetect] WT32_SC01
I (517) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (527) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (537) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (547) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1047) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1047) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[... 145 lines cut ...]
0x400862e5: esp_system_abort at /home/dizcza/tools/esp-idf/components/esp_system/esp_system.c:135

0x4008b726: abort at /home/dizcza/tools/esp-idf/components/newlib/abort.c:38

0x400d827b: app_main at /home/dizcza/Projects/Embedded/eclipse-workspace/other/LVGL8-WT32-SC01-IDF/main/main.cpp:114

0x4011f557: main_task at /home/dizcza/tools/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)
```

## Notes

I'm not using arduino. I'm using ESP-IDF.

```c
if( rtc_get_reset_reason(0)!=12 /* SW_CPU_RESET */ ) 
{
  lcd.init();
}
```

is a workaround that I don't want to include in my code.

> it was fixed with the appearance of a new esp-idf config flag that affected SPI behaviour, it came with esp-idf 4.4

What's this flag?

lovyan03, 2022-10-24:
@dizcza 
thank you for your report.

Ah, the LCD on the WT32-SC01 has the readout function disabled, so it cannot detect the LCD as it should.
We are forcing it to respond by determining the presence or absence of the touch controller via I2C communication, but maybe the I2C signal status is halfway through after the reboot.
We would like to work on confirming this within a few days, but may possibly remove it from the automatic detection.

lovyan03, 2022-10-25:
The develop branch has been updated.
With any luck, this will detect WT32-SC01.

dizcza, 2022-10-25:
Unfortunately, the fix didn't solve the problem. I've updated develop branch.

```
I (00:00:00.204) board: Board MAC 0xB8D918CD31E8
I (00:00:00.205) board: CHIP_ESP32 (1), features 0x0032, revision 100, cores 2
I (00:00:00.207) board: WT32-SC01 detected
I (00:00:00.212) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.222) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.232) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.254) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.255) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.264) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.274) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.285) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.294) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.305) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.315) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.325) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.335) LovyanGFX: [Autodetect] WT32_SC01
I (00:00:00.340) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.350) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.361) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.371) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.866) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.867) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.876) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.887) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.896) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.906) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.916) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.926) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (00:00:00.936) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:00.947) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:01.055) bsp: lv_tick timer started
I (00:00:01.056) bsp: lv_task_handler timer started
I (00:00:01.056) wt32: lv_display_init OK
I (00:00:01.060) sdcard: Mounting SD SPI: CS 25 HOST 1
I (00:00:01.206) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (00:00:01.245) sdspi_transaction: cmd=5, R1 response: command not supported
I (00:00:01.272) sdcard: SD card mounted at /sd
Name: SD16G
Type: SDHC/SDXC
Speed: 7272.73 MHz (limit: 7.50 MHz)
Size: 14916MB
CSD: ver=2, sector_size=512, capacity=30547968 read_bl_len=9
SSR: bus_width=1
Free: 13976MB
I (00:00:01.285) sdcard: /sd/RECORDS_MAG exists
I (00:00:01.743) sdcard: mkdir /sd/RECORDS_MAG/783 OK
I (00:00:01.818) sdcard: logging to '/sd/RECORDS_MAG/783/LOG.LOG'
I (00:00:01.820) RTC: DS3231 age offset 3
I (00:00:01.838) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (00:00:01.845) RTC: Attached RTC SQW handler at GPIO 35
I (16:00:34.000) RTC: Tue Oct 25 16:00:34 2022
I (16:00:34.080) sdcard: Opened /sd/RECORDS_MAG/783/RTC_SyncTime.csv
I (16:00:34.099) RTC: rtc_ds3231_write_synctime OK

>>> BEGIN '/sd/RECORDS_MAG/783/RTC_SyncTime.csv'
rtc_s,system_us
1666702834,1781666<<< END '/sd/RECORDS_MAG/783/RTC_SyncTime.csv'

I (16:00:34.183) RTC: rtc_ds3231_sync_clock_precise OK
I (16:00:34.193) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (16:00:34.205) RTC: Reset SQW GPIO 35
I (16:00:34.377) bsp: ui timer started
I (16:00:34.424) magrecord: I2C0 initialized, clk=400000
I (16:00:35.429) magrecord: Found RM3100 at 0x23
I (16:00:35.436) magrecord: RM3100 initialized
I (16:00:35.442) magrecord: RM3100 CycleCount x=200 y=200 z=200
I (16:00:35.448) magrecord: RM3100 CMM rate: 0x05
[... 133 lines cut ...]
I (13:00:44.624) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (13:00:44.634) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (13:00:44.664) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (13:00:44.664) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (13:00:44.673) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (13:00:44.683) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... cut ...]
I (13:00:44.824) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (13:00:44.833) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (13:00:44.844) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (13:00:44.854) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (13:00:44.864) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (13:00:44.874) wt32: lv_display_init failed
```

dizcza, 2022-10-26:
Now it works.

closed 2022-12-24 by dizcza