lovyan03/LovyanGFX #302: ST7735 - 0.96" TFT 160 x 80 misalignment in height

sukesh-ak, 2022-10-27:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  TinyS3
- Panel Driver IC:    ST7735S
- Bus type:           [SPI|
- LovyanGFX version:  latest 
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment :   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )
I have a rectangle to test like below. If I use height - 1 then bottom line is not visible.
```cpp
tft.fillScreen(TFT_BLACK);
tft.drawRect(0,0, tft.width() - 1, tft.height() - 2, 0x00FF00U);
```
### Expected Behavior ( 期待される動作 )
I should be able to get a rectangle with below code but see the result in photo
```cpp
tft.drawRect(0,0, tft.width() - 1, tft.height() - 1, 0x00FF00U);
```
[image]

### Actual Behavior ( 実際の動作 )
Need to give `tft.height() - 2` to get the bottom line of the rectangle

### Code to reproduce this issue ( 再現させるためのコード )

### Configuration
```cpp
#define SPI_MODE 0

#define TFT_MOSI  35  
#define TFT_MISO  37
#define TFT_SCLK  36  
#define TFT_DC    3
#define TFT_CS    34
#define TFT_RST   -1
#define TFT_BCK_LT 2 

#define TFT_WIDTH   80
#define TFT_HEIGHT  160

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ST7735S _panel_instance;
  lgfx::Bus_SPI _bus_instance; 
  
public:

  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = SPI3_HOST; 
      cfg.spi_mode =SPI_MODE;          
      cfg.freq_write = 27000000;
      cfg.freq_read = 14000000;  
      cfg.spi_3wire = true;      
      cfg.use_lock = true;       
 
      cfg.dma_channel = SPI_DMA_CH_AUTO; 

      cfg.pin_sclk = TFT_SCLK; 
      cfg.pin_mosi = TFT_MOSI; 

      cfg.pin_miso = TFT_MISO; 
      cfg.pin_dc = TFT_DC;    

      _bus_instance.config(cfg);              
      _panel_instance.setBus(&_bus_instance); 
    }

    {
      auto cfg = _panel_instance.config(); 

      cfg.pin_cs = TFT_CS;    
      cfg.pin_rst = TFT_RST;  
      cfg.pin_busy = -1; 

      cfg.panel_width = TFT_WIDTH;    // actual visible width
      cfg.panel_height = TFT_HEIGHT;   // actually visible height
      cfg.offset_x = 26;         // Panel X-direction offset amount
      cfg.offset_y = 1;         // Panel Y offset amount
      cfg.offset_rotation = 2;  // offset of rotational values from 0 to 7 (4 to 7 upside down)
      cfg.dummy_read_pixel = 17; // number of bits in dummy leads before pixel read
      cfg.dummy_read_bits = 1;  // number of bits in dummy leads before reading non-pixel data
      cfg.readable = false;      // set to true if data can be read
      cfg.invert = false;       // set to true if the light and dark of the panel is reversed
      cfg.rgb_order = false;    // set to true if the red and blue of the panel are swapped
      cfg.dlen_16bit = false;   // Set to true for panels that transmit data lengths in 16-bit increments in 16-bit parallel or SPI
      cfg.bus_shared = true;    

      // The following should only be set if the display is misalized by a driver with a variable number of pixels, such as st7735 or ILI9163.
      // This enabled makes it worse
      // cfg.memory_width = TFT_WIDTH; 
      // cfg.memory_height = TFT_HEIGHT; 

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance); // Set the panel to be used.
  }
};

```

### Code

```cpp
#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include "displayconfig.h"

static LGFX tft;

void setup(void)
{
  tft.init();
  tft.setRotation(tft.getRotation() ^ 1);
  tft.setBrightness(128);
  tft.setColorDepth(24);

  tft.fillScreen(0x000000u);  
  tft.drawRect(0,0, tft.width() - 1, tft.height() - 1, 0x00FF00U);

  tft.setTextColor(0xFF0000U);
  tft.drawString("Hello", 60, 30);

}

void loop(void)
{
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2022-10-27:
hi,

have you tried to poke with the values?

```cpp
      cfg.offset_x = 26;         // Panel X-direction offset amount
      cfg.offset_y = 1;         // Panel Y offset amount
```

sukesh-ak, 2022-10-27:
> hi,
> 
> have you tried to poke with the values?
> 
> ```c++
>       cfg.offset_x = 26;         // Panel X-direction offset amount
>       cfg.offset_y = 1;         // Panel Y offset amount
> ```

Yes spent half a day. Then gave up :)
If you are sure that's where the solution is... will spend another half day.

tobozo, 2022-10-27:
it's okay to be tired :yawning_face: 

looking at adafruit's init line for your display model (7735R mini 160x80):

```cpp
    _width = 160;
    _height = 80;
    _colstart = 24;
    _rowstart = 0;
```

that seems to fit with the 2px offset you had, so the correct values for you should look like this:

```cpp
      cfg.panel_width  = cfg.memory_width  = 160; // don't invert !!
      cfg.panel_height = cfg.memory_height = 80; // don't invert !!
      cfg.offset_x = 24;       // Panel X-direction offset amount
      cfg.offset_y = 0;         // Panel Y offset amount
```

sukesh-ak, 2022-10-27:
> that seems to fit with the 2px offset you had, so the correct values for you should look like this:
> 
> ```c++
>       cfg.panel_width  = cfg.memory_width  = 160; // don't invert !!
>       cfg.panel_height = cfg.memory_height = 80; // don't invert !!
>       cfg.offset_x = 24;       // Panel X-direction offset amount
>       cfg.offset_y = 0;         // Panel Y offset amount
> ```

Now right side line is gone but bottom is working. 
So tried `24 and 1` and it works.

@tobozo Thank you :)

Complete config below if someone else who might hit the same issue.

```cpp
class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ST7735S _panel_instance;
  lgfx::Bus_SPI _bus_instance; 
  lgfx::Light_PWM     _light_instance;
  
public:

  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = SPI3_HOST; 
      cfg.spi_mode =SPI_MODE;          
      cfg.freq_write = 27000000;
      cfg.freq_read = 14000000;  
      cfg.spi_3wire = true;      
      cfg.use_lock = true;       
 
      cfg.dma_channel = SPI_DMA_CH_AUTO; 

      cfg.pin_sclk = TFT_SCLK; 
      cfg.pin_mosi = TFT_MOSI; 

      cfg.pin_miso = TFT_MISO; 
      cfg.pin_dc = TFT_DC;    

      _bus_instance.config(cfg);              
      _panel_instance.setBus(&_bus_instance); 
    }

    {
      auto cfg = _panel_instance.config(); 

      cfg.pin_cs = TFT_CS;    
      cfg.pin_rst = TFT_RST;  
      cfg.pin_busy = -1; 

      cfg.panel_width = TFT_WIDTH;    // actual visible width
      cfg.panel_height = TFT_HEIGHT;   // actually visible height
      cfg.offset_x = 24;         // Panel X-direction offset amount
      cfg.offset_y = 1;         // Panel Y offset amount
      cfg.offset_rotation = 2;  // offset of rotational values from 0 to 7 (4 to 7 upside down)
      cfg.dummy_read_pixel = 17; // number of bits in dummy leads before pixel read
      cfg.dummy_read_bits = 1;  // number of bits in dummy leads before reading non-pixel data
      cfg.readable = false;      // set to true if data can be read
      cfg.invert = false;       // set to true if the light and dark of the panel is reversed
      cfg.rgb_order = false;    // set to true if the red and blue of the panel are swapped
      cfg.dlen_16bit = false;   // Set to true for panels that transmit data lengths in 16-bit increments in 16-bit parallel or SPI
      cfg.bus_shared = true;    

      _panel_instance.config(cfg);
    }

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = TFT_BCK_LT;              
      cfg.invert = false;           
      cfg.freq   = 44100;           
      cfg.pwm_channel = 7;          

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);  
      }

    setPanel(&_panel_instance); // Set the panel to be used.
  }
};
```

closed 2022-10-27 by sukesh-ak

lovyan03, 2022-10-27:
There seems to be some misunderstanding, so I will correct it here.

In the ST7735 series, `memory_width` and `memory_height` may differ depending on the product because memory resolution can be set by the GM0 and GM1 pins.

[image]

 - width 132 x height 162
 - width 132 x height 132
 - width 128 x height 160

I don't know which setting you are using for your panel, so you will need to explore the settings carefully.

It is important to note that the 80 x 160 resolution is merely a cropped representation of these memory areas.

Try `setRotation` from 0 to 7. If everything is displayed correctly, it is a success, but if any of width, height, or offset is incorrect, part of the image will extend off the screen.

Note that ST7735 has a vertical screen.
Do not set the width to 160 by mistake.