lovyan03/LovyanGFX #304: How to config touch-interface XPT2046 type for 480x272 on ESP32-S3?

TridentTD, 2022-10-28:
I can config the touch-interface XPT2046 type for 320x240 on ESP32. 
The following code for XPT2046 is worked well.

```c
/* ---- MCU :  ESP32  with XPT2046 ---- */

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

  #define TFT_WIDTH               320
  #define TFT_HEIGHT              240

  #define TOUCH_XPT2046
  #define TOUCH_HOST_TYPE         HSPI_HOST
  #define TOUCH_INT               36
  #define TOUCH_MOSI              32
  #define TOUCH_MISO              39
  #define TOUCH_SCLK              25
  #define TOUCH_CS                33
  
class LGFX : public lgfx::LGFX_Device {
  private:
    // ....
    lgfx::Panel_XXXXX   _panel_instance;
    lgfx::Touch_XPT2046 _touch_instance;
  public:
    LGFX(void)
    {
      // ....
      { // タッチスクリーン制御の設定を行います。（必要なければ削除）
        auto touch_cfg = _touch_instance.config();

        touch_cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
        touch_cfg.x_max      = TFT_HEIGHT-1;  // タッチスクリーンから得られる最大のX値(生の値)
        touch_cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
        touch_cfg.y_max      = TFT_WIDTH-1;  // タッチスクリーンから得られる最大のY値(生の値)
        touch_cfg.pin_int    = TOUCH_INT;   // INTが接続されているピン番号
        touch_cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
        touch_cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

        // SPI接続の場合
        touch_cfg.spi_host   = TOUCH_HOST_TYPE; //HSPI_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST)
        touch_cfg.freq       = 1000000;     // SPIクロックを設定
        touch_cfg.pin_sclk   = TOUCH_SCLK;     // SCLKが接続されているピン番号
        touch_cfg.pin_mosi   = TOUCH_MOSI;     // MOSIが接続されているピン番号
        touch_cfg.pin_miso   = TOUCH_MISO;     // MISOが接続されているピン番号
        touch_cfg.pin_cs     = TOUCH_CS;     //   CSが接続されているピン番号

        _touch_instance.config(touch_cfg);
        _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
      }
      setPanel(&_panel_instance);
    }
};
```

However I have another TFT 480x272 touch-screen with ESP32-S3 and XPT2046 touch-interface.
I have adapted touch code from previous code as the following...

```c
/* ---- MCU :  ESP32-S3  with XPT2046 ---- */

#define LGFX_USE_V1
#include <LovyanGFX.hpp>

  #define TFT_WIDTH               480
  #define TFT_HEIGHT              272

  #define TOUCH_XPT2046
  #define TOUCH_HOST_TYPE         SPI1_HOST  //  SPI1_HOST , SPI2_HOST, SPI3_HOST  
  #define TOUCH_SCLK              12
  #define TOUCH_MISO              13
  #define TOUCH_MOSI              11
  #define TOUCH_CS                38
  #define TOUCH_INT               18
  
class LGFX : public lgfx::LGFX_Device {
  private:
    // ....
    lgfx::Panel_XXXXX   _panel_instance;
    lgfx::Touch_XPT2046 _touch_instance;
  public:
    LGFX(void)
    {
      // ....
      { // タッチスクリーン制御の設定を行います。（必要なければ削除）
        auto touch_cfg = _touch_instance.config();

        touch_cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
        touch_cfg.x_max      = TFT_HEIGHT-1;  // タッチスクリーンから得られる最大のX値(生の値)
        touch_cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
        touch_cfg.y_max      = TFT_WIDTH-1;  // タッチスクリーンから得られる最大のY値(生の値)
        touch_cfg.pin_int    = TOUCH_INT;   // INTが接続されているピン番号
        touch_cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
        touch_cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

        // SPI接続の場合
        touch_cfg.spi_host   = TOUCH_HOST_TYPE; //HSPI_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST)
        touch_cfg.freq       = 1000000;     // SPIクロックを設定
        touch_cfg.pin_sclk   = TOUCH_SCLK;     // SCLKが接続されているピン番号
        touch_cfg.pin_mosi   = TOUCH_MOSI;     // MOSIが接続されているピン番号
        touch_cfg.pin_miso   = TOUCH_MISO;     // MISOが接続されているピン番号
        touch_cfg.pin_cs     = TOUCH_CS;     //   CSが接続されているピン番号

        _touch_instance.config(touch_cfg);
        _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
      }
      setPanel(&_panel_instance);
    }
};
```

I have tried from the upper XPT2046 touch code , 
the result is no response for touch interface.

How to solve the issue for ESP32-S3 with XPT2046  (480x272) touch-interface?
Thank you.

lovyan03, 2022-10-29:
If possible, please provide source code that builds and works as is.
Do not abbreviate or hide panel names.

TridentTD, 2022-10-29:
The tesing code for XPT2046 (480x272) is as the following code

```c
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

 #define TFT_WIDTH            480
 #define TFT_HEIGHT           272

#define TOUCH_XPT2046
#define TOUCH_HOST_TYPE       SPI1_HOST  //  SPI1_HOST , SPI2_HOST, SPI3_HOST  
#define TOUCH_SCLK            12
#define TOUCH_MISO            13
#define TOUCH_MOSI            11
#define TOUCH_CS              38
#define TOUCH_INT             18

class LGFX : public lgfx::LGFX_Device {
  public:
    LGFX();
    lgfx::Bus_RGB             _bus_instance;
    lgfx::Panel_RGB           _panel_instance;
    lgfx::Light_PWM           _light_instance;
    lgfx::Touch_XPT2046       _touch_instance;
};

LGFX::LGFX(void)
{
  {
    auto panel_cfg            = _panel_instance.config();
    auto panel_cfg2           = _panel_instance.config_detail();
    
    panel_cfg.memory_width    = panel_cfg.panel_width   = TFT_WIDTH;
    panel_cfg.memory_height   = panel_cfg.panel_height  = TFT_HEIGHT;
    panel_cfg.offset_x        = panel_cfg.offset_y      = 0;
    panel_cfg2.use_psram      = 1;
    
    _panel_instance.config(panel_cfg);
    _panel_instance.config_detail(panel_cfg2);
  }
  {
    auto bus_cfg              = _bus_instance.config();
    bus_cfg.panel             = &_panel_instance;
    bus_cfg.pin_d0            = GPIO_NUM_8;  // B0
    bus_cfg.pin_d1            = GPIO_NUM_3;  // B1
    bus_cfg.pin_d2            = GPIO_NUM_46; // B2
    bus_cfg.pin_d3            = GPIO_NUM_9;  // B3
    bus_cfg.pin_d4            = GPIO_NUM_1;  // B4
    bus_cfg.pin_d5            = GPIO_NUM_5;  // G0
    bus_cfg.pin_d6            = GPIO_NUM_6;  // G1
    bus_cfg.pin_d7            = GPIO_NUM_7;  // G2
    bus_cfg.pin_d8            = GPIO_NUM_15; // G3
    bus_cfg.pin_d9            = GPIO_NUM_16; // G4
    bus_cfg.pin_d10           = GPIO_NUM_4;  // G5
    bus_cfg.pin_d11           = GPIO_NUM_45; // R0
    bus_cfg.pin_d12           = GPIO_NUM_48; // R1
    bus_cfg.pin_d13           = GPIO_NUM_47; // R2
    bus_cfg.pin_d14           = GPIO_NUM_21; // R3
    bus_cfg.pin_d15           = GPIO_NUM_14; // R4

    bus_cfg.pin_henable       = GPIO_NUM_40;
    bus_cfg.pin_vsync         = GPIO_NUM_41;
    bus_cfg.pin_hsync         = GPIO_NUM_39;
    bus_cfg.pin_pclk          = GPIO_NUM_42;
    bus_cfg.freq_write        = 16000000;

    bus_cfg.hsync_polarity    = 0;
    bus_cfg.hsync_front_porch = 8;
    bus_cfg.hsync_pulse_width = 4;
    bus_cfg.hsync_back_porch  = 16;
    bus_cfg.vsync_polarity    = 0;
    bus_cfg.vsync_front_porch = 4;
    bus_cfg.vsync_pulse_width = 4;
    bus_cfg.vsync_back_porch  = 4;
    bus_cfg.pclk_idle_high    = 1;

    _bus_instance.config(bus_cfg);
    _panel_instance.setBus(&_bus_instance);
  }
  {
    auto light_cfg = _light_instance.config();
    light_cfg.pin_bl = GPIO_NUM_2;
    _light_instance.config(light_cfg);
    _panel_instance.light(&_light_instance);
  }

  { // タッチスクリーン制御の設定を行います。（必要なければ削除）
    auto touch_cfg = _touch_instance.config();

    touch_cfg.x_min           = 0;    // タッチスクリーンから得られる最小のX値(生の値)
    touch_cfg.x_max           = TFT_HEIGHT-1;  // タッチスクリーンから得られる最大のX値(生の値)
    touch_cfg.y_min           = 0;    // タッチスクリーンから得られる最小のY値(生の値)
    touch_cfg.y_max           = TFT_WIDTH-1;  // タッチスクリーンから得られる最大のY値(生の値)
    touch_cfg.pin_int         = TOUCH_INT;   // INTが接続されているピン番号
    touch_cfg.bus_shared      = false;//true; // 画面と共通のバスを使用している場合 trueを設定
    touch_cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

    // SPI接続の場合
    touch_cfg.spi_host        = TOUCH_HOST_TYPE; //HSPI_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST)
    touch_cfg.freq            = 1000000;     // SPIクロックを設定
    touch_cfg.pin_sclk        = TOUCH_SCLK;     // SCLKが接続されているピン番号
    touch_cfg.pin_mosi        = TOUCH_MOSI;     // MOSIが接続されているピン番号
    touch_cfg.pin_miso        = TOUCH_MISO;     // MISOが接続されているピン番号
    touch_cfg.pin_cs          = TOUCH_CS;     //   CSが接続されているピン番号

    _touch_instance.config(touch_cfg);
    _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
  }
  setPanel(&_panel_instance);
}

LGFX lcd;
uint32_t count = ~0;

void setup() {
  Serial.begin(115200); Serial.println();
  Serial.println("LCD4.3 test1");
  lcd.init();
  lcd.setTextSize((std::max(lcd.width(), lcd.height()) + 255) >> 8);

  if (lcd.touch())
  {
    if (lcd.width() < lcd.height()) lcd.setRotation(lcd.getRotation() ^ 1);

    // 画面に案内文章を描画します。
    lcd.setTextDatum(textdatum_t::middle_center);
    lcd.drawString("touch the arrow marker.", lcd.width() >> 1, lcd.height() >> 1);
    lcd.setTextDatum(textdatum_t::top_left);

    // タッチを使用する場合、キャリブレーションを行います。画面の四隅に表示される矢印の先端を順にタッチしてください。
    std::uint16_t fg = _TFT_WHITE;
    std::uint16_t bg = _TFT_BLACK;
    if (lcd.isEPD()) std::swap(fg, bg);

    uint16_t touch_data[8];
    lcd.calibrateTouch(touch_data, fg, bg, std::max(lcd.width(), lcd.height()) >> 3);
  }  
  
  lcd.setTextDatum(textdatum_t::middle_center);
  lcd.drawString("Hello World", lcd.width() >> 1, lcd.height() >> 1);
}

void loop() {
  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);

  lcd.setTextColor(_TFT_WHITE);
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30, 30, lcd.width() - 60, lcd.height() - 60, count * 7);
  lcd.drawFastHLine(0, 0, 10);

  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    lcd.fillRect(x - 2, y - 2, 5, 5, count * 7);
  }
}
```

The result : (as the lower image. )
  -  TFT can display.
  -  XPT2046 doesn't response.

[image]

lovyan03, 2022-11-01:
The coordinate values obtained from XPT2046 are probably in the neighborhood of 250-3900. Therefore, the following settings are always expected to be treated as off-screen

```
  #define TFT_WIDTH               320
  #define TFT_HEIGHT              240

    touch_cfg.x_min           = 0;    // タッチスクリーンから得られる最小のX値(生の値)
    touch_cfg.x_max           = TFT_HEIGHT-1;  // タッチスクリーンから得られる最大のX値(生の値)
    touch_cfg.y_min           = 0;    // タッチスクリーンから得られる最小のY値(生の値)
    touch_cfg.y_max           = TFT_WIDTH-1;  // タッチスクリーンから得られる最大のY値(生の値)
```

Try changing it as follows
```
    touch_cfg.x_min           = 100;  // Minimum X value (raw value) obtained from touch screen.
    touch_cfg.x_max           = 4000; // Maximum X value (raw value) obtained from touch screen.
    touch_cfg.y_min           = 100;  // Minimum Y value (raw value) obtained from touch screen.
    touch_cfg.y_max           = 4000; // Maximum Y value (raw value) obtained from touch screen.
```

If no SPI signal is generated at all, try setting TOUCH_INT to -1 and see if the situation changes.

closed 2022-12-25 by github-actions[bot]