lovyan03/LovyanGFX #305: How to solve the issue on GT911 Touch that gives wrong point?

TridentTD, 2022-10-28:
The GT911 touch-interface code is following...

```c
#define TFT_WIDTH               480
#define TFT_HEIGHT              272

#define TOUCH_GT911
#define TOUCH_I2C_PORT          0   // 使用するI2Cを選択 (0 or 1)
#define TOUCH_I2C_ADDR          0x5D //0x14
#define TOUCH_I2C_SDA           19
#define TOUCH_I2C_SCL           20
#define TOUCH_INT               -1
#define TOUCH_RST               38

class LGFX : public lgfx::LGFX_Device
{
public:
  //...
  lgfx::Touch_GT911 _touch_instance;

  LGFX(void)
  {  
    // ....
    // ....
    { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto touch_cfg = _touch_instance.config();
      touch_cfg.i2c_port   = TOUCH_I2C_PORT;         // 使用するI2Cを選択 (0 or 1)
      touch_cfg.i2c_addr   = TOUCH_I2C_ADDR;         // I2Cデバイスアドレス番号
      touch_cfg.pin_sda    = TOUCH_I2C_SDA;          // SDAが接続されているピン番号
      touch_cfg.pin_scl    = TOUCH_I2C_SCL;          // SCLが接続されているピン番号
      touch_cfg.pin_int    = TOUCH_INT;              // INTが接続されているピン番号
      touch_cfg.freq       = 400000;                 // I2Cクロックを設定

      touch_cfg.x_min      = 0;
      touch_cfg.x_max      = TFT_WIDTH -1;
      touch_cfg.y_min      = 0;
      touch_cfg.y_max      = TFT_HEIGHT-1;
      touch_cfg.bus_shared = false;  // true
      touch_cfg.offset_rotation = 0;   // 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

      _touch_instance.config(touch_cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }
    setPanel(&_panel_instance);
  }
};
```
The result as the following image.
[image]

How to solve the issue for GT911 (480x272) ?
Thank you.

tobozo, 2022-10-28:
1) you are assuming the touch has the exact same resolution as the TFT
2) maybe this touch interface has negative values for y_min and x_min

TridentTD, 2022-10-29:
> 1. are you assuming the touch has the exact same resolution as the TFT?
> 2. maybe this touch interface has negative values for y_min and x_min or can use default values?

Yes, the resolution is same with TFT (480x272).
When touching a point, the result occurs 1 right point, with other 3 wrong points together.

lovyan03, 2022-10-29:
It is probably not a touch problem, but a setRotate problem with the LCD.
Please remove the setRotation statement from the sample code.

TridentTD, 2022-10-29:
> It is probably not a touch problem, but a setRotate problem with the LCD. Please remove the setRotation statement from the sample code.

OK, when remove lcd.setRotation(..)   TFT 480x272 with GT911 is worked now.

By the way, I have another TFT 800x480 RGB with GT911 touch interface.
With the same code with changing the size of TFT_WIDTH and TFT_HEIGHT.

The TFT 800x480 can display well, 
exception GT911 touch is no response.

How to fix for GT911 touch interface for 800x480 TFT touch screen?

closed 2022-10-29 by TridentTD