lovyan03/LovyanGFX #306: Cannot get GT911 touch to work.

lucasmontec, 2022-11-04:
Hey! I hope you can help, I already spent 4h+ on this. Recently I bought this [ESP32 module](https://www.aliexpress.com/item/1005004632953455.html?spm=a2g0o.order_list.0.0.21ef194dV9xe9D). This is using a TFT hooked on the hardware SPI and a capacitive touch panel controlled by a GT911. I'm trying to hook up your library and cannot ever get the touch controller to work.

The hardware connections are:
[image]

I'm using this setup:
```cpp
lgfx::Panel_ST7796      _panel_instance;
lgfx::Bus_SPI       _bus_instance; 
lgfx::Light_PWM     _light_instance;
lgfx::Touch_GT911            _touch_instance;

auto cfg = _bus_instance.config(); 

// SPI (display is working)
      cfg.spi_host = HSPI_HOST;     // This display is hooked into HSPI (GPIO 13,12,14,15)
      // ESP-IDF VSPI_HOST , HSPI_HOST SPI2_HOST , SPI3_HOST
      cfg.spi_mode = 0;             // SPI (0 ~ 3)
      cfg.freq_write = 55000000;    // SPI (80MHz, 80MHz)
      cfg.freq_read  = 20000000;    // SPI
      cfg.spi_3wire  = false;        // MOSI true
      cfg.use_lock   = true;        // true
      cfg.dma_channel = 1; // DMA / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=)
      // ESP-IDF DMA SPI_DMA_CH_AUTO 1ch,2ch
      cfg.pin_sclk = 14;            // SPI SCLK
      cfg.pin_mosi = 13;            // SPI MOSI
      cfg.pin_miso = 12;            // SPI MISO (-1 = disable)
      cfg.pin_dc   = 2;            // SPI D/C (-1 = disable)
     // SD SPI

      _bus_instance.config(cfg);  
      _panel_instance.setBus(&_bus_instance);    

 auto cfg = _panel_instance.config();

      cfg.pin_cs           =    15;  // CS   (-1 = disable)
      cfg.pin_rst          =    2;   // RST  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSY (-1 = disable)
    
      cfg.panel_width      =   320;
      cfg.panel_height     =   480;
      cfg.offset_x         =     0;  
      cfg.offset_y         =     0;  
      cfg.offset_rotation  =     0;  // 0~7 (4~7)
      cfg.dummy_read_pixel =     8; 
      cfg.dummy_read_bits  =     1;  
      cfg.readable         =  true;  // true
      cfg.invert           = false;  // true
      cfg.rgb_order        = false;  // true
      cfg.dlen_16bit       = false;  // 16bit SPI true
      cfg.bus_shared       =  true;  // SD true(drawJpgFile)

      _panel_instance.config(cfg);

//----------------------------TOUCH <<< This is what is not working
auto cfg = _touch_instance.config();
      
      cfg.x_min      = 0;    // Minimum X value (raw value) obtained from touch screen
      cfg.x_max      = 319;  // Maximum X value (raw value) obtained from the touch screen
      cfg.y_min      = 0;    // Minimum Y value (raw value) obtained from touch screen
      cfg.y_max      = 479;  // Maximum Y value (raw value) obtained from the touch screen

      cfg.bus_shared = false; // Set to true if you are using the same bus as the screen
      cfg.offset_rotation = 0;// Adjustment when the display and touch orientation do not match Set with a value from 0 to 7
      
      cfg.pin_rst    = 25;
      cfg.pin_int    = 21;   // Pin number to which INT is connected

      cfg.i2c_port = 1;      // I2C(0 or 1) Must be 1 to use Wire library not Wire1 (I tested with Wire and it read)
      cfg.i2c_addr = GT911_ADDR1;   // I2C I tested the GT911_ADDR1 which is (uint8_t)0x5D using TAMC_GT911 and got touches
      cfg.pin_sda  = 33;     // SDA correct
      cfg.pin_scl  = 32;     // SCL correct
      cfg.freq = 400000;     // I2C from the example
      
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance); 
```

I even looked into your code and got this to spit a 'true':
```
  result = lgfx::i2c::init(1, 33, 32).has_value();
  Serial.print("Init port 1:");
  Serial.println(result); //prints 1
```

This line:
`_inited = lgfx::i2c::init(_cfg.i2c_port, _cfg.pin_sda, _cfg.pin_scl).has_value() && _writeBytes(gt911cmd_getdata, 3);`
The `_writeBytes(gt911cmd_getdata, 3);` part returns 0 or it seems to return 0.

lovyan03, 2022-11-04:
Sorry, there is not enough information, please use the Issue template and fill in the required fields

https://github.com/lovyan03/LovyanGFX/issues/new?assignees=&labels=&template=bug-report--------.md&title=

Please provide the smallest, buildable, code that can reproduce the phenomenon. Code with only the configuration part will not allow me to test under the same conditions.

lovyan03, 2022-11-04:
And, if possible, try to get it from the develop branch.
Issues that are not resolved in the release version may be resolved in the develop branch.

lovyan03, 2022-11-04:
@lucasmontec 
Thanks for the detailed information.

I got this information about ESP32-3248SO35 from an acquaintance, please try the following.
When CPU is 240MHz, INT GPIO21 does not function properly, so specify -1 for INT.
If you set CPU 160MHz, INT GPIO21 works.

lucasmontec, 2022-11-05:
I'll try it as soon as possible and answer here.
Just needed to add that I was able to work with LVGL and the TAMC GT911, so that GPIO21 should be working just fine.

rzeldent, 2022-11-07:
Hi Lucas,

I do have the same board and have pin 21 set to -1. I got it working and adapted a Tetris game: [https://github.com/rzeldent/ESP32-3248S035-Tetris](https://github.com/rzeldent/ESP32-3248S035-Tetris). In this repo, the pin21 is set to -1 and it "works".

However, I also made an integration with lvgl and here I really cannot get the GT911 using LovyanGFX @ ^0.4.18 to work, Even with the INT pin set to -1.  It looks like the GT911 implementation is not very stable/timing?

I have it working the TAMC GT911 driver and TFT_eSPI and this is working with lvgl. I ordered some resistive touch boards to see it this works.

Please let me know if you make any progress because this is also an impediment for me.

lucasmontec, 2022-11-08:
Yeahh! I actually did the same thing! I'm using TAMC GT911 and LVGL for this. The issue is, Lovyan is faster and has a better API than LVGL... I think the resistive touch probably works.

rzeldent, 2022-11-08:
I hope the resistive touch works better, most of the examples are with the 3238S036R (R = probably resistive) with the XPT2046. A few with the GT911 but code seems a bit hacked in the examples for the board. However, I'd prefer capacitive. Also, using LovyanGFX is interetsing because it looks less heavy and seems to perform better. The best option would, to use only the drivers. LVGL contains all the drawing stuff/fonts etc and is this is not needed from Lovyan or TFT_eSPI. I tried  [https://github.com/lvgl/lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) but this is not up to date for lvgl 8 and did not get it to work... Maybe you did? What now?

Zer0-bit, 2022-12-02:
@lovyan03 Hello and thank you for all your hard work!

I'll gladly create a separate issue if this is what's best but seeing how it's the same panel and  driver i thought there's no need.

I have the same issue using the pre-defined user settings from the develop branch https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx_user/LGFX_ESP32S3_RGB_ESP32-8048S043.h

I'm actually using the 4.3" IPS(800x480) screen this header is targeting and if i flash the provided calibration example including the above header i just end up seeing the drawing screen without being able to draw as no touch, the calibration phase is skipped which tells me it doesn't seem to detect any touch panel ?
I know the work on that header is in progress still but not sure whether in your tests it works and doesn't in mine ?

tobozo, 2022-12-02:
@Zer0-bit those predefined user settings are templates, have you tried to import the code into your sketch and use a different pinout?

@lucasmontec feel free to [create a discussion](https://github.com/moononournation/Arduino_GFX/discussions/new) in the Q&A section

Zer0-bit, 2022-12-02:
> @Zer0-bit those predefined user settings are templates, have you tried to import the code into your sketch and use a different pinout?
> 
But the pinout is exactly the same as specified there, i mean i'm using the exact model the template was created for unless i'm missing smth reading that ofc?

tobozo, 2022-12-02:
@Zer0-bit do you mean you're also using an ESP32-8048S043? Have you tried the suggested workarounds? Looks like you're expecting some kind of commercial support, but this is a community site :wink: 

Please make your own research and provide the results before asking for support from the community.

Zer0-bit, 2022-12-02:
> @Zer0-bit do you mean you're also using an ESP32-8048S043? Have you tried the suggested workarounds? Looks like you're expecting some kind of commercial support, but this is a community site 😉
> 
> Please make your own research and provide the results before asking for support from the community.

Lol i mean i've not been trying to be perceived as trying to get commercial level support tbh not even sure what in my sentence sparked this thought, i was just trying to understand whether the screen's touch has been working at all in testing since is saw a template in the user settings that was targeting this exact model ( yes i have  ESP32-8048S043 ).
I'm atm on my 4th day of trying to get this thing working so did quite a bit of research and tried all the workarounds i was able to find(including in this thread) prior to posting, i mean i'm running my own OOS project so i know how absurd some issues or requests can get, pls hold fire lol.

closed 2022-12-25 by github-actions[bot]

rzeldent, 2023-02-28:
Hi Simone,

Thanks for reviving this thread. I had lots of trouble getting it to work with Lovyan so started a github repo just for these boards for LVGL.
Besides the display it also covers the led, light sensor etc... [esp32-smartdisplay](https://github.com/rzeldent/esp32-smartdisplay)
Maybe this works also for you...

rayene01, 2025-08-19:
Hello, any solution to get GT911 Touch work with LVGL in ESP-IDF?